johanmeijer/grott #189: Incorrect timestamp values

mydliar, 2022-10-17:
Hi there - first of all, great piece of work, I'm just starting using grott, but it is already becoming an indispensable part of my HA automations.

I'm encountering this bizarre issue, has anyone else seen it? Looking at the grott messages in my MQTT broker, I can see the timestamps to be:
- Never quite current
- Sometimes behind by hours
- Sometimes behind by 5-6 minutes

For ref, these are screenshots from today:
[image]

[image]

[image]

[image]

I doubt it's a problem with the MQTT client, I'm using two different ones, and both show the same messages. I'm using eclipse-mosquitto broker running on Synology/Docker. And on the same Docker I'm also running Grott, with TZ: Europe/London. 

The actual Grott values which I process in Home Assistant 'seem' OK, i.e. aligning with what Growatt servers are showing. But it's hard to tell what if any impact there is, since Growatt servers are also fed from my Grott service. It just makes me nervous about seeing the timestamp fluctuate like this. 

Any ideas what I'm missing?

Thanks,

Martin

johanmeijer, 2022-10-17:
Some dataloggers (shinewifi) seems to have problems with keeping the time and show a enormous timedrift. Time is being set during initialisation phase e.g. in the morning when the inverter start again (this does not works when the inverter is always on). You can also force this by stopping / starting the connection to the growatt server (or stop/start Grott). 

Grott can correct this by using the server time instead of the time provided by the datalogger, You have to specify time = server  in grott.ini [Generic] section or gtime = "server" as docker environmental variable. 

I have not heard that growatt server has problems with it but can understand that if there are records buffered and sent later this can cause some issues. 

Grott sent time as part of the MQTT message and and it is also use for the influxdb timestap and PVOutput. If you do not use /ignore the time keyword from the MQTT (probably HA does that) it is not a problem. But for influxDB and PVOutput it is a problem.

mydliar, 2022-10-18:
Hi, thanks for this. I have added 'time = server' into my grott.ini and the timestamps are now within 2-3 seconds.

Yes, I can imagine an outage and buffered records could cause some sort of problem.

I am also adopting a nightly script to set the battery charge rate, so I'll consider updating the inverter time in this script too. Interestingly enough, whatever the timestamp from shinewifi logger shows, the inverter behaves correctly at specified times (e.g. starts charging the battery at 0:30am, as per my settings). So it must know the correct time event though the MQTT timestamp is wrong. 

Issue can be closed, thanks again.

closed 2022-10-19 by johanmeijer