johanmeijer/grott #196: Home Assistant addon not being maintained?

Maes152, 2022-10-28:
The wiki refers to the repo from [wjansenw](https://github.com/wjansenw/home-assistant-add-on) but it looks like this is not maintained anymore. Last update was March 2021. 

I did find a fork from this repo by [Emacee ](https://github.com/Emacee/grott-home-assistant-add-on) which seems to be a lot newer (July 26 2022) but this repo is also quite outdated, running grott 2.3.1

Maes152, 2022-10-28:
So it looks like the docker image updates itself, however the configuration file is out of date. This repo seems to be up-to-date:
[https://github.com/boris-burgos/home-assistant-add-on](url)

Maes152, 2022-10-28:
So the repo from boris-burgos isn't working either, it has an incorrect config file.
I made a fork and made some changes. Whoever is reading this, this one is fully working and up-to-date:
[](https://github.com/Maes152/grott-home-assistant-add-on/)

johanmeijer, 2022-10-31:
I am not using HA at this moment so keeping HA add-on config up to date is very hard for me. I rely on the users of HA to do this. 

I will remove the link to the wjansew repo. There are also (I think up to date? ) config sample in the example directory. 

There is also a lot of discussion / issues on this. The latest is issue #181 where @shinkie is looking into this.

Freekers, 2023-02-16:
This one is (also) still being maintained: https://github.com/egguy/grott-home-assistant-add-on