johanmeijer/grott #217: Making sense of Grott Values

farsonic, 2022-11-04:
I'm trying to make sense of the values being presented by Grott (and exported using MQTT) compared to what I'm seeing from the Growatt site. 

[image]

How do I rationalise the values?

johanmeijer, 2022-11-04:
Interesting. Are they from the same time?

farsonic, 2022-11-04:
Yes, my recordings from grott seem to have a higher value that matches the growatt site then a smaller value that doesn’t make sense.

johanmeijer, 2022-11-07:
What type of inverter do you have?

farsonic, 2022-11-07:
5000TL-X

W1ll3mp13, 2022-11-07:
I'm having the same issue. It looks like the values from Grott are shuffled.
There is some delay of the data in below screenshots (The grott data was from this afternoon and the GroWatt data from just now), but I guess the output data of for example epv1today (4619.6) is the total energy and should be placed at the value of pvenergytotal.
The inverter I have is 4000 TL3-S.
Is there a way to get the correct values at the correct properties? As a work around I can also manipulate the sensors I'm using in Home assisant to get the correct values in Home assistant. 
[image]
[image]

johanmeijer, 2022-11-11:
@W1ll3mp13 for the TL3-S you have to specify invtype = tl3 or ginvtype = "tl3" (docker).

johanmeijer, 2022-11-11:
@farsonic you also have a energy monitor (SDm630)  II believe. Is it possible that you try to compare the energy monitor data with the inverter (solar) data?

farsonic, 2022-11-11:
That’s possible, let me check today

W1ll3mp13, 2022-11-12:
> @W1ll3mp13 for the TL3-S you have to specify invtype = tl3 or ginvtype = "tl3" (docker).

Dankjewel Johan. Het werkt :-)

johanmeijer, 2022-11-12:
Super, fijn dat het wekt!

closed 2022-11-12 by johanmeijer