johanmeijer/grott #223: Bug in generated_pv_energy_today

RienduPre, 2022-11-05:
https://github.com/johanmeijer/grott/blob/198ef4735145622cc4cfeddaff2bb98992d0aafa/examples/Home%20Assistent/grott_ha.py#L193

value_json.epvTodaycontains a capital (T) and must be lowercase (t)

egguy, 2022-11-07:
Hi, thanks for the bug report.

It seems I was confused by the SPH inverter, which has an `epvToday` field while the others are `epvtoday`.
- https://github.com/johanmeijer/grott/blob/d6f6947f6d2ed722d4ea66172768cd7cd1a59a97/grottconf.py#L875

I will try to correct it soon in the -current version of the HA addon.

johanmeijer, 2022-11-08:
Oké. We can fix this but this can cause an incompatibility for people already using this in their mqtt processing or influxdb.

The solution would be not to change it, but add a second key I think (other name same offset). They can then be used next to it each other.

Backwards compatibility is always a think to keeping mind ;)

RienduPre, 2023-02-15:
WhT’s the status on this?

egguy, 2023-02-15:
Hello,
There's now an auto generation of the correct template, this prevents case problem, the epvtoday in the current version is using this function and address this problem.
ex: https://github.com/egguy/grott/blob/feature/ha-extension/examples/Home%20Assistent/grott_ha.py#L167