johanmeijer/grott #238: Reimport historic data

codac, 2022-11-18:
Everything is running ok so far. My Grott Docker didnt run für one day, which is why data is missing now in the influxdb.
As I am accumulating values over weeks or months or even years, this is destroying all my numbers.
So I would need to get the values of lets say the day before yesterday reimportet. Is that possible yet and if, how?

johanmeijer, 2022-11-19:
Normally if Grott is not running this will not result in any loss of data while the data that is not send is buffered in the datalogger and sent later. 

However some dataloggers have problems with keeping the time or do not send the time in the data record. Grott can then put these records in the influxdb (while that is a timeseries database). 

If you have the information kept in an other way (like pvoutput) we might be able to create a python tool to add the missed records.

codac, 2022-11-20:
The weird thing is. The data of one day is missing on the grott server (there is no entry in the protocol for one day) hence it is not in the InfluxDB, it is missing on the growatt server website as well.
But it is available in the growatt App and in the CSV Report, that I have enabled in order to be received by email.
I dont know the root cause, but if there is a python script in order to push the datatransfer again that is buffered on the datalogger, that would be a big help.

johanmeijer, 2023-02-06:
I am not sure the status of this one. We can start a script to force the buffered records. When a ne connection is being made the Datalogger start sending the Buffered records. 

Do you also use PVOutput? Grott seems to skip Influxdb writing if there is a PVOutput error (and that is pretty normal when sending a bulk of buffered records in a short timeframe).  I will fix that in a next release!

codac, 2023-02-07:
@johanmeijer I'm using pvpowerout.
Another case, but same interest: Due to a wrong retention setting I've lost the first month of data in the InfluxDB. 
I would love to re-add this data to InfluxDB.

codac, 2023-04-06:
@johanmeijer any update on this?

johanmeijer, 2023-04-13:
Euh. How can we re-add this? Do you still have the data (in pvoutput or growatt server/app)? Maybe we can create a script to add this data to influxDB?