johanmeijer/grott #239: Grott connection error: peer has disconnected

RedFoxy, 2022-11-21:
Hi!
I've an inverter Growatt SPH3000 connected to a ShineLanBox, I want get the information to show it in Home Assistant, so just I installed grott as docker in my docker compose, I've all Home Assistant components in the same docker-compose like:

```
version: "3.9"
services:

  mqtt:
    container_name: mqtt
    image: eclipse-mosquitto
    restart: always
    privileged: false
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - ./config/mosquitto:/mosquitto/config
      - ./data/mosquitto:/mosquitto/data
      - ./log/mosquitto:/mosquitto/log
    ports:
      - "1883:1883"
      - "9001:9001"
    networks:
      domotica:
        ipv4_address: 172.30.1.220
        aliases:
          - mqtt

  grott:
    container_name: grott
    image: ledidobe/grott
    restart: unless-stopped
    privileged: false
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - ./config/grott/grott.ini:/app/grott.ini
    ports:
      - "5279:5279"
    networks:
      domotica:
        ipv4_address: 172.30.1.222
        aliases:
          - grott
    environment:
      - TZ=Europe/Rome

networks:
  domotica:
    driver: bridge
    driver_opts:
      com.docker.network.bridge.name: br-domotica
    enable_ipv6: false
    ipam:
      driver: default
      config:
        - subnet: 172.30.1.0/24
```

grott.ini file is:
```
[Generic]
mode = proxy

[Growatt] 

[MQTT]
ip = 172.30.1.220
port = 1883
topic= energy/growatt
auth = True
user = [username]
password = [password]

[PVOutput]

[influx]

[extension] 
```

Than I logged in my Growatt Shinelink/-X Setting Center (IP: 192.168.0.18) and under NetWork Setting I changed:
ResolvDomain -> Off
Server IP -> 192.168.0.100 (Server IP with docker)

But I receive follows errors:
         - Grott connection error
         - peer has disconnected
         - ('192.168.0.18', 1057) has connected
         - Grott connection error
         - peer has disconnected
         - ('192.168.0.18', 1058) has connected
         - Grott connection error
         - peer has disconnected
         - ('192.168.0.18', 1059) has connected
         - Grott connection error
         - peer has disconnected
         - ('192.168.0.18', 1060) has connected
        
The server has lastes Debian with iptables not enable to filter any traffic, docker it's lastes version as the Grott docker image.

RedFoxy, 2022-11-21:
Today:
```
         - Grott connection error
         - peer has disconnected
         - ('192.168.0.18', 1112) has connected
         - Grott connection error
         - peer has disconnected
         - ('192.168.0.18', 1113) has connected
         - ('192.168.0.18', 1114) has connected
         - Grott connection error
         - peer has disconnected
         - ('192.168.0.18', 1116) has connected
[Errno 111] Connection refused
         - Can't establish connection with remote server.
         - Closing connection with client side ('192.168.0.18', 1117)

         - Growatt packet received:
                  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.30.1.222', 36104), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.30.1.222', 36104), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=7, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.30.1.222', 5279), raddr=('192.168.0.18', 1116)>
         - Data less then minimum record length, data not processed
         - ('47.91.67.66', 5279) has disconnected
[Errno 111] Connection refused
         - Can't establish connection with remote server.
         - Closing connection with client side ('192.168.0.18', 1118)
         - ('192.168.0.18', 1119) has connected
         - ('192.168.0.18', 1120) has connected

         - Growatt packet received:
                  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.30.1.222', 40850), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=7, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.30.1.222', 5279), raddr=('192.168.0.18', 1119)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.30.1.222', 40850), raddr=('47.91.67.66', 5279)>
         - Growatt original Data:
                 \x00\xce\x00\x05\x02\x19\x50\x03\x09\x33\x2c\x44\x58\x45\x41\x75\x4b\x5c\x39
                 \x2c\x37\x4c\x7f\x46\x5e\x47\x50\x35\x74\x47\x72\x6f\x77\x61\x76\x74\x47\x72
                 \x13\x77\x60\x74\x74\x47\x72\x6f\x13\x61\x10\x53\x57\x72\x6f\x7c\xd9\x7a\x64
                 \x15\x33\x5e\x59\x51\x54\x2e\x04\x10\x0e\x77\x60\x74\x74\x47\x72\x6a\xab\x61
                 \x6a\x75\x6b\x72\xa7\x77\x05\x74\x74\x09\x3f\x2c\x4f\x59\x40\x45\x77\x43\x2e
                 \x76\x61\x86\x50\x47\x73\x6f\x2d\x61\x74\x74\x47\x52\x4f\x39\x14\x1b\x02\x26
                 \x52\x2a\x19\x04\x06\x13\x2e\x13\x4f\x77\x61\x79\xd9\x47\x72\x68\x91\x61\x7f
                 \x74\x52\x72\x67\x77\x53\x74\x4c\x47\x73\x68\xd4\x68\x96\x67\x33\x61\xf3\x76
                 \x38\x7e\x21\x55\xfc\x7b\x69\x61\x74\x74\x47\x72\x6f\x77\x61\x73\xd7\x4e\x90
                 \x7c\x09\x72\xe6\x74\x0c\x72\xf9\x77\x6b\x74\x7e\x47\x77\x6f\x72\x61\x71\x74
                 \x42\x72\x6f\x77\x61\x74\x74\x47\x72\x66\x95\x61\x74\x26\x06\x33\x2e\x46\x59
                 \x45\x43\x77\x43\x6f\x77\x60\x45\x74\x47\x72\x6f\x64\xfd\x74\x6c\x4e\x90\x66
                 \xc3\x69\x62\x7c\x03\x72\x7b\x77\x64\x7d\x1b\x4f\x8e\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\xe4\x6f\x77\x60\x90\x74\xb8\x3c\x4f\x77\x9e\x3a
                 \x54\x47\x8d\x21\x57\x61\x8b\x3a\x67\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x74\x74\x47\x71\x87\x73\x05\x63\x04\x47\x72\x7d\xb7\x61\x46\x75\x91\x70
                 \x50\x65\xa1\x62\xdc\x45\xd2\x6b\x3b\x63\xc8\x70\x0b\x70\x37\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x42\x72\x6a\x77\x05\x74\x10\x47
                 \x72\x6f\x77\x61\x10\x74\x23\x72\x6f\x77\x60\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x46\x72\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x10\x47\x77
                 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x05
[... 367 lines cut ...]
                 \x5f\x4e\x5b\x44\x47\x7d\x42\x57\x12\x78
         - Grott automatic protocol detection
         - Grott data record length 181
         - layout   :  T055029
         - no matching record layout found, try generic
         - Record layout used :  T055029
         - Growatt data decrypted V2
         - Grott Growatt data decrypted
         - Growatt plain data:
                 00d4000500ad50294e4143333931353239334d5943373931353239334e4d43383834313031410
                 dad000400013500050001310006000233320008001e4d59433739313532393300000000000000
[... cut ...]
                  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.30.1.222', 40850), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=7, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.30.1.222', 5279), raddr=('192.168.0.18', 1119)>
         - Data less then minimum record length, data not processed

```

johanmeijer, 2022-11-21:
The end part of the log looks oké. The beginning has some connection errors. I am not sure why. I see both connection drops at the growatt server and inverter side.

Do you see this connection errors always or only at inverter startup?

RedFoxy, 2022-11-22:
It's like a mix, I receive a lot of:
```
         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.30.1.222', 47654), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.30.1.222', 47654), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.30.1.222', 47654), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.30.1.222', 47654), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.30.1.222', 5279), raddr=('10.0.0.178', 1038)>
         - Data less then minimum record length, data not processed

         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.30.1.222', 47654), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed
```

And I noticed that sometimes the datas that it receive are corrupted, like serial number with strange characters or unusefull datas, like:
```
         - MQTT jsonmsg: 
                         {"device": "9,7L\u007fF^GP5", "time": "2022-11-22T01:22:45", "buffered":
                         "no", "values": {"pvstatus": 30567, "pvpowerin": 1953777522, "pv1voltage":
                         28354, "pv1current": 24948, "pv1watt": 1950839403, "pv2voltage": 30561,
                         "pv2current": 29812, "pv2watt": 1198681975, "pvpowerout": 1198681975,
                         "pvfrequentie": 24948, "pvgridvoltage": 29767, "pvgridcurrent": 29295,
                         "pvgridpower": 2002875508, "pvgridvoltage2": 18290, "pvgridcurrent2": 28535,
                         "pvgridpower2": 1635021895, "pvgridvoltage3": 29295, "pvgridcurrent3":
                         30561, "pvgridpower3": 1953780984, "totworktime": 1950839407,
                         "pvenergytoday": 2002875508, "pvenergytotal": 1198681975, "epvtotal":
                         1635021895, "epv1today": 1919874261, "epv1total": 2132171653, "epv2today":
                         1870094708, "epv2total": 1950799584, "pvtemperature": 18016,
                         "pvipmtemperature": 28793}}
```

and not:
```

         - MQTT jsonmsg: 
                         {"device": "NMC884101A", "time": "2022-11-22T01:22:43", "buffered": "no",
                         "values": {"pvstatus": 6, "pvpowerin": 0, "pv1voltage": 437, "pv1current":
                         0, "pv1watt": 4, "pv2voltage": 0, "pv2current": 0, "pv2watt": 0,
                         "pvpowerout": 0, "pvfrequentie": 0, "pvgridvoltage": 0, "pvgridcurrent": 0,
                         "pvgridpower": 0, "pvgridvoltage2": 0, "pvgridcurrent2": 0, "pvgridpower2":
                         0, "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3": 5002,
                         "totworktime": 0, "pvenergytoday": 0, "pvenergytotal": 0, "epvtotal": 0,
                         "epv1today": 99252, "epv1total": 190979319, "epv2today": 0, "epv2total":
                         107663, "pvtemperature": 274, "pvipmtemperature": 534}}
```

I tryied to restart the docker 2 times but I don't got again the connections errors

RedFoxy, 2023-04-10:
Just I removed grott I'll test it again later

closed 2023-04-10 by RedFoxy