johanmeijer/grott #246: Can't change Growatt Data Logger IP address

supermas, 2022-11-29:
I am running Docker on a Raspberry Pi Zero following your instructions.

The Raspberry Pi IP address is 192.168.1.226

I run:  docker run ledidobe/grottrpi

and I get this output:
Grott Growatt logging monitor : 2.6.1g

Grott Command line parameters processed:
        verbose:         True
        config file:     grott.ini
        output file:     <_io.TextIOWrapper name='<stdout>' mode='w' encoding='UTF-8'>
        nomqtt:          False
        inverterid:      automatic
        pvoutput:        False
        blockcmd:        False
        noipf:           False

Grott process configuration file

Grott process environmental variables

Grott override settings if set in commandline

Grott process json layout files
t0060104sph.json

Grott layout records loaded
T02NNNN  :
{'decrypt': {'value': 'False'}, 'pvserial': {'value': 36, 'length': 10, 'type': 'text', 'divide': 10}, 'date': {'value': 56, 'divide': 10}, 'recortype1': {'value': 70, 'length': 2, 'type': 'num', 'incl': 'no'}, 'recortype2': {'value': 74, 'length': 2, 'type': 'num', 'incl': 'no'}, 'pvstatus': {'value': 78, 'length': 2, 'type': 'num'}, 'pvpowerin': {'value': 82, 'length': 4, 'type': 'num', 'divide': 10}, 'pv1voltage': {'value': 90, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1current': {'value': 94, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1watt': {'value': 98, 'length': 4, 'type': 'num', 'divide': 10}, 'pv2voltage': {'value': 106, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2current': {'value': 110, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2watt': {'value': 114, 'length': 4, 'type': 'num', 'divide': 10}, 'pvpowerout': {'value': 122, 'length': 4, 'type': 'num', 'divide': 10}, 'pvfrequentie': {'value': 130, 'length': 2, 'type': 'num', 'divide': 100}, 'pvgridvoltage': {'value': 134, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent': {'value': 138, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower': {'value': 142, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage2': {'value': 150, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent2': {'value': 154, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower2': {'value': 158, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage3': {'value': 166, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent3': {'value': 170, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower3': {'value': 174, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytoday': {'value': 182, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytotal': {'value': 190, 'length': 4, 'type': 'num', 'divide': 10}, 'totworktime': {'value': 198, 'length': 4, 'type': 'num', 'divide': 7200}, 'pvtemperature': {'value': 206, 'length': 2, 'type': 'num', 'divide': 10}, 'isof': {'value': 210, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'gfcif': {'value': 214, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'dcif': {'value': 218, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'vpvfault': {'value': 222, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'vacfault': {'value': 226, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'facfault': {'value': 230, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'tmpfault': {'value': 234, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'faultcode': {'value': 238, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'pvipmtemperature': {'value': 242, 'length': 2, 'type': 'num', 'divide': 10}, 'pbusvolt': {'value': 246, 'length': 2, 'type': 'num', 'divide': 10, 'incl': 'no'}, 'nbusvolt': {'value': 250, 'length': 2, 'type': 'num', 'divide': 10, 'incl': 'no'}, 'epv1today': {'value': 278, 'length': 4, 'type': 'num', 'divide': 10}, 'epv1total': {'value': 286, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2today': {'value': 294, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2total': {'value': 302, 'length': 4, 'type': 'num', 'divide': 10}, 'epvtotal': {'value': 310, 'length': 4, 'type': 'num', 'divide': 10}, 'rac': {'value': 318, 'length': 4, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'eractoday': {'value': 326, 'length': 4, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'eractotal': {'value': 334, 'length': 4, 'type': 'num', 'divide': 1, 'incl': 'no'}}
T05NNNN  :
{'decrypt': {'value': 'True'}, 'pvserial': {'value': 36, 'length': 10, 'type': 'text', 'divide': 10}, 'date': {'value': 56, 'divide': 10}, 'recortype1': {'value': 70, 'length': 2, 'type': 'num', 'incl': 'no'}, 'recortype2': {'value': 74, 'length': 2, 'type': 'num', 'incl': 'no'}, 'pvstatus': {'value': 78, 'length': 2, 'type': 'num'}, 'pvpowerin': {'value': 82, 'length': 4, 'type': 'num', 'divide': 10}, 'pv1voltage': {'value': 90, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1current': {'value': 94, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1watt': {'value': 98, 'length': 4, 'type': 'num', 'divide': 10}, 'pv2voltage': {'value': 106, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2current': {'value': 110, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2watt': {'value': 114, 'length': 4, 'type': 'num', 'divide': 10}, 'pvpowerout': {'value': 122, 'length': 4, 'type': 'numx', 'divide': 10}, 'pvfrequentie': {'value': 130, 'length': 2, 'type': 'num', 'divide': 100}, 'pvgridvoltage': {'value': 134, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent': {'value': 138, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower': {'value': 142, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage2': {'value': 150, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent2': {'value': 154, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower2': {'value': 158, 'length': 4, 'type': 'num', 'divide': 10}, 'pvgridvoltage3': {'value': 166, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridcurrent3': {'value': 170, 'length': 2, 'type': 'num', 'divide': 10}, 'pvgridpower3': {'value': 174, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytoday': {'value': 182, 'length': 4, 'type': 'num', 'divide': 10}, 'pvenergytotal': {'value': 190, 'length': 4, 'type': 'num', 'divide': 10}, 'totworktime': {'value': 198, 'length': 4, 'type': 'num', 'divide': 7200}, 'pvtemperature': {'value': 206, 'length': 2, 'type': 'num', 'divide': 10}, 'isof': {'value': 210, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'gfcif': {'value': 214, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'dcif': {'value': 218, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'vpvfault': {'value': 222, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'vacfault': {'value': 226, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'facfault': {'value': 230, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'tmpfault': {'value': 234, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'faultcode': {'value': 238, 'length': 2, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'pvipmtemperature': {'value': 242, 'length': 2, 'type': 'num', 'divide': 10}, 'pbusvolt': {'value': 246, 'length': 2, 'type': 'num', 'divide': 10, 'incl': 'no'}, 'nbusvolt': {'value': 250, 'length': 2, 'type': 'num', 'divide': 10, 'incl': 'no'}, 'epv1today': {'value': 278, 'length': 4, 'type': 'num', 'divide': 10}, 'epv1total': {'value': 286, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2today': {'value': 294, 'length': 4, 'type': 'num', 'divide': 10}, 'epv2total': {'value': 302, 'length': 4, 'type': 'num', 'divide': 10}, 'epvtotal': {'value': 310, 'length': 4, 'type': 'num', 'divide': 10}, 'rac': {'value': 318, 'length': 4, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'eractoday': {'value': 326, 'length': 4, 'type': 'num', 'divide': 1, 'incl': 'no'}, 'eractotal': {'value': 334, 'length': 4, 'type': 'num', 'divide': 1, 'incl': 'no'}}
T06NNNN  :
{'decrypt': {'value': 'True'}, 'pvserial': {'value': 76, 'length': 10, 'type': 'text', 'divide': 10}, 'date': {'value': 136, 'divide': 10}, 'recortype1': {'value': 150, 'length': 2, 'type': 'num', 'incl': 'yes'}, 'recortype2': {'value': 154, 'length': 2, 'type': 'num', 'incl': 'yes '}, 'pvstatus': {'value': 158, 'length': 2, 'type': 'num'}, 'pvpowerin': {'value': 162, 'length': 4, 'type': 'num', 'divide': 10}, 'pv1voltage': {'value': 170, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1current': {'value': 174, 'length': 2, 'type': 'num', 'divide': 10}, 'pv1watt': {'value': 178, 'length': 4, 'type': 'num', 'divide': 10}, 'pv2voltage': {'value': 186, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2current': {'value': 190, 'length': 2, 'type': 'num', 'divide': 10}, 'pv2watt': {'value':  [... cut ...]
        token:           **secret**
_Extension:
        extension:       False
        extname:         grottext
        extvar:          {'none': 'none'}

Grott proxy mode started
Hostname : cb88cbf3cd1b
IP :  172.17.0.2 , port :  5279

All appears okay so far although I'm not sure why the listed IP is 172.17.0.2

When I login to the Growatt server and try and change the IP address of my Data Logger to 192.168.1.226 and enter the date as password, it says success in green but nothing happens?

[image]

What am I doing wrong?

chirs-51, 2022-12-05:
I have the exact same problem :/

did you find out how?

supermas, 2022-12-05:
No answers yet but I have a horrible feeling we're going to have to do it from the Data Logger Dongle itself.

robupham, 2022-12-05:
Have you checked that the container is reachable at your IP? i.e. can you connect to 192.168.1.226 on port 5279?
If not, you need to fix your Docker forwarding, otherwise no amount of fiddling with the data logger settings will help.

johanmeijer, 2022-12-05:
If server IP is being set to the grott address and it does not reach grott then it will not be possible to communicate with the inverter from the growatt site (and no statics will be shown anymore). 

If you still are able to set  or read things then probably the address in not set (or grott is doing its thing).

supermas, 2022-12-05:
Good point.

The last lines of the Docker grottrpi say:

Grott proxy mode started
Hostname : 33c95e2349a3
IP :  172.17.0.2 , port :  5279

Do I assume the IP listed is the one I should be connecting to?

If so, why is it not in my network range, 192.168.*.*?

johanmeijer, 2022-12-05:
It will be your network range 192.168... the 172.17 is the internal docker IP address.

johanmeijer, 2022-12-05:
If you specified the docker port at 5279 (-p 5279:5279) then grott address will be 192.168.x.y:5279.

supermas, 2022-12-05:
Ah, okay.

At this point I should state that I am a Docker newby and this was my first project.

So after running "docker run ledidobe/grottrpi" and seeing all the config go past, the interface stays open (no prompt)

I have to Ctrl C to get back to prompt.

Is that expected behaviour?

supermas, 2022-12-05:
I installed the image and just ran it.

The instructions seemed to suggest that defaults would kick in and all would be well.

supermas, 2022-12-05:
Is the Growatt Data Logger clever enough to check it gets a valid response from the IP before it lets you switch to it?

johanmeijer, 2022-12-05:
I do not think so.

johanmeijer, 2022-12-05:
If you have started the docker container you can look in the log with docker logs command (docker logs grott). You can see if something is being sent via grott.

You can also try to go grott via a browser page by 192.168.x.y:5279. You will see nothing in the browser but should see something in the grott log.

supermas, 2022-12-05:
Okay a step further..

Re-ran docker with command line params: 
docker run -d -p 5279:5279 --restart on-failure -e gmqttip="192.168.1.226" ledidobe/grott

prompt came back.

Checked logs - looking good.

pointed browser at 192.168.1.226:5297 - spinning..

Re-checked logs and new message:  Growatt packet received: etc..

supermas, 2022-12-05:
Next issue:
My MQTT server has authentication.
When I try to run grottrpi with these variables:

docker run -d -p 5279:5279 --restart on-failure -e gmqttip="192.168.1.228" -e gmqttauth="True" -e gmqttuser="mqtt_user" -e gmqttpassword="**********" ledidobe/grottrpi

I get this error:

docker: Error response from daemon: driver failed programming external connectivity on endpoint practical_curie (b9d5a776d5c0fb6193b87dbd6d8751641f231a6d3c52b943233b1b1f05e74338): Bind for 0.0.0.0:5279 failed: port is already allocated.

supermas, 2022-12-05:
Now if I could just get my Data Logger IP to change..

johanmeijer, 2022-12-06:
Otherwise you have to reset your datalogger en start configuring it again.

johanmeijer, 2023-02-06:
@supermas is this one solved?

supermas, 2023-02-06:
I've put this on the back burner for now.

Thanks for all your help getting this far.

catch56, 2023-02-09:
I ran into the same problem, but just resolved it after an hour or so's trial and error, this is with a Shine Wifi-S

What didn't work: editing the datalogger configuration via the web UI and changing the IP. The 'Successful' message is a lie.

What did work:
Using the ShinePhone app, click the + in the top right hand corner and list dataloggers.

Go to 'Datalogger configuration'

Put the Shine Wifi-S into AP mode, this requires physically pressing the reset switch on the dongle itself.

Once in AP mode and connected to the hotspot (pw: 12345678), go to advanced settings - change the server IP there. The interface for this is completely different even though the setting is theoretically the same. You can also change the update frequency here if you want it lower than the standard 5 minutes.

Exit AP mode.

After this I started getting output from grott/mqtt.

ChristiaanW123, 2023-02-25:
> I ran into the same problem, but just resolved it after an hour or so's trial and error, this is with a Shine Wifi-S
>
> What didn't work: editing the datalogger configuration via the web UI and [...]

Change it to what exactly? I understand the 172.x.x.x IP is not the correct one and it should be 192.168.x.y. But how do I find what x and y it is using?