johanmeijer/grott #247: Adjust server on GrowattshineWIFI-x

Raspb-henk, 2022-12-03:
Hi,
I am installing grott on a RPI with Docker. I have a problem to redirect the output to the PI as there seems to be a new interface (?)
I do the following:
I have a Growatt 5KTL3-XH with a ShineWifi-X module.
-On my (android)Phone I start the Shine App
-On the bottom of the screen I choose "Installation"
-In the menu installation I choose the "+" at the right above corner of te screen
-In the list that opens I choose  the third option "Datalogger lijst"
-In the next screen are the data off the Shine
-I click on the text and get a menu with four options, I choose  the third "Datalogger configuratie"
-The screen "Datalogger configuratie "opens (this is the same screen when you install the datalogger the firtst time)
-I choose "Hotspot-modus"
-The screen asks to short press the button at the bottom of the Shine 
-I go to the Phone systeem screen opens to connect to the hotspot off the shine (shine serialnumber as Network ID)
-After connecting to the Shinenetwork I go back to the app and get on the next screen there a screen with the option "Advanced settings" 
-This screen opens:
[image], I do not change a thing
I go to the right to "Server instellingen"
[image]
Here I push  the button to the left (Password : growattyyyymmdd). 
Then the field growatt doman went blank and  I could put in the IP adress of my local RPI.

I saved everything and thought it would work but.......  it does not. No data to the RPI. What dit I do wrong? 
 
Henk

johanmeijer, 2022-12-03:
Hi Henk,

I have always used the growatt website for this (can you try this?). I am not using growatt as server any more (and not using the app) so I can not try it directly, 

If I see the message it seems that your not allowed to change this from the app (is only allowed by the installer). That seems the be the new default way of defining the inverter to growatt). 

Hopefully somebody else can response. Otherwise I will try it (have to enable sending to growatt server again). 

I am a little busy with other things (my daughter is moving houses this week and there is still a lot of work to do) so it might take a little bit longer.

Raspb-henk, 2022-12-04:
Hello Johan,

Thanks for the respons. I’ll post this  also on the Issue page because it can be of use for other users. I hope you find time to take a look at it.

 

The option to change the ip adress is still on the  the Growatt website. I searched and found that on the main page (BUTTON Dashboard) the website shows  MY Photovoltaic Devices it show a ShineWif-X. with a button “Settings”. There there is no option to change the IP.  BUT If you click on the right above the button Setting on “ALL devices” there is a second screen in wich the ShineWIFI X is shown as “Datalogger”.  There clicking on “Data Logger Setting” you do  get a menu where you can change the IP address. So I missed this at first ☹

 

I now  tried to change the ip address of the datalogger on the Growatt website and got “Succesfull” as respons. In the Growatt website I still get the data from the panels in the Growatt site.  The logger has status Connection. So I think the proxy on the RPI is working? 😊

 

But : No data is arriving at Pvoutput.org. I have activated port forwarding on the RPI as described in the Grott Wiki  “Rerouting Growatt” for sniffer mode. As I got no connection to Pvoutput I stopped the Docker file and restarted it in proxy mode with :

docker run -d -p 5279:5279 --restart on-failure -e gnomqtt="True" -e gblockcmd="True" -e gpvoutput="True" -e gpvapikey="ff78c5c5e43487cf0d160fd26c315d33c61b0b11" -e gpvsystemid="95101" ledidobe/grottrpi

Up to now no data.

 

Kind regards

Henk.

johanmeijer, 2022-12-05:
Hi Henk, if you run in proxy mode you should activate port forwarding (that bypasses then the Grott proxy).

Raspb-henk, 2022-12-13:
Hi
I have restarted everything which took some time.  I checked that the IP in the Shine has been set to my  RPI (192.168.1.103) . I use a RPI 4 with a fresh installed OS and docker with  Portainer. 
I installed Grottrpi and it is running.  I checked that the IP in the Shine has been set to my  RPI (192.168.1.103). In the log off Grottrpi I see that the data from the Growatt is received. On the website of Growatt  I also receive the data OK and the SHINE APP is also OK. So far so good.

**BUT i do not get data to PV output.**
As it is a three fase inverter I added the command < -e ginvtype ="tl3" > 
I started the container with this command:
docker run -d -p 5279:5279 --restart on-failure -e gnomqtt="True" -e gblockcmd="True" -e ginvtype="tl3" -e gpvoutput="True" -e gpvapikey="ff78c5c5e43487XXXXXXXXc315d33c61b0b11" -e gpvsystemid="95266" ledidobe/grottrpi
 In the log of Grott (attatched) I find this:
	 - Grott send data to PVOutput systemid:  95XX6 for inverter:  XJL0XXX01F
		 -  {'X-Pvoutput-Apikey': 'ff78c5c5e4348XXXXXXXXc61b0b11', 'X-Pvoutput-SystemId': '95XX6'}
		 -  {'d': '20221211', 't': '13:26', 'v1': 76500, 'v2': 39321.6, 'v6': 394.0}
	 - Grott PVOutput response: 
		 **-  Bad request 400: Energy value [76500] too high for system size [5200]**
	XXXXX added by me
In pvoutput the inverter is a Growatt 5000TL3-XH. The id fitts and  is  does have a size of 5200W. (13 x 400WP pannels)

So I Now I am quessing what is wrong ?? From where commes this high value? I think Grott is working OK but is maybe something wrong with the inverter setings? 
'NB when the logfile was made it was dark.

I hope you can point me in the right direction.
Henkj.

[_elastic_noyce_logs.txt](https://github.com/johanmeijer/grott/files/10222626/_elastic_noyce_logs.txt)

Raspb-henk, 2022-12-23:
Closed because initial problem is solved.

closed 2022-12-23 by Raspb-henk