johanmeijer/grott #248: Layout for sph - sunrino M12000 (Sunrino-SP Serie)

camelcamro, 2022-12-04:
Hello
is anyone having a sunrino sph (hybrid inverter layout ?
it seems to be recognized as: 
i tried to make a clone of another ..
finally, the values are maybe not correct ..
here to startup
it is only having 718 length

 - Grott automatic protocol detection
         - Grott data record length 359
         - layout   :  T060103SPH
         - Record layout used :  T060103SPH
         - Growatt data decrypted V2
         - Grott Growatt data decrypted
         - Growatt plain data:
                  00250006015f0103554d43304345583059410000000000000000000000000000000000000000525447304134363031390000000000000000000000000000000000000000221204022356030000002c0000000200010000000000000000000000013034362e30303033372e3031000000000000000200010000000000005254473041343630313900000000000101000000000000a0022602180000000000020000021802264e250640002d00590016000c000400020017003a000100000000000000000000000001180000000000000000000000000000000011f8104b0e750ffb000000000000000000000001d4c00001d4c00000000000000000000000000000000000000000005a008600000000000000000000000000000000000000000000000000000000000000000002000000000002000000000000000000000000000000000000150e00001c84000000000000000013ce0064000000000000095f17700000016f1d19
         - Growatt new layout processing
                 - decrypt       :  True
                 - offset        :  6
                 - record layout :  T060103SPH
        - Grott server date/time used
         - Grott values retrieved:
                 -  pvserial             :  RTG0A46019
                 -  pvstatus             :  0
                 -  pvpowerin            :  13107.3
                 -  pv1voltage           :  0.0
                 -  pv1current           :  0.0
                 -  pv1watt              :  0.0
                 -  pv2voltage           :  0.0
                 -  pv2current           :  0.1
                 -  *outPutPower         :  80872811.0
                 -  pvpowerout           :  3604533.6
                 -  pvfrequentie         :  0.0
                 -  pvgridvoltage        :  0.0
                 -  pvgridcurrent        :  0.2
                 -  pvgridpower          :  53.6
                 -  pvgridvoltage2       :  55.0
                 -  pvgridcurrent2       :  2000.5
                 -  pvgridpower2         :  10485764.5
                 -  pvgridvoltage3       :  8.9
                 -  pvgridcurrent3       :  2.2
                 -  pvgridpower3         :  78643.6
                 -  totworktime          :  0.0
                 -  eactoday             :  6553.6
                 -  eactotal             :  0.0
                 -  epvtotal             :  9.0
                 -  epv1today            :  28.0
                 -  epv1total            :  0.0
                 -  epv2today            :  0.0
                 -  epv2total            :  0.0
                 -  pvtemperature        :  13.4
                 -  pvipmtemperature     :  0.0
                 -  pvboosttemp          :  0.0
                 -  bat_dsp              :  0.0
                 -  pbusvolt             :  0.0
                 -  #nbusvolt            :  0.0
                 -  #ipf                 :  0.0
                 -  #realoppercent       :  0.0
                 -  #opfullwatt          :  0.0
                 -  #deratingmode        :  0
                 -  eacharge_today       :  0.2
                 -  eacharge_total       :  0.0
                 -  batterytype          :  0
                 -  uwsysworkmode        :  0
                 -  systemfaultword0     :  0
                 -  systemfaultword1     :  0
                 -  systemfaultword2     :  5070
                 -  systemfaultword3     :  100
                 -  systemfaultword4     :  0
                 -  systemfaultword5     :  0
                 -  systemfaultword6     :  0
                 -  systemfaultword7     :  2399
                 -  pdischarge1          :  39321600.0
                 -  p1charge1            :  2405916.1

but of course ... now is like reverse enigeering needed or anyone which already started ?

camelcamro, 2022-12-04:
Hello Johan,
not sure, if i can reach you on this way ... trying it ...

My Question is, if you can help me to make a "record layout" which is having the right values ?
i would offer you to make a donation of 100,-. USD or EURO.

i would like to ask you, if you would help me with making a "record layout - 0Sunrino Hybrid Inverter"
Mine is a "Sacolar 12000W - Hybrid Dual Split Phase Inverter" (hot 1 + hot 2 = 240 Volt / hot1 + neutral = 120 Volt)
(typically for America power system)
with 2 MPPT - each 60 A - battery system 48V
=> SUNRINO SP serie
https://www.sacolar.com/products/info_8_itemid_187.html
https://www.sacolar.com/uploadfiles/2020/03/20200327151005105.pdf

Basically, it is a growatt product or most of it .
also the WIFI USB logger is a growatt product:
https://www.sacolar.com/products/info_8_itemid_140.html
https://www.sacolar.com/uploadfiles/2019/04/20190424115107517.pdf

basically, it is the same as the growatt WIFI USB dongle .... only configured for domain server.pvbutler.com
but also possible to reconfigure server.growatt.com

i guess, it is the small brother of them.
as they are using for this as default: pvbutler.com, which is also on growatt network.

###################################
anyhow .....

now, I'm using your grott server solution, but the "record layout" doesn'T fit fully - only few values are OK, and the length is also limited on 718.

so, i used a template: t060104sph.json as based and tried it with that ...

My Question is, if you can help me to make a "record layout" which is having the right values ?
i would offer you to make a donation of 100,-. USD or EURO.
[grott.zip](https://github.com/johanmeijer/grott/files/10149403/grott.zip)

if that is ok for you.
attached:
.)grott.ini -> config, to be sure, if i did anything wrong ...
.) growatt_server.log -> from your server -> the log with all data and raw data(XLS
.) RTG0A46019 storage data - 2022-12-04_2022-12-04.xls -> pvbutler (growatt) export in XLS
time range for both is: 13:00 to 13:30 - 2min cycle ( (the date is on my raspery server is having UTC, so, that is why different as I'm in Costa Rica (-6hrs) 6 hrs different as on the growatt webservice configured
.) t060103sph.json -> record layout what i was beginning todo ..

maybe you can help ?

camelcamro, 2022-12-05:
to keep you up2date ...

meanwhile i was beginning with reverse engineering, and made a template , which is:
1.) figure out the length of the data (in my case on this inverter = 718
2.) every field define in a "record layout" file -> all fields as default in steps of 2 byte (4 lenght in total), and divide by 10, and then looking on the real values to compare on server.pvbutler / server.growatt.com

attached the draft version v0.1 for this kind of inverter, few values, i need to check when the PV system getting sun ;-)
[t060104sph.zip](https://github.com/johanmeijer/grott/files/10149967/t060104sph.zip)

johanmeijer, 2022-12-05:
With the excel and the original data (the "t060104") record from the same time is should be relatively easy to create te right layout. I suppose you want to use the names of the fields as specified in the excel? 

Can yo u also sent the log from the start of a connection till the first "04" record so we can see if the used protocol is also the same? 

This week I am busy with other things than Grott. But next week I can look at it I think.

camelcamro, 2022-12-05:
Hello Johann,

i already did the reserve engineering ... almost done for the new record layout

about the offset ?
do you mean the offset in grottconf.py or only to set the grott.ini
this values ?

# Compat is True and valoffset needs to be set if offset / growatt protocol has been changed..
compat = True
valueoffset = 4

and then restarting and output psoting ?

johanmeijer, 2022-12-05:
If you use the layout files do not specify compat and valueoffset. That is only being used with the first version of Grott (is still there for compatibility reasons).

johanmeijer, 2022-12-05:
Maybe a stupid question but if the inverter is alomost the same as a growatt sph is then the standard  sph layout not giving the right info?

camelcamro, 2022-12-05:
but you were asking, if i can do a "offset 4" start ?
so, hoiw i should configure it to prepare the debug ttrace for you ?

removing my template (which i copied to root grott folder) ?
verbose = True
trace = True
offset ? changing in grottconfig.py ?
or what do you need ?

template data for now - almost fine ...

t060104sph.json
```
{"T060104SPH": {
    "decrypt"           : {"value" :"True"},
    "pvserial"          : {"value" :76, "length" : 10, "type" : "text", "divide" : 10},
    "date"              : {"value" :136, "divide" : 10},
    "recortype"         : {"value" :154, "length" : 2, "type" : "num"},
    "pvstatus"          : {"value" :158, "length" : 2, "type" : "num"},
    "vPV1"              : {"value" :162, "length" : 2, "type" : "num", "divide" : 10},
    "vPV2"              : {"value" :166, "length" : 2, "type" : "num", "divide" : 10},
    "data_170"          : {"value" :170, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "PPV"               : {"value" :174, "length" : 2, "type" : "num", "divide" : 10},
    "data_178"          : {"value" :178, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "data_182"          : {"value" :182, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "iChargePV1"        : {"value" :186, "length" : 2, "type" : "num", "divide" : 10},
    "data_190"          : {"value" :190, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "data_194"          : {"value" :194, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "outPutPower"       : {"value" :198, "length" : 2, "type" : "num", "divide" : 10},
    "data_202"          : {"value" :202, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "outPutPowerTotal"  : {"value" :206, "length" : 2, "type" : "num", "divide" : 10},
    "data_210"          : {"value" :210, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "pacGrid"           : {"value" :214, "length" : 2, "type" : "num", "divide" : 10},
    "data_218"          : {"value" :218, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "pacGridAll"        : {"value" :222, "length" : 2, "type" : "num", "divide" : 10},
    "vBat"              : {"value" :226, "length" : 2, "type" : "num", "divide" : 100},
    "capacity"          : {"value" :230, "length" : 2, "type" : "num", "divide" : 1},
    "data_234"          : {"value" :234, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "vGrid"             : {"value" :238, "length" : 2, "type" : "num", "divide" : 10},
    "freqGrid"          : {"value" :242, "length" : 2, "type" : "num", "divide" : 100},
    "outPutVolt"        : {"value" :246, "length" : 2, "type" : "num", "divide" : 10},
    "freqOutPut"        : {"value" :250, "length" : 2, "type" : "num", "divide" : 100},
    "data_254"          : {"value" :254, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "Temperature1"      : {"value" :258, "length" : 2, "type" : "num", "divide" : 10},
    "Temperature2"      : {"value" :262, "length" : 2, "type" : "num", "divide" : 10},
    "LoadPercent"       : {"value" :266, "length" : 2, "type" : "num", "divide" : 10},
    "W_data_270"        : {"value" :270, "length" : 2, "type" : "num", "divide" : 10},
    "data_274"          : {"value" :274, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "data_278"          : {"value" :278, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "data_282"          : {"value" :282, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "data_286"          : {"value" :286, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "data_290"          : {"value" :290, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "outPutCurrent1"    : {"value" :294, "length" : 2, "type" : "num", "divide" : 10},
    "outPutCurrent2"    : {"value" :298, "length" : 2, "type" : "num", "divide" : 10},
    "data_302"          : {"value" :302, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "pacGrid"           : {"value" :306, "length" : 2, "type" : "num", "divide" : 10},
    "data_310"          : {"value" :310, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "data_314"          : {"value" :314, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "data_318"          : {"value" :318, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "data_322"          : {"value" :322, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "data_326"          : {"value" :326, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "data_330"          : {"value" :330, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "W_data_334"        : {"value" :334, "length" : 2, "type" : "num", "divide" : 10},
    "EtoGridToday"      : {"value" :338, "length" : 2, "type" : "num", "divide" : 10},
    "W_data_342"        : {"value" :342, "length" : 2, "type" : "num", "divide" : 10},
    "data_346"          : {"value" :346, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "data_350"          : {"value" :350, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "data_354"          : {"value" :354, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "data_358"          : {"value" :358, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
[... cut ...]
    "data_698"          : {"value" :698, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "data_702"          : {"value" :702, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "data_706"          : {"value" :706, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "data_710"          : {"value" :710, "length" : 2, "type" : "num", "divide" : 10, "incl" : "no"},
    "data_714"          : {"value" :714, "length" : 2, "type" : "num", "divide" : 10}
} }

```

camelcamro, 2022-12-05:
> Maybe a stupid question but if the inverter is alomost the same as a growatt sph is then the standard sph layout not giving the right info?

no,
1.) only 718 bytes, and failing if i use the default 
2.) huge difference .... don'T ask me, why it is showing the "T060104SPH" 
it is also finny, on first start it is looking for a "T060103SPH". Then after next update (in my case 1x), then it is always using the "T060104SPH"
maybe the "offset" test with 4, is giving us better infos. you tell me ;-)

johanmeijer, 2022-12-05:
The 03 is an announce record only sent once after session start. The 04 is te real data record that is sent at regulair basis and contains the real information.

Grott does decode the 03 record (is not needed).

camelcamro, 2022-12-05:
FYI: with the give template above, i alreadyy getting more values which i was possible to define, as it is by the export from growatt ;-)
few information, still are not clear for me, but can be something internal things, which they hidding from the users ...
but, the template is really great, and having everything what others are having.

johanmeijer, 2022-12-06:
I think if you use invtype = spf (and ignore op_va because I detected a bug there) I might fit your record layout pretty close.

johanmeijer, 2022-12-07:
I really mean spf (not sph)you if you run grott with invtype = spf you might see some usable values. 
I tested with your data amd think it will fit.

camelcamro, 2023-01-24:
well, i can only tell you,
i tested also "spf", but that is having only few values, and not using 716 bytes. i guess, spf is having only 378 or something what i have seen, and then there would be several values missing. so, that would be only a workaround. and also not all given vlaues are then also good named.