johanmeijer/grott #249: grotcsv - Grott extension processing started / Grott import extension failed: grotcsv

camelcamro, 2022-12-05:
Hello any idea what is here wrong ?

the normal output is fine, and also the mqtt out (which is disabled)

but not file created ...
1.) directory is existing and having rights for 777
2.) config in grott.ini =
[extension]
extension = True
extname = grotcsv
extvar = {"outpath": "/home/custom/grott/grottlog", "csvheader":"decrypt,pvserial,date,recortype,pvstatus,data_162,data_166,data_170,PPV,data_178,data_182,iChargePV1,data_190,data_194,outPutPower,data_202,data_206,data_210,data_214,data_218,data_222,vBat,capacity,data_234,vGrid,freqGrid,outPutVolt,freqOutPut,data_254,Temperature1,Temperature2,LoadPercent,data_270,data_274,data_278,data_282,data_286,data_290,outPutCurrent1,outPutCurrent2,data_302,pacGrid,data_310,data_314,data_318,data_322,data_326,data_330,data_334,EtoGridToday,data_342,data_346,data_350,data_354,data_358,eChargeToday,data_366,eChargeTotal,data_374,data_378,data_382,data_386,data_390,eacChargeToday,data_398,eacChargeTotal,data_406,eBatDisChargeToday,data_414,eBatDisChargeTotal,data_422,data_426,data_430,data_434,data_438,data_442,data_446,data_450,data_454,data_458,data_462,data_466,data_470,data_474,pBat,data_482,data_486,data_490,data_494,data_498,data_502,data_506,eopDischrToday,data_514,eopDischrTotal,data_522,data_526,data_530,data_534,data_538,data_542,data_546,data_550,data_554,data_558,data_562,data_566,data_570,epvTotal,epvToday,data_582,data_586,data_590,data_594,data_598,data_602,data_606,data_610,data_614,data_618,data_622,data_626,data_630,data_634,data_638,data_642,data_646,data_650,data_654,data_658,data_662,data_666,data_670,data_674,data_678,data_682,data_686,data_690,data_694,data_698,data_702,data_706,data_710,data_714"}

in code i see: in grottdata.py 
    if conf.extension : 
        
        if conf.verbose :  print("\t - " + "Grott extension processing started : ", conf.extname)
        import importlib
        try: 
            module = importlib.import_module(conf.extname, package=None)
        except :
            if conf.verbose : print("\t - " + "Grott import extension failed:", conf.extname)  
            return

maybe a package needs to be installed ?
eg:

pip3 install importlib

but still giving me the same error as described above ...

johanmeijer, 2022-12-05:
sudo pip3 install importlib ?

johanmeijer, 2022-12-18:
Was this the solution?

camelcamro, 2022-12-20:
yes, sorry, i guess - this ticket can be closed, because i have seen it somewhere, that it was already described, that this is also needed to be installed.

closed 2022-12-24 by johanmeijer