johanmeijer/grott #260: Pvoutput.org. Bad request 400: Energy value [76500] too high for system size [5200]**

Raspb-henk, 2022-12-23:
I have restarted everything which took some time. I checked that the IP in the Shine has been set to my RPI (192.168.1.103) . I use a RPI 4 with a fresh installed OS and docker with Portainer.
I installed Grottrpi and it is running. I checked that the IP in the Shine has been set to my RPI (192.168.1.103). In the log off Grottrpi I see that the data from the Growatt is received. On the website of Growatt I also receive the data OK and the SHINE APP is also OK. So far so good.

BUT i do not get data to PV output.
As it is a three fase inverter I added the command < -e ginvtype ="tl3" >
I started the container with this command:
docker run -d -p 5279:5279 --restart on-failure -e gnomqtt="True" -e gblockcmd="True" -e ginvtype="tl3" -e gpvoutput="True" -e gpvapikey="ff78c5c5e43487XXXXXXXXc315d33c61b0b11" -e gpvsystemid="95xxx6" ledidobe/grottrpi

In the log of Grott (attatched) I find this:

	 - Grott send data to PVOutput systemid:  95XX6 for inverter:  XJL0XXX01F
		 -  {'X-Pvoutput-Apikey': 'ff78c5c5e4348XXXXXXXXc61b0b11', 'X-Pvoutput-SystemId': '95XX6'}
		 -  {'d': '20221211', 't': '13:26', 'v1': 76500, 'v2': 39321.6, 'v6': 394.0}
	 - Grott PVOutput response: 
		 **-  Bad request 400: Energy value [76500] too high for system size [5200]**
XXXXX added by me
and I see very strange values (?) for the pvgridpower. ( pvgridpower3 : 194445312.0) (the values change but stay to high)

In pvoutput the inverter is a Growatt 5000TL3-XH. The ID is correct and is does have a size of 5200W. (13 x 400WP pannels)

So I Now I am quessing what is wrong ?? From where this high value(s)? I think Grott is working OK but is maybe something wrong with the inverter setings?
'NB when the logfile was made it was dark .Update: At twelve midday with bright sun the same high values and bad request error.
NB 2. Tried starting docker with -e ginvtype="sph"
NB 3. Tried starting docker with both -e ginvtype="tl3"and -e ginvtype="sph"
All same result: bad request error 400 Can it be that the problem is that I have a three fase 3tl -XH inverter?

I hope you can point me in the right direction.
Henk
[_elastic_noyce_logs.txt](https://github.com/johanmeijer/grott/files/10222626/_elastic_noyce_logs.txt)

johanmeijer, 2022-12-24:
This is a strange one. You are the first one with this type of pvoutput problems. I have to look deeper in to this.

Raspb-henk, 2022-12-24:
Hi
Not a situation I like It is a new type of hybrid  inverter. So i do not now whether some other user has it up and running. 
To check the settings. Is there an  inverter setting you suggest? If you need more logs or so tell me and I send them.
Happy Christmas,
Henk

Raspb-henk, 2022-12-26:
Hi
If I do not use docker but install as "first time installation" , using gitt clone with no -e ginvtype added I do not get the error from pvoutput. Everything works. 
What is the difference with the dockerfile.? I d like to use docker as if is better to manage.

Raspb-henk, 2023-01-28:
Hi,
I hope this gets an Respons ;).
I have checked the logs  and found that the error mentioned above is a wrong decimal pointini the output to PVoutput.org. 
The data send by the inverter is: 
               -  pvenergytoday        :  185.7
	      -  pvenergytotal          :  2.6
but the data send to PVoutput is:
                 - Grott send data to PVOutput systemid:  9xxx6 for inverter:  XJL0CE901F
		 -  {'X-Pvoutput-Apikey': 'ff78c5c5e434xxxxxxxxxx61b0b11', 'X-Pvoutput-SystemId': '95xxx6'}
		 -  {'d': '20230128', 't': '14:05', 'v1': 185700, 'v2': 124518.4, 'v6': 395.1}
	 - Grott PVOutput response: 
		 -  Bad request 400: Energy value [185700] too high for system size [5200]
 
V1: is 185700 which is to high I think . As mentioned befor this occurs only if I try to run grott in docker.

I hope it is possible to correct this issue?

kind regards
Henk.

johanmeijer, 2023-01-30:
Hi Henk.  As said your the only one reporting this problem so I really have to look deeper into that. Changing the energy value is not the right thing I think because that might affect the others. 

I think pvenergy today 185.7 is high and pvenergy total 2.7 low. So I think is is more in the decoding of the data then the send values to pvoutput. 

To be honest if I see your decoding values in the _elastic_noyce_logs.txt it does not seems to make any sense to me at all.

Do you have a log with also the grott startup values and a data processing value from the part it says growatt original data till the pvoutput reply?

johanmeijer, 2023-01-30:
Can also try it with the ledidobe/grottrpi:beta docker image?

johanmeijer, 2023-02-05:
@Raspb-henk Have you already tried the Beta image? It is now the latest version! 

If you want me to debug the problem, I need the log as state above.

Raspb-henk, 2023-02-05:
HiSorry I have not been able to work on this for a short time. I am now home again.I have installed Grott rpi: bèta in docker. This now works ok. Verzonden vanaf mijn Galaxy

bwbboot, 2023-02-12:
Hello, i am running in to this same issue:

-Grott Growatt logging monitor : 2.7.8
-Growatt MOD 4000TL3-X

	 - Grott data record date/time processing started
	 - date-time:  2023-02-12T10:39:16
	 - Grott values retrieved:
		 -  datalogserial        :  XGD....
		 -  pvserial             :  DJG....
		 -  pvstatus             :  1
		 -  pvpowerin            :  231.6
		 -  pv1voltage           :  396.2
		 -  pv1current           :  0.5
		 -  pv1watt              :  198.1
		 -  pv2voltage           :  181.5
		 -  pv2current           :  0.0
		 -  pv2watt              :  0.0
		 -  pvpowerout           :  0.0
		 -  pvfrequentie         :  0.0
		 -  pvgridvoltage        :  0.0
		 -  pvgridcurrent        :  0.0
		 -  pvgridpower          :  0.0
		 -  pvgridvoltage2       :  0.0
		 -  pvgridcurrent2       :  0.0
		 -  pvgridpower2         :  0.0
		 -  pvgridvoltage3       :  0.0
		 -  pvgridcurrent3       :  0.0
		 -  pvgridpower3         :  14857511.4
		 -  totworktime          :  54.6
		 -  pvenergytoday        :  9267198.4
		 -  pvenergytotal        :  26660455.2
		 -  epvtotal             :  0.1
		 -  epv1today            :  8644.0
		 -  epv1total            :  5460423.3
		 -  epv2today            :  0.1
		 -  epv2total            :  8354.6
		 -  pvtemperature        :  46.3
		 -  pvipmtemperature     :  0.0
	 - MQTT jsonmsg: 
			 {"device": "DJG....", "time": "2023-02-12T10:39:16", "buffered": "no",
			 "values": {"datalogserial": "XGD..., "pvserial": "DJG...",
			 "pvstatus": 1, "pvpowerin": 2316, "pv1voltage": 3962, "pv1current": 5,
			 "pv1watt": 1981, "pv2voltage": 1815, "pv2current": 0, "pv2watt": 0,
			 "pvpowerout": 0, "pvfrequentie": 0, "pvgridvoltage": 0, "pvgridcurrent": 0,
			 "pvgridpower": 0, "pvgridvoltage2": 0, "pvgridcurrent2": 0, "pvgridpower2":
			 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3": 148575114,
			 "totworktime": 393216, "pvenergytoday": 92671984, "pvenergytotal":
			 266604552, "epvtotal": 1, "epv1today": 86440, "epv1total": 54604233,
			 "epv2today": 1, "epv2total": 83546, "pvtemperature": 463,
			 "pvipmtemperature": 0}}
	 - Grott MQTT topic used : energy/growatt
	 - MQTT message message sent
DJG0...
	 - Grott send data to PVOutput systemid:  xxxx for inverter:  DJG...
		 -  {'X-Pvoutput-Apikey': '.xxxxxxx', 'X-Pvoutput-SystemId': 'xxxxx'}
		 -  {'d': '20230212', 't': '10:39', 'v2': 0.0, 'v6': 0.0, 'v1': 9267198400}
	 - Grott PVOutput response: 
		 -  Bad request 400: Energy value [9267198400] too high for system size [4560]
	 - Grott Send data to Influx disabled 
	 - Grott extension processing started :  grott_ha
	Grott HA 0.0.7-rc2 - creating DJG... config in HA
	 - Grott extension processing ended :  0

Raspb-henk, 2023-02-12:
Hi,
I have this solved with running the last beta in docker.
I start it with:
`docker run -d -p 5279:5279 --restart on-failure -e gmqttip="192.168.1.103"  -e gpvoutput="True" -e gpvapikey="ff78c5c5e43xxxxxxxxxx6c315d33c61b0b11" -e gpvsystemid="9xxxxx" ledidobe/grottrpi:beta
`
It runs OK .

Henk

johanmeijer, 2023-02-12:
I still exist after the running the new docker beta. Please sent me the original grott data from the log so I can debug it.

bwbboot, 2023-02-12:
> I still exist after the running the new docker beta. Please sent me the original grott data from the log so I can debug it.

If my logfile (attached above) does not include the correct details. Can you give instructions please?

johanmeijer, 2023-02-12:
Sorry missed the file. Info is ok. I am going to look at it!

bwbboot, 2023-02-12:
Just to add some more detail. I have 3 inverters processed with Grott. The 2x one-phase inverters upload the information to PVOUTPUT without any issues. Seems the 3-phase inverter is causing an issue.

johanmeijer, 2023-02-13:
@bwbboot for the tl3-x server you have to specify invtype=tl3 in the grott.ini (or docker run with -e ginvtype="tl3" 

Do you run a grott instance per inverter or do you have one grott for all inverters ? 

If you have 1 grott you have to specify more to make it possible for Grott (and PVOutput) to see the difference is systems.

johanmeijer, 2023-02-13:
For multi inverter support see: 
https://github.com/johanmeijer/grott/wiki/Multi-Inverter-support 
https://github.com/johanmeijer/grott/wiki/PVOutput.org-support

bwbboot, 2023-02-13:
Thank you, and apologies for not spotting this myself. So my config should look as follows, assuming that the two other inverters (TL-X) can use the default setting. Correct?

invtypemap = {'DJGXXXX': 'tl3', 'CMGXXXX': 'default', 'PYHXXXX': 'default'}

johanmeijer, 2023-02-13:
Yes that should do it.

Be aware the multi inverter page I created today so it not strange you haven't seen it before :)

bwbboot, 2023-02-14:
Thanks Johan. I can confirm that everything works perfect now for me.

johanmeijer, 2023-02-14:
@Raspb-henk @bwbboot can we close this issue?

Raspb-henk, 2023-02-14:
Yes.The bèta runs ok. So my problem is als solved. I hope it stays so  Thanks for the support.Henk.Verzonden vanaf mijn Galaxy

bwbboot, 2023-02-14:
> @Raspb-henk @bwbboot can we close this issue?

Yes

closed 2023-02-14 by johanmeijer