johanmeijer/grott #261: Service will not start.

Banabas, 2022-12-27:
Hello.
I have a problem starting the service.
I have installed on the Raspberry /home/pi/grott/.
It also runs when I start manually. python3 grott.py -v
I get data and the data is written to influx.

But the service just does not want to start.
Can someone help me with this?

Super would be in german because my english is really bad.

`pi@solaranzeige:~/grott $ sudo systemctl start grott
pi@solaranzeige:~/grott $ sudo service grott status
● grott.service - Grott Growatt monitor
   Loaded: loaded (/etc/systemd/system/grott.service; enabled; vendor preset: enabled)
   Active: failed (Result: start-limit-hit) since Tue 2022-12-27 12:02:04 CET; 2s ago
  Process: 18510 ExecStart=/usr/bin/python3 -u /home/pi/grott/grott.py -v (code=exited, status=1/FAILURE)
 Main PID: 18510 (code=exited, status=1/FAILURE)

Dez 27 12:02:04 solaranzeige systemd[1]: grott.service: Service RestartSec=1s expired, scheduling restart.
Dez 27 12:02:04 solaranzeige systemd[1]: grott.service: Scheduled restart job, restart counter is at 5.
Dez 27 12:02:04 solaranzeige systemd[1]: Stopped Grott Growatt monitor.
Dez 27 12:02:04 solaranzeige systemd[1]: grott.service: Start request repeated too quickly.
Dez 27 12:02:04 solaranzeige systemd[1]: grott.service: Failed with result 'start-limit-hit'.
Dez 27 12:02:04 solaranzeige systemd[1]: Failed to start Grott Growatt monitor.
`

Greets
Sven

johanmeijer, 2022-12-27:
Hallo Sven, wie seht deine grott.service aus? 

Sehen sie vielecht etwas mehr when Du journalctl -u grott -e eingibst? Was ist die werkliche fehler (nicht die start-limit).

Banabas, 2022-12-27:
Hallo.
Die grott.service habe ich nicht verändert.
Habe sie nur nach /etc/systemd/system kopiert und danach dann "sudo systemctl enable grott"

pi@solaranzeige:~ $ journalctl -u grott -e
Dez 27 17:07:58 solaranzeige grott[25901]:   File "/home/pi/grott/grottdata.py", line 17, in <module>
Dez 27 17:07:58 solaranzeige grott[25901]:     import paho.mqtt.publish as publish
Dez 27 17:07:58 solaranzeige grott[25901]: ModuleNotFoundError: No module named 'paho'
Dez 27 17:07:58 solaranzeige systemd[1]: grott.service: Succeeded.
Dez 27 17:07:59 solaranzeige systemd[1]: grott.service: Service RestartSec=1s expired, scheduling restart.
Dez 27 17:07:59 solaranzeige systemd[1]: grott.service: Scheduled restart job, restart counter is at 4.
Dez 27 17:07:59 solaranzeige systemd[1]: Stopped Grott Growatt monitor.
Dez 27 17:07:59 solaranzeige systemd[1]: Started Grott Growatt monitor.
Dez 27 17:07:59 solaranzeige grott[25903]: Traceback (most recent call last):
Dez 27 17:07:59 solaranzeige grott[25903]:   File "/home/pi/grott/grott.py", line 19, in <module>
Dez 27 17:07:59 solaranzeige grott[25903]:     from grottconf import Conf
Dez 27 17:07:59 solaranzeige grott[25903]:   File "/home/pi/grott/grottconf.py", line 9, in <module>
Dez 27 17:07:59 solaranzeige grott[25903]:     from grottdata import format_multi_line, str2bool
Dez 27 17:07:59 solaranzeige grott[25903]:   File "/home/pi/grott/grottdata.py", line 17, in <module>
Dez 27 17:07:59 solaranzeige grott[25903]:     import paho.mqtt.publish as publish
Dez 27 17:07:59 solaranzeige grott[25903]: ModuleNotFoundError: No module named 'paho'
Dez 27 17:07:59 solaranzeige systemd[1]: grott.service: Succeeded.
Dez 27 17:08:00 solaranzeige systemd[1]: grott.service: Service RestartSec=1s expired, scheduling restart.
Dez 27 17:08:00 solaranzeige systemd[1]: grott.service: Scheduled restart job, restart counter is at 5.
Dez 27 17:08:00 solaranzeige systemd[1]: Stopped Grott Growatt monitor.
Dez 27 17:08:00 solaranzeige systemd[1]: grott.service: Start request repeated too quickly.
Dez 27 17:08:00 solaranzeige systemd[1]: grott.service: Failed with result 'start-limit-hit'.
Dez 27 17:08:00 solaranzeige systemd[1]: Failed to start Grott Growatt monitor.

Banabas, 2022-12-27:
now it's on.
was a mistake on my part.
have only "sudo pip install paho-mqtt" not pip3.
with "sudo pip3 install paho-mqtt" as it says in the instructions it works. Sorry my mistake.

johanmeijer, 2022-12-27:
Super!

closed 2022-12-27 by johanmeijer