johanmeijer/grott #262: Growatt SPA not reading

mvdh02, 2022-12-27:
Hello readers,
i've got a Growatt SPA3000 Battery system. I try to get it working with grott. But i don't know the right setting. Grott is running. And the Inverter is communication trough Grott. But in NodeRed my MQTT nog messages are recieved. 

In the record it reads there is no layout for the SPAcd..

Dec 26 23:48:18 raspberrypi grott[453]:          - Grott automatic protocol detection
Dec 26 23:48:18 raspberrypi grott[453]:          - Grott data record length 839
Dec 26 23:48:18 raspberrypi grott[453]:          - layout   :  T060104XSPA
Dec 26 23:48:18 raspberrypi grott[453]:          - no matching record layout found, try generic
Dec 26 23:48:18 raspberrypi grott[453]:          - no matching record layout found, standard processing performed
Dec 26 23:48:18 raspberrypi grott[453]:          - Record layout used :  none
Dec 26 23:48:18 raspberrypi grott[453]:          - Growatt data decrypted V2
Dec 26 23:48:18 raspberrypi grott[453]:          - Grott Growatt data decrypted
Dec 26 23:48:18 raspberrypi grott[453]:          - Growatt plain data:
Dec 26 23:48:18 raspberrypi grott[453]:                  002d0006033f01044a50433641313539413900000000000000000000000000000000000000005
Dec 26 23:48:18 raspberrypi grott[453]:                  a474441424b533030370000000000000000000000000000000000000000160c1a1730100303e8
Dec 26 23:48:18 raspberrypi grott[453]:                  0464000600000000000000000000000000000800000000000000000002110012000007d000000
Dec 26 23:48:18 raspberrypi grott[453]:                  00000000000000007d000000000000000000000000000000000000007d0000000000000000000
Dec 26 23:48:18 raspberrypi grott[453]:                  0007d00000005000060bd51f400000008f0000277a00000000000005cb000000080000047c000
Dec 26 23:48:18 raspberrypi grott[453]:                  00008000004bf000000a100002a04000000000000000000000000000000000000000000000000
Dec 26 23:48:18 raspberrypi grott[453]:                  0000000000000000000003e80000006100000000001214b000000007177000000000000000000
Dec 26 23:48:18 raspberrypi grott[453]:                  00000641644000000000000000000000000000059505950000000000000000100000000000000
Dec 26 23:48:18 raspberrypi grott[453]:                  000000000000000000000000000000000000000000046504e1000d0000061a000000e303b9000
Dec 26 23:48:18 raspberrypi grott[453]:                  0000000000000000000000000001200000b3d00000011000008ee000000000000000000000000
Dec 26 23:48:18 raspberrypi grott[453]:                  00000000000000000000000000000000000000000000000000000000000000000000000000000
Dec 26 23:48:18 raspberrypi grott[453]:                  00000000000000000000000000000000000000000100000000000100000000000000000000000
Dec 26 23:48:18 raspberrypi grott[453]:                  00000000000000000000000000000000000000000000000000000000000000000000000000000
Dec 26 23:48:18 raspberrypi grott[453]:                  0000000000007d0084c0006000000000000000000000000000000000000000000000000000000
Dec 26 23:48:18 raspberrypi grott[453]:                  00000000000000000000000000000000000000000000000000000000000000000000000000000
Dec 26 23:48:18 raspberrypi grott[453]:                  00000ffffff44138b09250005000000b600000000000000000000000000000000000000000000
Dec 26 23:48:18 raspberrypi grott[453]:                  00000088000015b900d1d4b200000000000000000000000000000000000000000000000000000
Dec 26 23:48:18 raspberrypi grott[453]:                  00000000000000000000000000000000000000000000000000000000000000000000000000000
Dec 26 23:48:18 raspberrypi grott[453]:                  0000000099008200c8000002140ede0ee70000000000000000000000120000089800000012000
Dec 26 23:48:18 raspberrypi grott[453]:                  00b3d0000000d0000061a000000bc0000000100000000000000000000367a
Dec 26 23:48:18 raspberrypi grott[453]:          - Grott data ack record or data record not defined no processing done
Dec 26 23:48:18 raspberrypi grott[453]:          - Growatt packet received:
Dec 26 23:48:18 raspberrypi grott[453]:                   <socket.socket fd=7, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('
Dec 26 23:48:18 raspberrypi grott[453]:          - Data less then minimum record length, data not processed
Dec 26 23:49:58 raspberrypi grott[453]:          - Growatt packet received:
Dec 26 23:49:58 raspberrypi grott[453]:                   <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('
Dec 26 23:49:58 raspberrypi grott[453]:          - Data less then minimum record length, data not processed
Dec 26 23:49:58 raspberrypi grott[453]:          - Growatt packet received:
Dec 26 23:49:58 raspberrypi grott[453]:                   <socket.socket fd=7, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('
Dec 26 23:49:58 raspberrypi grott[453]:          - Data less then minimum record length, data not processed
Dec 26 23:52:58 raspberrypi grott[453]:          - Growatt packet received:
Dec 26 23:52:58 raspberrypi grott[453]:                   <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('
Dec 26 23:52:58 raspberrypi grott[453]:          - Data less then minimum record length, data not processed
Dec 26 23:52:58 raspberrypi grott[453]:          - Growatt packet received:
Dec 26 23:52:58 raspberrypi grott[453]:                   <socket.socket fd=7, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('
Dec 26 23:52:58 raspberrypi grott[453]:          - Data less then minimum record length, data not processed
Dec 26 23:53:18 raspberrypi grott[453]:          - Growatt packet received:
Dec 26 23:53:18 raspberrypi grott[453]:                   <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('
Dec 26 23:53:18 raspberrypi grott[453]:          - Growatt original Data:
Dec 26 23:53:18 raspberrypi grott[453]:                  \x00\x2e\x00\x06\x03\x3f\x01\x04\x0d\x22\x2c\x41\x20\x45\x41\x7e\x33\x56\x77
Dec 26 23:53:18 raspberrypi grott[453]:                  \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
Dec 26 23:53:18 raspberrypi grott[453]:                  \x35\x30\x25\x35\x36\x0c\x21\x5f\x47\x56\x74\x74\x47\x72\x6f\x77\x61\x74\x74
Dec 26 23:53:18 raspberrypi grott[453]:                  \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x77\x78\x6e\x50\x47\x7e\x74\x62
Dec 26 23:53:18 raspberrypi grott[453]:                  \x9c\x70\x23\x72\x69\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
Dec 26 23:53:18 raspberrypi grott[453]:                  \x7f\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x76\x56\x72\x7d\x77\x61\x37\x8c\x47
Dec 26 23:53:18 raspberrypi grott[453]:                  \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x34\x99\x74\x74\x47\x72\x6f\x77\x61\x74
Dec 26 23:53:18 raspberrypi grott[453]:                  \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x2c\x8f\x61\x74\x74\x47\x72\x6f\x77
Dec 26 23:53:18 raspberrypi grott[453]:                  \x61\x74\x74\x04\x8a\x6f\x77\x61\x24\x74\x41\x79\xba\x68\x21\x74\x74\x47\xe2
Dec 26 23:53:18 raspberrypi grott[453]:                  \x6f\x77\x46\x0f\x74\x47\x72\x6f\x77\x61\x71\xbf\x47\x72\x6f\x7f\x61\x74\x70
Dec 26 23:53:18 raspberrypi grott[453]:                  \x3b\x72\x6f\x77\x69\x74\x74\x43\xcd\x6f\x77\x61\xd6\x74\x47\x58\x6a\x77\x61
Dec 26 23:53:18 raspberrypi grott[453]:                  \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
Dec 26 23:53:18 raspberrypi grott[453]:                  \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6c\x9f\x61\x74\x74\x26
Dec 26 23:53:18 raspberrypi grott[453]:                  \x72\x6f\x77\x61\x74\x66\x53\xc2\x6f\x77\x61\x73\x63\x37\x72\x6f\x77\x61\x74
Dec 26 23:53:18 raspberrypi grott[453]:                  \x74\x47\x72\x6f\x77\x61\x10\x62\x03\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
Dec 26 23:53:18 raspberrypi grott[453]:                  \x61\x74\x74\x47\x2b\x3f\x2e\x31\x74\x74\x47\x72\x6f\x77\x61\x75\x74\x47\x72
Dec 26 23:53:18 raspberrypi grott[453]:                  \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
[... 3 lines cut ...]
Dec 26 23:53:18 raspberrypi grott[453]:                  \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
Dec 26 23:53:18 raspberrypi grott[453]:                  \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
[... cut ...]
Dec 26 23:53:18 raspberrypi grott[453]:                  \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\xee\x61\xf6\x74\x8f
Dec 26 23:53:18 raspberrypi grott[453]:                  \x72\x6f\x75\x75\x7a\x90\x49\x9f\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
Dec 26 23:53:18 raspberrypi grott[453]:                  \x66\x47\x72\x67\xef\x61\x74\x74\x55\x72\x6f\x7c\x5c\x74\x74\x47\x7f\x6f\x77
Dec 26 23:53:18 raspberrypi grott[453]:                  \x67\x6e\x74\x47\x73\x66\x77\x61\x74\x75\x47\x72\x6f\x77\x61\x74\x74\x47\x72
Dec 26 23:53:18 raspberrypi grott[453]:                  \x6f\xd3\x64

johanmeijer, 2022-12-27:
There is no layout for SPA yet. Sp please select invtype=spf. That should work.

johanmeijer, 2023-01-02:
Sorry I made a mistake. It should more like a SPH inverter (invtype = SPH). But I see there some strange values as well (ans Storage info is missing). 

I have to dive deeper in it.

mvdh02, 2023-01-03:
@johanmeijer That would be nice. I have tried several options, but none of them worked. Thanks in advance

johanmeijer, 2023-02-12:
@mvdh02 

I added this first SPA support in the Beta version (2.8.1) 
If you can test it, it would be nice. Please add invtype=spa to grott.ini. 

I have create a layout based on growatt documentation and test it with your record above. I am not sure if this make any sense at all. 

Be aware output to PVOutput is not supported (yet?).

mvdh02, 2023-02-13:
@johanmeijer 
Thanks for you're reaction and action. I will certanly try the beta version. Where can i find this version?

johanmeijer, 2023-02-13:
https://github.com/johanmeijer/grott/tree/Beta-(2.8.x) or do you use docker?

mvdh02, 2023-02-17:
@johanmeijer There where some problems with the Growatt server, so it was difficult to change the IP to Grott. I've changed it today, and the values of the SPA3000 are vissable in Node-red / MQTT

The values i can see are:
{"device": "xxxxxxxxxx", xxxxxxx", "time": "2023-02-17T15:41:17", "buffered": "no", "values": {"datalogserial": "xxxxx", "pvserial": "xxxxxxxx", "pvstatus": 6, "uwsysworkmode": 3, "systemfaultword0": 0, "systemfaultword1": 0, "systemfaultword2": 0, "systemfaultword3": 0, "systemfaultword4": 0, "systemfaultword5": 0, "systemfaultword6": 0, "systemfaultword7": 2048, "pdischarge1": 0, "pcharge1": 0, "vbat": 519, "SOC": 20, "pactouserr": 600, "pactousers": 0, "pactousert": 0, "pactousertot": 600, "pactogridr": 0, "pactogrids ": 0, "pactogrid t": 0, "pactogridtot": 0, "plocaloadr": 3100, "plocaloads": 0, "plocaloadt": 0, "plocaloadtot": 3100, "ipm": 0, "battemp ": 100, "spdspstatus": 6, "spbusvolt": 45374, "etouser_tod": 70, "etouser_tot": 17985, "etogrid_tod": 0, "etogrid_tot": 1663, "edischarge1_tod": 0, "edischarge1_tot": 1809, "eharge1_tod": 1, "eharge1_tot": 1812, "elocalload_tod": 84, "elocalload_tot": 20118, "dwexportlimitap": 0, "epsfac": 0, "epsvac1": 0, "epsiac1": 0, "epspac1": 0, "e...

I have to figure out what value what means. But the basics are working. Thanks a lot

johanmeijer, 2023-02-19:
Good luck with it and please let me know if you think something is wrong with the mapping!

xhemp, 2023-03-03:
I started testing Grott with a SPA3000 and so far, so good, working as intended.
Can't use it with pvoutput but I read here that it's not working (yet?).
Great job on this tool, thank you.

johanmeijer, 2023-03-14:
I will add PVOUTPUT soon for the SPA record if everything is working ok for you. 

Hopefully @egguy can help with the HA addon.

xhemp, 2023-03-17:
Hey! 
I am not sure if there's something else that needs to be exported but I know that extraacpowertogrid is PV Power coming through the panels.
Is there anything else grott send to PVOutput?

egguy, 2023-03-18:
Hello, the integration should work with the new layout and fields, but the new fields are going to be seen as raw values without unit.

logicland, 2023-04-20:
@johanmeijer @egguy 

FYI with the help of my son, managed to get Beta version (2.8.x) working in Home Assistant, it works fantastic so a very big "thank you" for work on this.

The combination is "SPA 3000TL BL" + Shinelan  + "GBLI6532 6.5kWh Battery"  and the Dashboard closely matches mqtt, but there is a lot of data to validate. 

Attached are the MQTT messages, the "Battery type" shows "Unknown" the actual data reads as 1,  it is in fact LiFePO4 (Lithium) 

If you need any data analysis or info, I'll be happy to help.

[MQTT jsonmsg V1.txt](https://github.com/johanmeijer/grott/files/11285415/MQTT.jsonmsg.V1.txt)

logicland, 2023-04-22:
Indeed Growatt's documentation suggests 0=Lead-Acid, 1=Lithium, 2=Other 

To explain further, in HA -> "Settings" -> "Devices & Services" -> "core-mosquitto" -> "devices and entities" -> <inverter S/No> -> "Sensors" shows the following data :-

"Batteries Type  Unknown"

 "Bmcell3volt   0.01"   

a jpg of the full HA display data is attached.

From the Grott Log the MQTT jsonmsg: shows "batterytype" :1  and "bmscell3volt": 1 seems too low compared to the others, but at the moment I have no way to verify. There's no fault, the Inverter is running normally although Inverter Status=6, bmsoh = 100%,  so I have some investigating to do.

[image]

egguy, 2023-04-22:
Hello,

It seems the problem comes from my plugin. Grott HA, i think the code for the template doesn't make a correct check:

```jinja2
"{% if value_json.batterytype == 0 %}Lithium{% elif value_json.batterytype == '1' %}Lead-acid{% elif value_json.batterytype == '2' %}Other{% else %}Unknown{% endif %}",
```
I will try to correct it in the new release.

Benchamoneh, 2023-06-23:
For some positive feedback, I've just pulled 2.8.2 and seems to be recognising my SPA3000TL "AcCouple" device with ShineWifi-S logger correctly. Stats look accurate when compared with the Growatt Dashboard (I'm discharging right now but will check tomorrow that charging displays properly.

```
         - Grott automatic protocol detection
         - Grott data record length 839
         - layout   :  T060104XSPA
         **- no matching record layout found, try generic**
         - Record layout used :  T06NNNNXSPA
         - Growatt data decrypted V2
         - Grott Growatt data decrypted
         - Growatt plain data: [omitted]
         - Growatt new layout processing
                 - decrypt       :  True
                 - offset        :  6
                 - record layout :  T06NNNNXSPA

         - Grott data record date/time processing started
         - date-time:  2023-06-23T19:36:08
         - Grott values retrieved:
                 -  datalogserial        :  JPC8CDH05J
                 -  pvserial             :  WPDACBP067
                 -  pvstatus             :  6
                 -  uwsysworkmode        :  81
                 -  systemfaultword0     :  0
                 -  systemfaultword1     :  0
                 -  systemfaultword2     :  0
                 -  systemfaultword3     :  0
                 -  systemfaultword4     :  0
                 -  systemfaultword5     :  0
                 -  systemfaultword6     :  0
                 -  systemfaultword7     :  0
                 -  pdischarge1          :  70.0
                 -  pcharge1             :  0.0
                 -  vbat                 :  53.1
                 -  SOC                  :  87
                 -  pactouserr           :  0.0
                 -  pactousers           :  0.0
                 -  pactousert           :  0.0
                 -  pactousertot         :  0.0
                 -  pactogridr           :  0.0
                 -  pactogrids           :  0.0
                 -  pactogrid t          :  0.0
                 -  pactogridtot         :  0.0
                 -  plocaloadr           :  500.0
                 -  plocaloads           :  0.0
                 -  plocaloadt           :  0.0
                 -  plocaloadtot         :  500.0
                 -  ipm                  :  0.0
                 -  battemp              :  26.3
                 -  spdspstatus          :  0.6
                 -  spbusvolt            :  5431.2
                 -  etouser_tod          :  10.1
                 -  etouser_tot          :  2500.5
                 -  etogrid_tod          :  1.3
                 -  etogrid_tot          :  525.1
                 -  edischarge1_tod      :  4.8
                 -  edischarge1_tot      :  1361.6
                 -  eharge1_tod          :  7.4
                 -  eharge1_tot          :  1443.2
                 -  elocalload_tod       :  24.2
                 -  elocalload_tot       :  5292.9
                 -  dwexportlimitap      :  0.0
                 -  epsfac               :  0.0
                 -  epsvac1              :  0.0
                 -  epsiac1              :  0.0
                 -  epspac1              :  0.0
                 -  epsvac2              :  0.0
                 -  epsiac2              :  0.0
                 -  epspac2              :  0.0
                 -  epsvac3              :  0.0
[... 93 lines cut ...]
                         "epsiac1": 0, "epspac1": 0, "epsvac2": 0, "epsiac2": 0, "epspac2": 0,
                         "epsvac3": 0, "epsiac3": 0, "epspac3": 0, "loadpercent": 0, "pf": 1000,
                         "bmsstatusold": 0, "bmsstatus": 97, "bmserrorold": 0, "bmserror": 0,
                         "bmssoc": 87, "bmsbatteryvolt": 5310, "bmsbatterycurr": 65436,
                         "bmsbatterytemp": 26, "bmsmaxcurr": 10420, "bmsgaugerm": 10690,
                         "bmsgaugefcc": 12240, "bmsfw": 34181, "bmsdeltavolt": 4, "bmscyclecnt": 232,
                         "bmssoh": 100, "bmsconstantvolt": 5680, "bmswarninfoold": 0, "bmswarninfo":
                         0, "bmsgaugeiccurr": 0, "bmsmcuversion": 34181, "bmsgaugeversion": 0,
                         "bmswgaugefrversionl": 0, "bmswgaugefrversionh": 0, "bmsbmsinfo": 16720,
                         "bmspackinfo": 16720, "bmsusingcap": 0, "bmscell1volt": 3324,
                         "bmscell2volt": 3322, "bmscell3volt": 1, "bmscell4volt": 3600,
                         "bmscell5volt": 3323, "bmscell6volt": 3323, "bmscell7volt": 3323,
                         "bmscell8volt": 3321, "bmscell9volt": 3322, "bmscell10volt": 3322,
                         "bmscell11volt": 3322, "bmscell12volt": 3322, "bmscell13volt": 3322,
                         "bmscell14volt": 3323, "bmscell15volt": 3323, "bmscell16volt": 3325,
                         "acchargeenergytoday": 81, "acchargeenergytotal": 17084, "acchargepower": 0,
                         "70%_invpoweradjust": 986, "extraacpowertogrid": 4300, "eextratoday": 180,
                         "eextratotal": 33991, "esystemtoday": 166, "esystemtotal": 34778,
                         "inverterstatus": 6, "pacs": 846, "fac": 4999, "vac1": 2402, "iac1": 5,
                         "pac1": 716, "eactoday": 45, "eactot": 13369, "timetotal": 59503577,
                         "Temp1": 331, "Temp2": 314, "Temp3": 200, "Temp4": 0, "uwbatvoltdsp": 533,
                         "pbusvoltage": 3916, "nbusvoltage": 3909, "remotectrlen": 0,
                         "remotectrlpower": 0, "eacchargetoday": 81, "eacchargetotal": 17084,
                         "priority": 0, "batterytype": 1, "autoproofreadcmd": 0}}
```

Benchamoneh, 2023-07-01:
While comparing the output from Grott to the Growatt Cloud I notice there is a value not reported by Grott to MQTT: **PpvInverter**.

This value measures solar generation and is created by the SPA by reading from a CT clamp over the inverter feed in (I have a separate inverter). The value is then divided by 1000 to give the solar energy created.

Could this value be added? Do you need more information from me?