johanmeijer/grott #265: Grott not working since Linux upgrade

JCdeAlmeria, 2023-01-07:
Hello, I have been using Grott for 2 months perfectly.
Grott installed with docker to use the data with homeassistant in other container, under Ubuntu.

Yesterday I installed RetroPie in the PC with grott, with a sudo apt update && sudo apt-get dist-upgrade before.
From that installation (or perhaps from the upgrade), Grott doesn't send info to homeassistant.
In the logs I only see:
         - Growatt packet received:
                  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('XXX.XX.X.X', 5279), raddr=('IP_OF_DATALOGGER', 1029)>
         - Data less then minimum record length, data not processed

The fd sometimes is 4 (with the raddr= IP of Growatt server, and others 5, with the internal IP of my datalogger).
The problem is there is no other kind of packet received, so no data from the inverter (SPH3000) is received, as in previous days.

I have rebooted the PC and stop and rebuild grott, with no results.

Am I missing something?
Is perhaps some problem with changed dependencies on host?
What can I do to recover grott working?

Please, I need help.

My docker-compose configuration is (although I have no changed it):
version: '3.8'
services:
  grott:
    container_name: grott
    image: ledidobe/grott
    restart: unless-stopped
    ports:
      - "5279:5279"
    environment:
      - ginvtype=sph
      - gblockcmd=True
      - gnomqtt=False
      - gmqttip=IP of the PC with mosquitto, homeassistant, etc.
      - gmqttauth=True
      - gmqttuser=*******
      - gmqttpassword=*****
      - TZ=Europe/Madrid

johanmeijer, 2023-01-08:
I think that the growatt server can not be reached from grott (the retro pie). 

What you see is that the inverter tries to connect to the growatt server with a ping (a smll record that is ignored by grott, but can be shown if you specif gminrecl = 1 in the docker compose). After the growatt server responses to the ping the real communication will be started and data will be sent by the inverter and processed by grott (and send via MQTT to HA). 

Can you perform a ping from ubuntu and the retro pie to the growatt server(ping server.growatt.com) so we can see if there is something blocking the connection setup?

JCdeAlmeria, 2023-01-08:
Thanks for your fast response.
I have done a ping to server.growatt.com from a terminal in Ubuntu without any problem.
But after the ping I checked the logs with docker logs -f grott and now a new message appears:
- ('47.91.67.66', 5279) has disconnected
I have reboot after that and when grott starts, it doesn't connect after the Grott proxy mode started line. In previous starts, there is a line that says - ('my_datalogger_IP', port) has connected, and then the lines with the packets received.
Now I have the Grott proxy mode started line, a line with the Hostname, a line with the internal IP of docker and the port 5279 and nothing more in the log...
Perhaps my IP is blocked by the growatt server after the ping?

JCdeAlmeria, 2023-01-08:
I changed several times the datalogger server domain from the PC with grott IP to growatt server, but no data in the Shine phone app...
Perhaps there is something wrong in my datalogger after changes?

JCdeAlmeria, 2023-01-08:
Ok, now I have again the lines about packets received with fd=4 and fd=5

johanmeijer, 2023-01-09:
If you see the growatt packet received messages I supose there is nothing wrong with your datalogger settings. It is still sending the data. 

I think there is something going wrong with the connection to the growatt server. It should be something with your network settings. Something that might been changed because of the retropie installation. 

Can runt the grott container with gminrecl=1 added to the compose settings? I do understand if you do not want to share the docker log via the internet. You can sent it to grott@ledidobe.nl . 

In the meanwhile lets try to look on the internet if we can find some issue's with ubuntu, docker and retropie.

johanmeijer, 2023-01-09:
Can you also look at the syslog if you find any additional problems (e.g. with journalctl). I would look for docker, network or security issue's

JCdeAlmeria, 2023-01-12:
Hello,
I have send you logs to your email.
I have discovered that Grott restarts every minute (using docker ps). The status never grows bigger than 59 seconds. At 60 seconds, the status appear as restarting and the time in seconds start again:
CONTAINER ID   IMAGE            COMMAND                  CREATED        STATUS                                    PORTS                                       NAMES
************   ledidobe/grott   "python -u grott.py …"   33 hours ago   Restarting (1) Less than a second ago     0.0.0.0:5279->5279/tcp, :::5279->5279/tcp   grott

I suspect this is the problem, that Grott is restarting every minute...

johanmeijer, 2023-01-12:
Received and replied the email. 

I have tested it and gminrecl=1 with grott version 2.6.1g is the reason why the grott containers stops and is being restarted every minute.

This is solved in the newer version of Grott (ledido/grott:beta or ledidobe/grott:2.7.6). 

I am not sure if the newer version of Grott also solve the real issue. Please try.

johanmeijer, 2023-01-17:
@JCdeAlmeria have you already be able to test with the beta version?

JCdeAlmeria, 2023-01-17:
Yes, I have tryed grott 2.7.6 and everything is working now.
I am obtaining buffered data packets at every 30 seconds corresponding to info from previous days (that was saved in the datalogger every 5 minutes), mixed with actual data every 5 minutes... after two days I am receiving data perfectly.
I suppose this is the solution, although I don't understand what is inside beta version to solve the problem RetroPie created.
Thanks a lot.
Do you want me to do any additional test to help improving grott?