johanmeijer/grott #269: No internet, No grott !

mahmoud-os93, 2023-01-25:
Hello,
I have noticed that whenever there an internet outage in my home, there is no updates using mqtt in hass although everything is reachable locally.
Before troubleshooting and getting into the logs. is this done by design ? should I get values with no communication with the growatt server ?
I am currently using the : docker pull ledidobe/grott:beta
thanks.

johanmeijer, 2023-01-25:
Yes grott standalone is a proxy that sits between the inverter en the growatt server. 

If you do not a connection with internet (growatt server) there is no data to proces.

Good news is that the data is beeing buffered in the inverter and send at the moment the connection is  established.

If you want to be totally independe you have to use grottserver. That will eliminate the need for yhe connection with the growatt server.

Grottserver is under development but works rather well. See #98 for information.

The grottserver docker image will run both grott and grottserver.

mahmoud-os93, 2023-01-25:
> Yes grott standalone is a proxy that sits between the inverter en the growatt server.
>
> If you do not a connection with internet (growatt server) there is no data to proces.
>
> Good news is that the [...]

I see, thank you for that ! Does it support higher refresh rates ( <1min )

johanmeijer, 2023-01-26:
Yes. Grottserver is replacing the growatt server (hence being independent  from China). I know some people want to have both and the server functionality (sending commands via an API) and sending data to Growatt. But at this moment that is not available. Was not the initial design. Might be available in the future. 

Yes you can set higher refresh rate with Grottserver. I managed to have refresh rates <20 seconds. 

You can also set other refresh rates via the Growatt webpages but the growatt server is not handling that very well and does not like refresh rates lower then the current 5 minutes.

closed 2023-01-26 by mahmoud-os93