johanmeijer/grott #272: shinewifi-s lost connection

smallbitt, 2023-02-02:
Not sure if it could depend on grout server on my raspberry, but since yesterday the shinewifi device blink green led (so its connected to wifi but no internet connection or no connection to server).

has been changed something ing Growatt server side or what?

I tried to reset and reinstall, all seems going well but still "connection lost".

in the datalogger panel I see 95.246.84.116:42758 what is that? seems its looking at a different IP address.

I'm trying to change ip or settings via setup of the cataloguer but I get "Datalogger communication lost"

does anyone knows what happened?

PS ofc my wifi and internet connection is working and I have changed anything in my setup.

johanmeijer, 2023-02-02:
Do you see something in your grott log?

The 95. ip adres is probably your internet address?

smallbitt, 2023-02-02:
95... it is not my IP address.

the log seems ok but don't know what to search.

johanmeijer, 2023-02-02:
When you stop/start grott do you see that a communication flow starts ?

95.246.84.xxx IP addresses belongs to telecomitallia. If this is your internet provider then this probably your outside internet IP.

smallbitt, 2023-02-02:
flow doesn't start in fact.

ok that ip is provider ip indeed. so the datalogger is connected but growatt says its disconnected and also not sending anything to grott

smallbitt, 2023-02-02:
I mean, the server start but journalctl -f gives me no new messages, like the cataloguer not sending anything

smallbitt, 2023-02-02:
Maybe I found the problem, I left grottserver ip assigned by dhcp and now changed its ip, so the datalogger can't find it and its like in a n "isolated" state. Is there a way to change the datalogger setup directly without shine phone?

johanmeijer, 2023-02-02:
Otherwise assign fix ip adress to your server

smallbitt, 2023-02-02:
fixed, through shinephone app I could change the server ip from "advanced" menu. thank you..

johanmeijer, 2023-02-06:
I asume this one can be closed

closed 2023-02-06 by johanmeijer