johanmeijer/grott #279: influxdb stopped collecting after a while.

cracyfloyd, 2023-02-08:
all seems to be running but after a while the collecting is stopping. when i am observe the log it seems to be all the collecting running fine. but when you are looking in detail you can see the time of the collecting is wrong.

[image]
[image]

johanmeijer, 2023-02-09:
Some dataloggers show a big time drift (do not keep up with the real time) . I am not sure why,  bu, this seems to be a hardware error. 

The first time the inverter/datalogger  starts (in the morning) the growatt server sent a time command to set the clock (with the time/date in the zone you configured at the Growatt site). Normally this is done ones a day. So if the datalogger has a timing problem the time can vary during the day. 
Normally the datalogger date/time is inserted in the data and used by Grott to create the MQTT, PVOutput en Influxdb records, 

If there are problems with the time drift of the datalogger we can do some things to bypass the problems.

* Restart Grott at regular base (there will be a Time sync command sent from the Growatt site)
The question is how big if is the time drift and how often you have to do this. Advantage is that all records will be processed by Grott (also eventually "buffered"records because of interupted connection)  
* If you are using Grott server you can sent a time update command via the API.
Same Advantages and disadvantages as a above. 
* Let Grott ignore the date/time in the data record and use the Server date/time by specifying time = server (gtime="server")
If the time of the server Grott is running on is set oke (think about time zones etc) this will be a solution for real time data. With this parameter active, Grott will not proces buffered records because there is no clue about when this record is created. This might not be a real problem if there are no problems with the connection, 

I am not sure if a datalogger upgrade can fix or introduce this problems. But upfront I do not rule this out.

cracyfloyd, 2023-02-11:
Very very thnx for your detailed information.

For me it seems thats the problem is solved by self. 2 days now are the collecting to influx are without to stop the collect and the time is every time correct in the log.  Now i use in the docker the env gtime=server, maybe this make it more stable. 

Thnx again Markus

johanmeijer, 2023-02-12:
Super. Can we close this one?

cracyfloyd, 2023-02-14:
Yes, you can close. it runs and runs and runs....
Thnx so much..

closed 2023-02-14 by johanmeijer