johanmeijer/grott #281: SPH6000 hybrid inverter layout - missing battery?

catch56, 2023-02-09:
Just got grott up and running. I seem to be getting data for both solar arrays OK, but I'm not seeing anything for the battery.

This is for an SPH6000.

cli output:

```
 - Growatt original Data:
		 \x00\x0f\x00\x06\x03\x3f\x01\x03\x0d\x22\x2c\x4f\x22\x33\x26\x77\x36\x37\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x21\x20\x22\x20\x37\x02\x2a\x5f\x35\x51\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x76\x76\x7d\x49\x57\x48\x74\x61
		 \x74\x74\x3b\x72\x6e\x7f\x61\x74\x74\x47\x16\x6f\x13\x46\x64\x74\x47\x65\x1f
		 \x79\x71\x26\x35\x76\x5c\x5f\x57\x3b\x37\x36\x06\x72\x69\x77\x60\x74\x74\x42
		 \xae\x6f\x69\x61\x35\x74\x8f\x72\x0b\x77\x61\x3a\x23\x04\x26\x2c\x32\x39\x44
		 \x36\x77\x73\x6f\x85\x14\x74\x75\x47\x28\x6f\x77\x61\x74\x54\x67\x52\x21\x12
		 \x16\x54\x31\x29\x17\x1d\x10\x18\x54\x54\x67\x72\x6e\x7a\xcc\x74\x74\x40\x95
		 \x6f\x75\x61\x7d\x74\x49\x72\x49\x77\x42\x74\x70\x40\x42\x65\x49\x73\xfa\x60
		 \x17\x75\x5f\x7d\xd0\x66\x28\x53\x22\x6f\x77\x61\x74\x74\x47\x72\x6f\x70\xb0
		 \x7e\x4a\x55\xfc\x7b\x27\x61\x0b\x74\x73\x72\x10\x77\x7a\x77\x9e\x47\x6b\x6f
		 \x6c\x61\x6f\x74\x47\x72\x6f\x77\x61\x74\x74\x4e\xb6\x6f\x77\x33\x35\x35\x06
		 \x40\x5e\x45\x51\x44\x42\x47\x72\x6e\x46\x61\x74\x74\x47\x61\xdf\x77\x79\x7d
		 \x96\x4e\xc6\x67\x61\x69\x30\x74\x53\x72\x6a\x7e\x0e\x7c\x88\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\xe2\x47\x72\x6e\x93\x61\x8b\x3a\x67\x72
		 \x90\x39\x41\x74\x8b\x09\x52\x6f\x88\x2f\x54\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x77\xaf\x76\x0b\x60\x11\x74\x74\x55\xb2\x6f\x45\x60
		 \xa2\x76\x78\x60\xaf\x61\xc9\x26\xd4\x43\x3e\x6d\xcb\x65\x38\x76\x1f\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x10\x74\x47\x72\x65\x77
		 \x61\x74\x75\x47\x72\x6f\x13\x61\x10\x74\x47\x72\x6e\x77\x61\x74\x74\x47\x72
		 \x6f\x76\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x75\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x05
		 \x74\x10\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x0b\x77\x6b\x74\x75\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x51\x6c\x6e\x69\x61\x74\x75\x59\x77\x6f\x77\x60\x71\x74\x51\x6c\x6f\x77
		 \x61\x7e\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x60\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x76\x0a\x73\x80\x33\x24
		 \x0c\x42\x5d\x46\x58\x41\x46\x75\x40\x56\x43\x27\x31\x46\x00\x22\x24\x47\x53
		 \x45\x4d\x72\x40\x5d\x45\x58\x41\x45\x7e\x43\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x03\x41
	 - Grott automatic protocol detection
	 - Grott data record length 839
	 - layout   :  T060103X
	 - no matching record layout found, try generic
	 - Record layout used :  T060103X
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 000f0006033f01034a50433843475230445800000000000000000000000000000000000000004
		 e57435443455830423000000000000000000000000000000000000000001702090e2527030000
		 007c000108000000006400642710000017700e105241312e30205a43424100060001000005dc0
		 01e004100c8006400004e5743544345583042300100f2750001005a000000002020204e657720
		 456e6572677920202000010dad000007e700020009000e00260023000407300a3e128e1450073
		 00ab1125c1450000000000000000007d10a3e128e1450007f0034007f001b03ea0019001b001b
		 000000000000000009c4000052414141323132303036000001310000000013b0001809e209b40
		 816084400140005096f08fc0000000000000000000000000096000001e400ff4e2000ff4e2000
		 ff4e2000ff4e20000000000000000000000000000003e804641770000012c0003201d6023f12c
		 016a852a0044c02bc044c02580000000000000000000000000000000000000000000000000000
		 000000000000000000000000000000000000000000000000000000640000000a0000000100000
		 06400640000000100000000000000010000000000000000000000000000000100000000000000
		 00000000000000000000000064006400000000000000000000000000000000000000000000000
		 00000000000000000000000000064000a00010000000000000000000000000000161e011e0000
		 011e050000010500161e0000000a0000000000000000000000000000000000000001000000000
		 00000000000046504e147504b3032313935323232393446453247504b30323139353232323935
		 31393100000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 0000000000000000000000000000000000000000000000000000000007420
```

mqtt message looks like this:

```
{"device": "NWCTCEX***", "time": "2023-02-09T14:42:16", "buffered": "no", "values": {"datalogserial": "JPC8CGR***", "pvserial": "NWCTCEX***", "pvstatus": 5, "pvpowerin": 9700, "pv1voltage": 2203, "pv1current": 27, "pv1watt": 6122, "pv2voltage": 3690, "pv2current": 9, "pv2watt": 3626, "pvpowerout": 0, "pvfrequentie": 0, "pvgridvoltage": 0, "pvgridcurrent": 0, "pvgridpower": 0, "pvgridvoltage2": 0, "pvgridcurrent2": 0, "pvgridpower2": 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3": 196088712, "totworktime": 0, "pvenergytoday": 0, "pvenergytotal": 0, "epvtotal": 68, "epv1today": 3978, "epv1total": 10703136, "epv2today": 31, "epv2total": 790, "pvtemperature": 393, "pvipmtemperature": 532}}
```

johanmeijer, 2023-02-09:
Did you specify invtype=sph in the grott.ini (or -e ginvtype="sph" in docker run)?

catch56, 2023-02-09:
I thought I'd tried that and got the same results, but I'd actually used 'SPH' instead of 'sph'...

Here's the results with that, it does look a bit different, using a different layout, but still no sign of the battery:

` - Growatt original Data:
		 \x00\x94\x00\x06\x03\x3f\x01\x04\x0d\x22\x2c\x4f\x22\x33\x26\x77\x36\x37\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x21\x20\x22\x20\x37\x02\x2a\x5f\x35\x51\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x76\x76\x7d\x57\x46\x5a\x74\x61
		 \x74\x74\x3b\x72\x6a\x77\x61\x74\xbc\x41\x59\x6f\x77\x61\x74\x74\x57\x7e\xa0
		 \x77\x61\x74\x74\x47\x99\x6f\x76\x33\xd4\x7d\xe9\x75\xc0\x7f\xed\x74\x8e\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x67
		 \xec\x67\xf0\x4e\x1f\x6f\x62\x61\x74\x64\x86\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x2c\x74
		 \x47\x7d\xf1\x77\xc2\xfa\x49\x47\x72\x6f\x53\x61\x74\x77\x5c\x72\x6f\x77\x21
		 \x74\x74\x4f\x30\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x7c\x12\x75
		 \x5c\x46\x7f\x6e\x61\x61\x74\x76\x4e\x62\xd7\x67\x85\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x65\x74\x74\x40\xe6\x6f\x77\x61\x74\x74\x47\x72\x6e\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x77\xaf\x76\x0b\x77\x64\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x7c\x53\x61\x74\x74\x47\x70\x65
		 \x77\x55\x74\x74\x47\x70\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x76\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x72\x98
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x7c\x9b\x61\x74\x74\x55\x72\x6a\x7c
		 \x17\x55\x40\x47\x72\x6f\x6b\x61\x74\x6e\xcf\x72\x6f\x77\x63\x74\x74\x47\x0f
		 \x6f\x77\x61\x5d\x74\x47\x78\x82\x77\x61\x74\x5f\x47\x72\x65\xeb\x61\x74\x74
		 \x3c\x72\x6f\x52\xae\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x62\x9c\x74\x47\x73\x0c\x77\x61\x74\x74\x47\x46\x7b\x13\x9d\xe4\x74\x55
		 \x6b\x6f\x6e\x27\x5b\x8c\x09\x3c\x6f\x70\x61\x5d\x74\x23\x64\x5f\x77\x61\x74
		 \x74\x47\x72\x21\x39\x61\x74\x74\x47\x72\x6f\x36\x31\x35\x24\x47\x72\x63\xb6
		 \x6d\xce\x74\x45\x7c\x2d\x7b\xdf\x78\xca\x4b\xcc\x63\xc9\x6d\xb5\x78\x87\x7e
		 \xaf\x7b\xa1\x78\xc9\x4b\xcf\x63\xcb\x6d\xce\x74\x47\x76\x0a\x73\x80\x74\x70
		 \x47\x72\x68\xe3\x61\x74\x74\x47\x76\x74\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x3e\x74\x74\x57\xbc\x6f\x77\x61\x2a\x74\x47\x62\x09
		 \x77\x61\x67\x98\x47\x72\x7c\x9b\x61\x74\x74\x24\x72\x6d\x77\x61\x70\x91\x47
		 \x70\x6f\x77\x64\x7a\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x46\x7e\xae\x7b\xdb
		 \x74\x76\x47\x52\x6f\x61\x61\x6c\x74\xfa\x72\xde\x77\x65\x74\x75\x47\x72\x6f
		 \x42\x61\x40\x74\x47\x72\x6f\x77\x61\x74\x74\x47\xb1\x6f\xb7\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x0f\x82
	 - Grott automatic protocol detection
	 - Grott data record length 839
	 - layout   :  T060104X
	 - no matching record layout found, try generic
	 - Record layout used :  T06NNNNX
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00940006033f01044a50433843475230445800000000000000000000000000000000000000004
		 e5743544345583042300000000000000000000000000000000000000000170209103435030000
		 007c0005000000c8062b0000000000100ccf0000000000eb000152a009ae07af088c00fa00000
		 00000000000000000000000000000000000000000000000000000000000000000000000108d13
		 84096d0015000010c1000000000000000000000000000000000000000000000000005800000f9
		 e00a38e3d000000240000031b0000004000000842000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000b730128010d0
		 1160000020910b810e40000000000000000000000000000000000000000000000000000000400
		 00079400000000000000010000000000000000000003e80464000500000000000000000000000
		 0000000000000132400000000020a003400000002000000000000000000000002000000000000
		 00000000000000000000000013ec0000000000000000000013ec0000001200050b76213400000
		 01c00001a88000000020000007d0000002900000aed0000002b00000a9c0000007b000025cf00
[... 44 lines cut ...]
		 -  epvtotal             :  8.8
		 -  epv1today            :  399.8
		 -  epv1total            :  1071878.1
		 -  epv2today            :  3.6
		 -  epv2total            :  79.5
		 -  pvtemperature        :  29.6
		 -  pvipmtemperature     :  52.1
	 - MQTT jsonmsg: 
			 {"device": "NWCTCEX***", "time": "2023-02-09T16:52:53", "buffered": "no",
			 "values": {"datalogserial": "JPC8CGR***", "pvserial": "NWCTCEX***",
[... cut ...]
			 "pvtemperature": 296, "pvipmtemperature": 521}}
`

Not sure if it's me misunderstanding the keys or an actual issue, but while pv1 and pv2 match what I see on the growatt server, evp1today and epv2today look mismatched.

If it makes any difference I have two 3.3kwh batteries connected as opposed to one single battery, but they don't show up as two batteries anywhere that I can see.

SOC is still showing fine in the growatt app/website so it's passing through OK (as you'd expect, but just for completeness).

johanmeijer, 2023-02-09:
Hmm I think think still the sph layout is not used. I expect a message about layout T06NNNNXSPH being used.

Can you sent the log information from grott startup (that shows the used parameters).

catch56, 2023-02-09:
Apologies that last output was PEBKAC, I'd completely failed to add invtype properly.

Layout isn't  specifically T06NNNNXSPH but I'm getting battery data now!

`Grott data record length 585
	 - layout   :  T060136XSPH
	 - no matching record layout found, try generic
	 - Record layout used :  T060136XSPH
`

I'd assumed the autodetect would pick up the SPH, would it be worth amending the documentation to suggest slightly more strongly trying out invtype if any data is missing?

Data output for posterity:
`
 -  datalogserial        :  JPC8CGR***
		 -  pvserial             :  NWCTCEX***
		 -  pvstatus             :  5
		 -  pvpowerin            :  0.0
		 -  pv1voltage           :  137.4
		 -  pv1current           :  0.0
		 -  pv1watt              :  1.4
		 -  pv2voltage           :  146.2
		 -  pv2current           :  0.0
		 -  pv2watt              :  2.2
		 -  pvpowerout           :  434.5
		 -  pvfrequentie         :  50.0
		 -  pvgridvoltage        :  243.4
		 -  pvgridcurrent        :  2.1
		 -  pvgridpower          :  426.6
		 -  pvgridvoltage2       :  0.0
		 -  pvgridcurrent2       :  0.0
		 -  pvgridpower2         :  0.0
		 -  pvgridvoltage3       :  0.0
		 -  pvgridcurrent3       :  0.0
		 -  pvgridpower3         :  0.0
		 -  totworktime          :  1489.2
		 -  eactoday             :  8.9
		 -  pvenergytoday        :  8.9
		 -  eactotal             :  399.9
		 -  epvtotal             :  293.1
		 -  epv1today            :  3.6
		 -  epv1total            :  79.5
		 -  epv2today            :  6.4
		 -  epv2total            :  211.4
		 -  pvtemperature        :  28.9
		 -  pvipmtemperature     :  25.9
		 -  pvboosttemp          :  25.6
		 -  bat_dsp              :  51.9
		 -  eacharge_today       :  0.4
		 -  eacharge_total       :  194.0
		 -  batterytype          :  1
		 -  uwsysworkmode        :  5
		 -  systemfaultword0     :  0
		 -  systemfaultword1     :  0
		 -  systemfaultword2     :  0
		 -  systemfaultword3     :  0
		 -  systemfaultword4     :  0
		 -  systemfaultword5     :  0
		 -  systemfaultword6     :  0
		 -  systemfaultword7     :  0
		 -  pdischarge1          :  520.0
		 -  p1charge1            :  0.0
		 -  vbat                 :  52.0
		 -  SOC                  :  0.5
		 -  pactouserr           :  9.6
		 -  pactousertot         :  9.6
		 -  pactogridr           :  0.0
		 -  pactogridtot         :  0.0
		 -  plocaloadr           :  520.0
		 -  plocaloadtot         :  520.0
		 -  spdspstatus          :  0.5
		 -  spbusvolt            :  292.3
		 -  etouser_tod          :  2.9
		 -  etouser_tot          :  679.3
		 -  etogrid_tod          :  0.2
		 -  etogrid_tot          :  12.5
		 -  edischarge1_tod      :  4.3
		 -  edischarge1_tot      :  279.9
		 -  eharge1_tod          :  4.3
		 -  eharge1_tot          :  271.6
		 -  elocalload_tod       :  12.6
		 -  elocalload_tot       :  968.2
`

Going to go ahead and mark this as closed. Thanks for the help and apologies for missing the obvious!

closed 2023-02-09 by catch56