johanmeijer/grott #286: grott_ha.py - Can't configure device for HA MQTT

Freekers, 2023-02-14:
I'm trying to use the [grott_ha.py](https://github.com/johanmeijer/grott/blob/master/examples/Home%20Assistent/grott_ha.py) extension by @egguy to automatically configure my Growatt inverters in HA. However, it returns code 7 which according to the documentation means 'Can't configure device for HA MQTT'. I'm not sure how to proceed.  Grott is able to reach the MQTT broker; I can see it connecting but it doesn't publish anything.

I'm running everything in Docker:
- Home Assistant: lscr.io/linuxserver/homeassistant:latest (Currently: 2023.2.4)
- MQTT: eclipse-mosquitto:latest
- Grott: ledidobe/grott:latest (Currently: 2.6.1g)

Here's the logging of Grott:

```
Grott settings:
      
:
	version:     	 2.6.1g
	verbose:     	 True
	trace:       	 False
	config file: 	 grott.ini
	minrecl:     	 100
	decrypt:     	 True
	compat:      	 False
	invtype:     	 default
	include_all: 	 False
	blockcmd:    	 False
	noipf:       	 False
	time:        	 auto
	sendbuf:     	 True
	timezone:    	 local
	valueoffset: 	 6
	offset:      	 6
	inverterid:  	 automatic
	mode:        	 proxy
	grottip      	 default
	grottport    	 5279
_MQTT:
	nomqtt       	 True
	mqttip:      	 172.22.0.252
	mqttport:    	 1883
	mqtttopic:   	 energy/growatt
	mqtttauth:   	 True
	mqttuser:    	 XXXXXX
	mqttpsw:     	 **secret**
_Growatt server:
	growattip:   	 47.91.67.66
	growattport: 	 5279
_PVOutput:
	pvoutput:    	 False
	pvurl:       	 https://pvoutput.org/service/r2/addstatus.jsp
	pvapikey:    	 yourapikey
	pvinverters: 	 1
	pvsystemid:  	 12345
_Influxdb:
	influx:      	 False
	influx2:     	 False
	database:    	 grottdb
	ip:          	 localhost
	port:        	 8086
	user:        	 grott
	password:    	 **secret**
	organization:	 grottorg
	bucket:      	 grottdb
	token:       	 **secret**
_Extension:
	extension:   	 True
	extname:     	 grott_ha
	extvar:      	 {'ha_mqtt_host': '172.22.0.252', 'ha_mqtt_port': '1883', 'ha_mqtt_user': 'XXXXX', 'ha_mqtt_password': 'XXXXXX'}
Grott proxy mode started
Hostname : 2a1c6a075cdf
IP :  172.22.0.6 , port :  5279 
	 - ('192.168.30.195', 21259) has connected
	 - Growatt packet received:
[... 222 lines cut ...]
		 -  pvtemperature        :  27.8
		 -  pvipmtemperature     :  35.6
	 - MQTT jsonmsg: 
			 {"device": "TEG2B02XXX", "time": "2023-02-14T16:59:42", "buffered": "no",
			 "values": {"recortype1": 3000, "recortype2": 3124, "pvstatus": 1,
			 "pvpowerin": 2520, "pv1voltage": 2034, "pv1current": 9, "pv1watt": 1940,
			 "pv2voltage": 1893, "pv2current": 3, "pv2watt": 580, "pvpowerout": 2489,
			 "pvfrequentie": 5004, "pvgridvoltage": 2341, "pvgridcurrent": 12,
			 "pvgridpower": 2493, "pvgridvoltage2": 0, "pvgridcurrent2": 0,
			 "pvgridpower2": 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3":
			 0, "totworktime": 44721643, "pvenergytoday": 93, "pvenergytotal": 46204,
			 "epvtotal": 47690, "epv1today": 49, "epv1total": 18846, "epv2today": 44,
			 "epv2total": 28844, "pvtemperature": 278, "pvipmtemperature": 356}}
	 - No MQTT message sent, MQTT disabled
	 - Grott Send data to PVOutput disabled 
	 - Grott Send data to Influx disabled 
	 - Grott extension processing started :  grott_ha
	[Grott HA] 0.0.7-rc2 Can't configure device: TEG2B02XXX
	 - Grott extension processing ended :  7

```

Please let me know if you need more info.

Thanks!

Freekers, 2023-02-14:
To add: Same issue happens with rc3 of the grott_ha.py file, which is newer than the one in this repo. It can be found here: https://raw.githubusercontent.com/egguy/grott/feature/ha-extension/examples/Home%20Assistent/grott_ha.py

egguy, 2023-02-15:
Hello, 

I will take a look to see if I can see locate the origin of the problem.

In the meantime, it seems you are running version 2.6.1g, the plugin is mostly made for  2.7 series.
And from a quick glances, the error 7 could be triggered by a missing layout, which could happen in the 2.6 series.

Could you try to run a 2.7 container? 

Thanks

Freekers, 2023-02-16:
2.7 container in combination with rc5 works. Thanks for your help!

closed 2023-02-16 by Freekers

truwer, 2023-04-04:
Thank you. Had the same issue, which was solved with pulling a dedicated version (ledidobe/grott:2.7.8)  instead of the latest tag. After that the plugin works.