johanmeijer/grott #290: Docker is frequently restarting because of erroe message.

codac, 2023-02-17:
After using the latest version of the docker image, the Grott Docker frequently restarts after the following error message:
` Grott data ack record or data record not defined no processing done`
That happens serveral times until the docker ist stable again. This scenario happens probably once a day, usually in the morning times.

johanmeijer, 2023-02-17:
That is not ok 

I need more information. This message is pretty normal in the morning. When the inverter starts it sends an announcement record with layout t06xx03 (xx is 01 or 50 or sometimg like that) Grott wil ignore this record with message you described.

This message will occur after inverter communication starts/restarts (also after restart of grott / grott-container).

Are there more messages in the log.  Do you see also python error messages?

Be aware sometimes the inverter can start multiple times in the morning (e.g when it is not sunny enough) and grott will log this message. But it should not crash grott then.

Can you send more log information from the time of the crash?

Do you use Grott or grottserver?

codac, 2023-02-17:
I'm using Grott.
I found this bit here if it helps:

```
	 - Grott write to influxdb v2
	 - Grott InfluxDB error 
(422)
Reason: Unprocessable Entity
HTTP response headers: HTTPHeaderDict({'Content-Type': 'application/json; charset=utf-8', 'X-Influxdb-Build': 'OSS', 'X-Influxdb-Version': 'v2.6.1', 'X-Platform-Error-Code': 'unprocessable entity', 'Date': 'Fri, 17 Feb 2023 08:33:59 GMT', 'Content-Length': '135'})
HTTP response body: {"code":"unprocessable entity","message":"failure writing points to database: partial write: points beyond retention policy dropped=1"}
Grott Influxdb write error, grott will be stopped
Grott Growatt logging monitor : 2.6.1g
```

johanmeijer, 2023-02-17:
Ok. It seems to be an influxdb error. And for me a strange error because it says that it trying to write an old (outdated) record. Do you also have the information of the record (json message) that grott tries to write? It should be just above this messages in the log.

 Which docker image do you use the latest or the beta?

To be honest the grott:latest is still version 2.6.1g en not 2.7.8 (no time found to change that) and from te log you send still think you are using the 2.6.1g version.

codac, 2023-02-18:
```
	 - MQTT send failed: [Errno 99] Cannot assign requested address

	 - Grott Send data to PVOutput disabled 

	 - Grott InfluxDB publihing started

	 - Timezone local specified default timezone used

	 - Grott original time :  2005-11-16T12:58:15 adjusted UTC time for influx :  2005-11-16T11:58:15

	 - Grott influxdb jsonmsg: 

			 [{'measurement': 'XXXXXXXXXX', 'time': '2005-11-16T11:58:15', 'fields':

			 {'pvserial': 'XXXXXXXXXX', 'recortype1': 3000, 'recortype2': 3124,

			 'pvstatus': 0, 'pvpowerin': 0, 'pv1voltage': 661, 'pv1current': 0,

			 'pv1watt': 0, 'pv2voltage': 0, 'pv2current': 0, 'pv2watt': 0, 'pvpowerout':

			 0, 'pvfrequentie': 4998, 'pvgridvoltage': 2323, 'pvgridcurrent': 1,

			 'pvgridpower': 0, 'pvgridvoltage2': 0, 'pvgridcurrent2': 0, 'pvgridpower2':

			 0, 'pvgridvoltage3': 0, 'pvgridcurrent3': 0, 'pvgridpower3': 0,

			 'totworktime': 4837777, 'pvenergytoday': 0, 'pvenergytotal': 574,

			 'epvtotal': 585, 'epv1today': 0, 'epv1total': 585, 'epv2today': 0,

			 'epv2total': 0, 'pvtemperature': 49, 'pvipmtemperature': 122}}]

	 - Grott write to influxdb v2

	 - Grott InfluxDB error 

(422)

Reason: Unprocessable Entity

HTTP response headers: HTTPHeaderDict({'Content-Type': 'application/json; charset=utf-8', 'X-Influxdb-Build': 'OSS', 'X-Influxdb-Version': 'v2.6.1', 'X-Platform-Error-Code': 'unprocessable entity', 'Date': 'Sat, 18 Feb 2023 07:20:44 GMT', 'Content-Length': '135'})

HTTP response body: {"code":"unprocessable entity","message":"failure writing points to database: partial write: points beyond retention policy dropped=1"}

Grott Influxdb write error, grott will be stopped

Grott Growatt logging monitor : 2.6.1g
```

I'm using ledidobe/grott:latest

johanmeijer, 2023-02-18:
This strange the date of the record it totally outdated ( 2005-11-16T11:58:15) that causes an influx error. I stop Grott after that. Maybe that is not the best idea, but the date is really wrong.

Can you sent just a little more log. With the information about the date/time processing.

Other question what datalogger do you have shinewifi or shinelink?

johanmeijer, 2023-02-19:
see also #295 I will change grott so it will stops after a influxxdb error. 

Be aware the date/time in your record is still strange and we might want to investigate that!

codac, 2023-02-21:
I'm using the Growatt ShineWiFI-X - WiFi-Stick.
As due to a wrong retention time, I have lost one month of data. I'm currently trying to readd them with uploading data with the line protocol ([Details here](https://community.influxdata.com/t/weird-query-behaviour-with-different-time-ranges/28255/6)). I have added two data points for testing. One on the 26.11.2022 and another one on the 08.12.2022.
But I dont get the 16th of 11 2005. The whole system started on the 6th of November 2022.

How can I furhter investigate this date, as there is no datapoint in the database?

This is the log of the 20th of Feb:

```
07:45:08	 - Grott data record date/time processing started

07:45:08	 - date-time:  2005-11-16T13:02:54

07:45:08	 - Grott values retrieved:

07:45:08		 -  pvserial             :  XXXXXXXXXX

07:45:08		 -  recortype1           :  3000

07:45:08		 -  recortype2           :  3124

07:45:08		 -  pvstatus             :  0

07:45:08		 -  pvpowerin            :  0.0

07:45:08		 -  pv1voltage           :  70.4

07:45:08		 -  pv1current           :  0.0

07:45:08		 -  pv1watt              :  0.0

07:45:08		 -  pv2voltage           :  0.0

07:45:08		 -  pv2current           :  0.0

07:45:08		 -  pv2watt              :  0.0

07:45:08		 -  pvpowerout           :  0.0

07:45:08		 -  pvfrequentie         :  50.0

07:45:08		 -  pvgridvoltage        :  231.1

07:45:08		 -  pvgridcurrent        :  0.1

07:45:08		 -  pvgridpower          :  0.0

07:45:08		 -  pvgridvoltage2       :  0.0

07:45:08		 -  pvgridcurrent2       :  0.0

07:45:08		 -  pvgridpower2         :  0.0

07:45:08		 -  pvgridvoltage3       :  0.0

07:45:08		 -  pvgridcurrent3       :  0.0

07:45:08		 -  pvgridpower3         :  0.0

07:45:08		 -  totworktime          :  687.8

07:45:08		 -  pvenergytoday        :  0.0

07:45:08		 -  pvenergytotal        :  57.7

07:45:08		 -  epvtotal             :  58.8

07:45:08		 -  epv1today            :  0.0

07:45:08		 -  epv1total            :  58.8
[... 278 lines cut ...]

07:45:12	database:    	 grott

07:45:12	ip:          	 <ip>

07:45:12	port:        	 8086

07:45:12	user:        	 <user>

07:45:12	password:    	 **secret**

07:45:12	organization:	 <org>

07:45:12	bucket:      	 <bucket>

07:45:12	token:       	 **secret**

07:45:12	_Extension:

07:45:12	extension:   	 False

07:45:12	extname:     	 grottext

07:45:12	extvar:      	 {'ip': '<ip>', 'port': '8000'}
```

johanmeijer, 2023-04-13:
In normal processing Grott is retrieving the date/time from the data record from inverter datalogger. Sometimes this timestap is wrong. Some wifishine-x are a little bit buggy with keeping in sync the time. but then start drifting away from the time during the day. 

I have not heard about a occasionally wrong time record. This will be hard to detect for Grott.  Grott can override the time from the record with the time from the server (docker container if you use that). You have to specify time=server in the grott.ini (-e gtime="server" for docker).  Con of this is that grott can not process the eventually buffered records anymore (if we are doing that records with wrong data/time will be inserted).  

I am not sure how we can filter your wrong 2005 record. That one is a really strange one.