johanmeijer/grott #296: Temporary failure in name resolution

dannytsang, 2023-02-19:
I'm using Grott in Docker on 2.6.1g.

It looks like it's failing to handle a DNS resolution when it can't connect to pvoutput.org. Stack trace below:

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/urllib3/connection.py", line 175, in _new_conn
    (self._dns_host, self.port), self.timeout, **extra_kw
  File "/usr/local/lib/python3.7/site-packages/urllib3/util/connection.py", line 73, in create_connection
    for res in socket.getaddrinfo(host, port, family, socket.SOCK_STREAM):
  File "/usr/local/lib/python3.7/socket.py", line 752, in getaddrinfo
    for res in _socket.getaddrinfo(host, port, family, type, proto, flags):
socket.gaierror: [Errno -3] Temporary failure in name resolution

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/urllib3/connectionpool.py", line 706, in urlopen
    chunked=chunked,
  File "/usr/local/lib/python3.7/site-packages/urllib3/connectionpool.py", line 382, in _make_request
    self._validate_conn(conn)
  File "/usr/local/lib/python3.7/site-packages/urllib3/connectionpool.py", line 1010, in _validate_conn
    conn.connect()
  File "/usr/local/lib/python3.7/site-packages/urllib3/connection.py", line 358, in connect
    conn = self._new_conn()
  File "/usr/local/lib/python3.7/site-packages/urllib3/connection.py", line 187, in _new_conn
    self, "Failed to establish a new connection: %s" % e
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPSConnection object at 0x149e18e63250>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/usr/local/lib/python3.7/site-packages/urllib3/connectionpool.py", line 756, in urlopen
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
  File "/usr/local/lib/python3.7/site-packages/urllib3/util/retry.py", line 574, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='pvoutput.org', port=443): Max retries exceeded with url: /service/r2/addstatus.jsp (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x149e18e63250>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "grott.py", line 35, in <module>
    proxy.main(conf)
  File "/app/grottproxy.py", line 85, in main
    self.on_recv(conf)
  File "/app/grottproxy.py", line 165, in on_recv
    procdata(conf,data)    
  File "/app/grottdata.py", line 427, in procdata
    reqret = requests.post(conf.pvurl, data = pvdata, headers = pvheader)
  File "/usr/local/lib/python3.7/site-packages/requests/api.py", line 117, in post
    return request('post', url, data=data, json=json, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/requests/api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/requests/sessions.py", line 542, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python3.7/site-packages/requests/sessions.py", line 655, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/requests/adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPSConnectionPool(host='pvoutput.org', port=443): Max retries exceeded with url: /service/r2/addstatus.jsp (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x149e18e63250>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))
```

johanmeijer, 2023-02-26:
I have to look closer to thi. Have not seen it before.

gasparakos, 2024-04-26:
I got similar "Temporary failure in name resolution" with 2.8.3 and grott was terminated. I have no logs, because recreated container.

johanmeijer, 2024-05-17:
Looks like a PVOutput.org connection error. I can not repair that. The only think I can do is improve the Grott error handling in this area.