johanmeijer/grott #298: Grott server: make error messages valid JSON with status code headers?

catch56, 2023-02-20:
I've been setting up some hass rest sensors to get the values of things via the grottserver rest API that aren't available in mqtt, like 'Stop charge SOC' and 'Battery first slot 1'. For a successful API response, grottserver puts this in JSON like { value: 1 }. However, when there's an error (timeouts seem quite common), the error comes back as a string message.

I'm wondering two things:

1. If the inverter request times out or similar, could the API response add a 503 header?
2. For any error, could it come back in the format { error: 'The error message' } so that it's valid JSON at all times - this would make it easier to account for those errors.

johanmeijer, 2023-02-21:
Sorry pushen the close button by accident.

egguy, 2023-03-05:
Also, if I can add my 2cts, could it have a numerical code? Like `{"code": 987, "message": "time out pushing value" }`.

This allows to have a stable error handling from an external point of view (like JS App), you can change the message without affecting the error handling on the client side (if the client uses the error codes).