johanmeijer/grott #301: Mqttretain doesn't seem to work properly

Caros2017, 2023-02-25:
Copied from the discussion section:

I am not sure what I am doing wrong. I am running the latest version (2.7.8). I have multiple(5+) connections to my `Mosquitto` MQTT broker and they all stay persistent. But I can't get Grott working to be persistent.

In docker-compose I use this:
```   environment:   
      - gblockcmd=False    
      - gnomqtt=False
      - gmqttretain=True
```
when I look at docker logs from Grott I see:
`        Pulled 'gmqttretain=True' from the environment
`
and
```
_MQTT:
        nomqtt                   False
        mqttip:                  **hidden**
        mqttport:                **hidden**
        mqtttopic:               energy/growatt
        mqttmtopic:              False
        mqttmtopicname:          energy/meter
        mqttinverterintopic:     False
        mqtttretain:             True
```
and
`         - Grott MQTT message retain enabled
`
But when I look at my Mosquitto logs I still see automatic disconnections from Grott:
```
1676205753: New connection from 172.21.0.1:54384 on port 1883.
1676205753: New client connected from 172.21.0.1:54384 as automatic (p2, c1, k60, u'grott').
1676205753: Client automatic disconnected.
1676205861: New connection from 172.21.0.1:55590 on port 1883.
1676205861: New client connected from 172.21.0.1:55590 as automatic (p2, c1, k60, u'grott').
1676205861: Client automatic disconnected.
1676205899: New connection from 172.21.0.1:56216 on port 1883.
1676205899: New client connected from 172.21.0.1:56216 as automatic (p2, c1, k60, u'grott').
1676205899: Client automatic disconnected.
1676205957: New connection from 172.21.0.1:56350 on port 1883.
1676205957: New client connected from 172.21.0.1:56350 as automatic (p2, c1, k60, u'grott').
1676205957: Client automatic disconnected.
```

Am I doing something wrong or is `mqtttretain` not working properly?

johanmeijer, 2023-02-26:
Grott is not designed at this moment to have a  persistent mqtt session. It works at fire and forget basis (publish only). 

This is easier to program and not really an issue if you send a message once per 5 minutes (original Growatt interval).

Now with grottserver we can achieve a much shorter interval it might be a consideration to implement a persistent mqtt connection. It needs some rethinking while a not working mqqtt connection does not affect the communication flow with growatt (the primary objective).

 If we create more and more persistency and statefullness within Grott I need to add all kind of error handling as well. I am not sure if this improve Grott (makes it at least more complex) and going think about it. 

Even with a 10 second interval I do not think that not having a persistent mqtt connection is a real issue.

Be aware retain is something different then persistency.

Caros2017, 2023-02-27:
> Grott is not designed at this moment to have a persistent mqtt session. It works at fire and forget basis (publish only).
>
> This is easier to program and not really an issue if you send a message [...]

Thanks for the comprehensive explanation. My request is for persistency indeed and not for retain. My bad!