johanmeijer/grott #303: Use grott with MIC 600 TL-X

Yogi4301, 2023-02-27:
Hello,

I set up grott in docker on my rpi in order to sent the data to pvoutput. The original data is still routes to Growatt but I don't see any data in pvoutput. Is it in general possible to use grott with the above mentioned inverter?

johanmeijer, 2023-02-27:
Somebody stated that a: MIC 2000TL-X  is working (see issue #60). It is with a container > 2.7 so I am not sure which you are running at this moment but try grott:beta (2.7.8). 

I still have the latest version to this (stable) version. 

Maybe we miss some information but the PV part should be oke (not the battery part). Lets first try to get it active and then see what info we miss ans see if we can find that.

Yogi4301, 2023-02-27:
Many thanks! I was using latest or 2.6.1g - I will try 2.7.8 as soon as the inverter switches back on. Do I have to configure a inverter type?

Yogi4301, 2023-02-28:
Many thanksfor pointing me to the beta - it is working with the mentioned inverter. 

But I have observerd, that at the time the growatt cloud skips some requests the data is also not shown in pvoutout.org. Is ist possible to transfer the data to pvoutput.org even if the growatt cloud is down?

At the corresponding time I find the following error messages in the log:

```
          - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.27.0.2', 57394), raddr=('47.91.67.66', 5279)>
         - Grott - grottproxy forward error :  [Errno 111] Connection refused
         - Grott - grottproxy forward error :  [Errno 111] Connection refused
         - Grott - grottproxy forward error :  [Errno 111] Connection refused
         - Grott - grottproxy forward error :  [Errno 111] Connection refused
         - Grott - grottproxy forward error :  [Errno 111] Connection refused
         - Grott - grottproxy forward error :  [Errno 111] Connection refused
         - Grott - grottproxy forward error :  [Errno 111] Connection refused
         - Grott - grottproxy forward error :  [Errno 111] Connection refused
         - Grott - grottproxy forward error :  [Errno 111] Connection refused
         - Grott - grottproxy forward error :  [Errno 111] Connection refused
         - Grott - grottproxy forward error :  [Errno 111] Connection refused
         - Grott - grottproxy forward error :  [Errno 111] Connection refused
         - Grott - grottproxy forward error :  [Errno 111] Connection refused
         - Grott - grottproxy forward error :  [Errno 111] Connection refused
```

Yogi4301, 2023-03-01:
Maybe I found the reason for the missing data in pvoutput - also today I see no data between 13:15 and 14:35 (marked on the right side). But I think I found the data at the beginning of the day (see the marked area at the left side). 
Is it possible to instruct grott to use the local time in the communicaton to pvoutput and mqtt?
[image]

johanmeijer, 2023-03-01:
Ah Ok. 

Normally the time/date of the shinewifi is being set at start up by Growatt. So you have to define your time zone their in the right way (website or phone app I think). 
Normally this time is used in the data record. Some shinewifi sticks have problems with keeping up with the time (HW error I think)
Wrong times are being inserted then. Grott can recognize this if it is a real invalid time format, but if is only of a couple of hours  of Grott will not detect this. 

To prevent this you can configure grott to use the server (container time).  

Requirement is that the timezone of the server / container is specified right. To enable that for docker you have to add the -e TZ=Your/timezone (e.g. -e TZ=Europe/Amsterdam) at the docker startup (docker compose). 

You also have to set the time=server parameter in the grott.ini (-e gtime="server" for docker). 

Now the server time is being used. 

The disadvantage of this that Grott has to ignore the buffered records (because we can not determine when this one has being sent). Buffered records are created when no connection with the growatt server is available.

johanmeijer, 2023-03-01:
The buffered records are sent to growatt, but grott will ignore them (not sent to mqtt, pvoutput) if time=server. while that make no sence (time wise). 

Sendig V7-V12 is not possible yet. I have plans to make this configurable in one of the next releases.

Yogi4301, 2023-03-07:
Works perfectly with time=server

closed 2023-03-07 by Yogi4301