johanmeijer/grott #306: Missing layour for 40ktl-3

Alferez, 2023-03-02:
Hello,

I have an installation with 3 40ktl-3 controlled with a ShineMaster.

I have configured the docker with the 2.7.8 and I have defined the 3 inverters inside the grott.ini, also I have configured the invtype as tl-3.

The problem I observe, is that it only returns me data from pv1 and pv2, but this inverter has 4 mppts.

I see this in verbose mode:
	 - Grott automatic protocol detection
	 - Grott data record length 585
	 - layout   :  T060204X
	 - no matching record layout found, try generic
	 - Record layout used :  T06NNNNX
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Possible Inverter serial UMHDXXXXXX
	 - Inverter serial not recognised - using inverter type default
	 - Growatt new layout processing
		 - decrypt       :  True
		 - offset        :  6
		 - record layout :  T06NNNNX
	 - Grott server date/time used
         ....

is possible create template for this inverter??

Best Regards.

johanmeijer, 2023-03-06:
Can you try it with Invtype=tl3 in grott.ini (or -e ginvtype="tl3" in docker run)?

Alferez, 2023-03-06:
I had in the configuration tl-3 as I put in the first message.
I changed it to tl3, but it is already night. Tomorrow I will tell the result when the sun rises.

Thank you.

johanmeijer, 2023-03-06:
Yes sorry. Tl3 should give information about 3 mppts at this moment if you als specify includeall = True. In grott.ini. I will see if I can add more mppts in the comming weeks.

Alferez, 2023-03-06:
I reconfigure with includeall.

_Generic:
	version:             	 2.7.8
	verbose:             	 True
	trace:               	 False
	config file:         	 grott.ini
	minrecl:             	 100
	decrypt:             	 True
	compat:              	 False
	invtype:             	 tl3
	invtypemap:          	 {}
	include_all:         	 True
	blockcmd:            	 False
	noipf:               	 False
	time:                	 server
	sendbuf:             	 True
	timezone:            	 local
	valueoffset:         	 6
	offset:              	 6
	inverterid:          	 automatic
	mode:                	 proxy
	grottip              	 0.0.0.0
	grottport            	 5279

And now i can see mpp3:
- Grott values retrieved:
		 -  datalogserial        :  XXXXXXXXX
		 -  pvserial             :  XXXXXXXX
		 -  recortype1           :  0
		 -  recortype2           :  124
		 -  pvstatus             :  1
		 -  pvpowerin            :  900.3
		 -  pv1voltage           :  364.4
		 -  pv1current           :  0.3
		 -  pv1watt              :  109.3
		 -  pv2voltage           :  617.0
		 -  pv2current           :  0.4
		 -  pv2watt              :  246.8
		 -  pv3voltage           :  688.0
		 -  pv3current           :  0.2
		 -  pv3watt              :  137.6
		 -  pvpowerout           :  886.8
		 -  pvfrequentie         :  50.0
		 -  pvgridvoltage        :  245.1
		 -  pvgridcurrent        :  1.6
		 -  pvgridpower          :  392.1
		 -  pvgridvoltage2       :  242.6
		 -  pvgridcurrent2       :  1.7
		 -  pvgridpower2         :  412.4
		 -  pvgridvoltage3       :  242.6
		 -  pvgridcurrent3       :  1.6
		 -  pvgridpower3         :  388.1
		 -  Vac_RS               :  422.6
		 -  Vac_ST               :  418.6
		 -  Vac_TR               :  423.3
		 -  pvenergytoday        :  58.0
		 -  pvenergytotal        :  4294.3
		 -  totworktime          :  379.2
		 -  epv1today            :  11.2
		 -  epv1total            :  969.8
		 -  epv2today            :  15.0
		 -  epv2total            :  1051.6
		 -  epvtotal             :  4073.3
		 -  pvtemperature        :  59.1
		 -  pvipmtemperature     :  54.2
		 -  pvboottemperature    :  52.7
		 -  temp4                :  0.0
		 -  uwBatVolt_DSP        :  0.0
		 -  pbusvolt             :  3573
		 -  nbusvolt             :  3573

I will watch for future updates if they include the 4th mppt option.

Best regards.

johanmeijer, 2023-03-14:
This is nice.  I will add the 4th mppt option soon!

Alferez, 2023-03-20:
I have reviewed the template code and created a new pull request.
https://github.com/johanmeijer/grott/pull/313

I have tested it on my installation and it works:

- date-time:  2023-03-20T15:08:32
 - Grott values retrieved:
	 -  datalogserial        :  XXXXXX
	 -  pvserial             :  YYYYYY
	 -  recortype1           :  0
	 -  recortype2           :  124
	 -  pvstatus             :  1
	 -  pvpowerin            :  6997.9
	 -  pv1voltage           :  752.7
	 -  pv1current           :  6.4
	 -  pv1watt              :  4817.2
	 -  pv2voltage           :  660.8
	 -  pv2current           :  1.3
	 -  pv2watt              :  859.0
	 -  pv3voltage           :  523.1
	 -  pv3current           :  1.0
	 -  pv3watt              :  523.1
	 -  pv4voltage           :  665.5
	 -  pv4current           :  1.2
	 -  pv4watt              :  798.6
	 -  pvpowerout           :  6951.8
	 -  pvfrequentie         :  50.0
	 -  pvgridvoltage        :  248.0
	 -  pvgridcurrent        :  9.5
	 -  pvgridpower          :  2356.0
	 -  pvgridvoltage2       :  247.9
	 -  pvgridcurrent2       :  9.3
	 -  pvgridpower2         :  2305.4
	 -  pvgridvoltage3       :  245.4
	 -  pvgridcurrent3       :  9.5
	 -  pvgridpower3         :  2331.3
	 -  Vac_RS               :  428.8
	 -  Vac_ST               :  426.4
	 -  Vac_TR               :  428.4
	 -  pvenergytoday        :  111.9
	 -  pvenergytotal        :  4675.7
	 -  totworktime          :  534.3
	 -  epv1today            :  27.1
	 -  epv1total            :  1310.6
	 -  epv2today            :  30.1
	 -  epv2total            :  1185.3
	 -  epvtotal             :  4487.0
	 -  pvtemperature        :  66.1
	 -  pvipmtemperature     :  74.9
	 -  pvboottemperature    :  68.4
	 -  temp4                :  0.0
	 -  uwBatVolt_DSP        :  0.0
	 -  pbusvolt             :  3705
	 -  nbusvolt             :  3726
 - MQTT jsonmsg: 
		 {"device": "YYYYYY", "time": "2023-03-20T15:08:32", "buffered": "yes",
		 "values": {"datalogserial": "XXXXXXX", "pvserial": "YYYYYYY",
		 "recortype1": 0, "recortype2": 124, "pvstatus": 1, "pvpowerin": 69979,
		 "pv1voltage": 7527, "pv1current": 64, "pv1watt": 48172, "pv2voltage": 6608,
		 "pv2current": 13, "pv2watt": 8590, "pv3voltage": 5231, "pv3current": 10,
		 "pv3watt": 5231, "pv4voltage": 6655, "pv4current": 12, "pv4watt": 7986,
		 "pvpowerout": 69518, "pvfrequentie": 5000, "pvgridvoltage": 2480,
		 "pvgridcurrent": 95, "pvgridpower": 23560, "pvgridvoltage2": 2479,
		 "pvgridcurrent2": 93, "pvgridpower2": 23054, "pvgridvoltage3": 2454,
		 "pvgridcurrent3": 95, "pvgridpower3": 23313, "Vac_RS": 4288, "Vac_ST": 4264,
		 "Vac_TR": 4284, "pvenergytoday": 1119, "pvenergytotal": 46757,
		 "totworktime": 3846700, "epv1today": 271, "epv1total": 13106, "epv2today":
		 301, "epv2total": 11853, "epvtotal": 44870, "pvtemperature": 661,
		 "pvipmtemperature": 749, "pvboottemperature": 684, "temp4": 0,
		 "uwBatVolt_DSP": 0, "pbusvolt": 3705, "nbusvolt": 3726}}