johanmeijer/grott #309: Docker compose mqtt ip etc

tonbor, 2023-03-10:
With docker compose containers in the same docker have NO default static ip address. Localhost of the host Ubuntu computer does not work. to point to each other. So containers just are connected by container name. That does not work with Grott. - gmqttip=mosquitto in the yaml file Grott still uses localhost. gmqttip=172.18.0.5, does work but the ip address of mosquitoo is not static. Please solve this issue.
My Growatt inverter has a lan connection with a static ip numberno wifi . Cannot find where to put this static ip number in the compose yaml. 
Mosquitto cannot handle message from Grot 1678452725: Client <unknown> disconnected due to malformed packet.

tonbor, 2023-03-12:
Well I have a tempenairety workaround for MQTT with the changing ip address, please adjust your implemantion. But there is no output to Mosquitto. WHERE did I go wrong? 
[_grott_logs (3).txt](https://github.com/johanmeijer/grott/files/10951362/_grott_logs.3.txt)

johanmeijer, 2023-03-13:
Ok. I have not done a lot with docker networks. Until now I always externalised the ports per container (mosquitto and grott) and then pointed to the server server ip address as the mqtt ip address.

uhavin, 2024-01-10:
Turns out you can set a fixed IP on a container. The way I got this working was to first create a docker network with a specific subnet, e.g.:
`sudo docker network create --subnet 10.0.8.0/24 mqtt-net` 

Then, in your docker compose yaml file, configure the network itself and configure the mosquitto service to use that network and give it the fixed ip. Both home assistant and grott will need to be connected to mqtt-net as well

```yaml
services:
  home-assistant:
    hostname: home-assistant
    ...
    networks:
      - "other-networks"  # if your home assistant also connects on another network, add it here.  
      - "mqtt-net"

  mosquitto:
    image: eclipse-mosquitto:2
    # ...
    networks:  # use a fixed ip on mqtt-net
      mqtt-net:
        ipv4_address: "10.0.8.100"
        aliases:
          - "mosquitto"
  grott:
    image: ledidobe/grott
    # ...
    env:
      - "gmqttip=10.0.8.100"
      # ...
    networks:
      - "mqtt-net"

networks:
  mqtt-net:
    name: mqtt-net
    external: true
```