johanmeijer/grott #310: New loglevels in grott

Yogi4301, 2023-03-13:
Hello,

due to my experience with grott (many thanks to @johanmeijer this is a great tool) I propose to implement new loglevels in grott:

level 2: the same like gverbose=True
level 1: the same like gverbose=False
level 0: only show an hourly summary of the messages send to growatt, MQTT, pvoutput.org and the blocked and non-blocked commands

Please also include a timestamp in the loggng for all three levels. Errors should also be displayed in alle levels

What do you think? From my point it is to much information in the log files for the normal operation today.

johanmeijer, 2023-03-13:
I agree that logging can be Improved. I will have a look at using log frameworks/standards.

I am not sure I want to implement summary messages. Until now the design of grott is to make it as stateless as possible. Keep track of counters and time will change this a bit. But I will take it in consideration. 

I have not missed timestamp because I log standard to the syslog and then time is included. But I can understand if logging to a file using timestamp can be of added value.

martin072, 2023-05-16:
Thanks for this tool, really appreciate it. 
Is there anyway to stop the logging to syslog? 
I am using a rPI with an sd card, so less writing, the better. Would only enable the log if stuff is not working. 

Thanks

typxxi, 2023-05-19:
> Is there anyway to stop the logging to syslog? I am using a rPI with an sd card, so less writing, the better. 

The better is to get a ssd asap cause you will otherwise loose data and time also cause the rpi with even a cheap ssd is a lot faster than any ssd , even the fastest Samsung one I had used before a complete crash with 0 chance to get anything back except the latest backup which was a 32 GB from over a week ago - in particualr also a month ago.

I have bought a good adapter which amazon had and was well known as a good boot device. I guess 10€ for that and 16€ for a fast, but used smaller SSD. 
Lot of different opportunities each cheaper and better than any SD card cause they dy sooner or later. Was not my first sd card I had lost (in a pi).

johanmeijer, 2023-05-19:
Yes a ssd for the rpi does not hurts I think. I am very happy using one.

You can stop logging by removing the -v in the ExecStart=-/usr/bin/python3 -u /home/pi/grott/grott.py -v line in the grott.service.

martin072, 2023-05-19:
Thanks for the ssd advice, understand this, but have to deal with the SD card for the moment. 
The -v option is not supplied, still it shows a lot of messages in the syslog.

rany2, 2023-05-19:
@martin072 when I used an RPI I just setup the system log to only save to RAM, worked like a charm. Now it is actually cheaper for me to just use a used laptop (and in fact it's better than any SBC at the same price point) :laughing: