johanmeijer/grott #316: Hosted Grott/AWS

thorstenfleischmann, 2023-03-25:
Would it be possible/advisable to run Grott on aws?

 I like to use Grott for pvoutput.org and change protection

rany2, 2023-03-25:
@thorstenfleischmann No it is no advisable. If you do, anyone could connect to it and spam your DB/submit on your behalf to PVOutput/etc. There is no authentication when it comes to inverter id so it could also be spoofed and your influxdb records for your inverter could be ruined.

tl;dr keep it on your local network only, if you're going to insist on doing it on AWS at least do it over a site-to-site VPN or setup firewall rules to allow connections from your network ONLY

thorstenfleischmann, 2023-03-25:
Thanks! How is the logger authenticated at Growatt? Don't we have the same problems with the Growatt servers?

rany2, 2023-03-25:
@thorstenfleischmann You provide Growatt with a check code that Growatt tied to your SN. Not very secure...

rany2, 2023-03-25:
@thorstenfleischmann There is no authentication actually... anyone could contact Growatt's servers, spoof your SN and cause you to have fake data

johanmeijer, 2023-03-25:
Luckily for us it is not that good protect so Grott is able intercept and decode the data. 

For Grott (grottserver) I can implement IP-filtering (Check your public IP address). Dat make a a little bit more secure if you want to run it in AWS (or other Public cloud).  I know this is not a total secure solution but it might help a little bit.

thorstenfleischmann, 2023-03-25:
> For Grott (grottserver) I can implement IP-filtering (Check your public IP address). Dat make a a little bit more secure if you want to run it in AWS (or other Public cloud). I know this is not a total secure solution but it might help a little bit.

I'm not very familiar with aws but I think they already have something like this built in ("security groups").

kreotak, 2023-03-27:
> I'm not very familiar with aws but I think they already have something like this built in ("security groups").
Yes, IP filtering should be possible on AWS side. Do you just want to avoid a "local" instance ? Or do you want to have one central instance for multiple inverter locations ?

thorstenfleischmann, 2023-03-27:
I thought of hosting it for me and maybe a friend.

But finally I concluded that aws is to expensive and I managed to run it natively on my low budget Synology.

But I like the idea of Grott as a service... The friend of mine has no Pi/Synology or anything running.

closed 2023-03-28 by thorstenfleischmann