johanmeijer/grott #317: Energy today/total is switched for MOD-3000TL3-XH

JohNan, 2023-03-27:
I have a MOD-3000TL3-XH with environment variable `ginvtype=tl3` set and it looks like these two values have been switched. According to https://server.growatt.com/ they should be the other way around.

```
grott-grott-1  |                 -  pvenergytoday        :  12.1
grott-grott-1  |                 -  pvenergytotal        :  0.3
```

Im also missing information about my battery such as SOC and how much it have charged/discharged.

```
grott-grott-1  |        - Growatt new layout processing
grott-grott-1  |                 - decrypt       :  True
grott-grott-1  |                 - offset        :  6
grott-grott-1  |                 - record layout :  T06NNNNXTL3
grott-grott-1  |
grott-grott-1  |         - Grott data record date/time processing started
grott-grott-1  |         - date-time:  2023-03-27T11:24:19
grott-grott-1  |         - Grott values retrieved:
grott-grott-1  |                 -  datalogserial        :  XGD6DA41JN
grott-grott-1  |                 -  pvserial             :  PZL0CGR02T
grott-grott-1  |                 -  pvstatus             :  1
grott-grott-1  |                 -  pvpowerin            :  131.9
grott-grott-1  |                 -  pv1voltage           :  219.9
grott-grott-1  |                 -  pv1current           :  0.6
grott-grott-1  |                 -  pv1watt              :  131.9
grott-grott-1  |                 -  pv2voltage           :  119.9
grott-grott-1  |                 -  pv2current           :  0.0
grott-grott-1  |                 -  pv2watt              :  0.0
grott-grott-1  |                 -  pvpowerout           :  26214.4
grott-grott-1  |                 -  pvfrequentie         :  9.3
grott-grott-1  |                 -  pvgridvoltage        :  406.3
grott-grott-1  |                 -  pvgridcurrent        :  401.2
grott-grott-1  |                 -  pvgridpower          :  26194739.2
grott-grott-1  |                 -  pvgridvoltage2       :  0.0
grott-grott-1  |                 -  pvgridcurrent2       :  0.0
grott-grott-1  |                 -  pvgridpower2         :  0.0
grott-grott-1  |                 -  pvgridvoltage3       :  10.9
grott-grott-1  |                 -  pvgridcurrent3       :  0.9
grott-grott-1  |                 -  pvgridpower3         :  85249228.8
grott-grott-1  |                 -  Vac_RS               :  0.5
grott-grott-1  |                 -  Vac_ST               :  0.0
grott-grott-1  |                 -  Vac_TR               :  7.8
grott-grott-1  |                 -  pvenergytoday        :  12.1
grott-grott-1  |                 -  pvenergytotal        :  0.3
grott-grott-1  |                 -  totworktime          :  0.0
grott-grott-1  |                 -  epv1today            :  0.0
grott-grott-1  |                 -  epv1total            :  0.0
grott-grott-1  |                 -  epv2today            :  0.0
grott-grott-1  |                 -  epv2total            :  0.0
grott-grott-1  |                 -  epvtotal             :  638.1
grott-grott-1  |                 -  pvtemperature        :  32.3
grott-grott-1  |                 -  pvipmtemperature     :  48.0
grott-grott-1  |                 -  pvboottemperature    :  31.5
grott-grott-1  |                 -  uwBatVolt_DSP        :  36.6
grott-grott-1  |                 -  pbusvolt             :  3201
grott-grott-1  |                 -  nbusvolt             :  3181
```
[image]

JohNan, 2023-03-27:
> That does not seem to be ok.
> 
> Looking into the documentation I have for TL3-X it refers to types MAX, MID, MAC, not MOD. In these documentation SOC is not included for these types.
> 

I understand. 

> But maybe I can create a different layout for it. Do you have the Growatt original Data (with the \x in it). That is easier to test for me.
> 

How can I get you the original data? 

> If I look into your record it confirms more with a "normal" TL-X inverter. I think specifying no invtype= will give better results (but no battery information, that is something I have to create).

I can try that and post the result.

johanmeijer, 2023-03-27:
The original data is at the log just before the part you sent above.

JohNan, 2023-03-27:
Here you go. I also removed the invtype setting. 

```
grott-grott-1  |         - Growatt original Data:
grott-grott-1  |                 \x00\xd9\x00\x06\x02\x41\x01\x04\x1f\x35\x2b\x41\x25\x35\x40\x76\x38\x21\x77
grott-grott-1  |                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
grott-grott-1  |                 \x3f\x2d\x2d\x44\x37\x00\x20\x5f\x45\x35\x74\x74\x47\x72\x6f\x77\x61\x74\x74
grott-grott-1  |                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x76\x77\x6f\x53\x60\x73\x75\x6a
grott-grott-1  |                 \xcc\x78\x73\x72\x6e\x77\x61\x74\x74\x45\x86\x6f\x77\x61\x74\x74\x47\x73\xec
grott-grott-1  |                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
grott-grott-1  |                 \x72\x6f\x77\x61\x74\x75\x13\x72\x6f\x73\xd8\x74\x74\x47\xf9\x7c\xfc\x68\x5f
grott-grott-1  |                 \x74\x43\x72\x6f\x74\xcb\x7d\x68\x47\x76\x6f\x77\x62\xd0\x7d\x40\x72\x6b\x77
grott-grott-1  |                 \x61\x77\xe8\x48\xa9\x60\xd4\x6e\xef\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
grott-grott-1  |                 \x6e\x23\x61\x7e\x5f\xa7\x72\x6f\x77\x69\x74\x74\x47\x23\x6f\x77\x61\x0a\x74
grott-grott-1  |                 \x47\x72\x67\x77\x61\x74\x0a\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
grott-grott-1  |                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
grott-grott-1  |                 \x77\x61\x74\x74\x4b\x72\x6f\x77\x0a\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
grott-grott-1  |                 \x72\x67\x77\x61\x74\x7a\x5d\x32\x6f\x77\x61\x74\x74\x47\x72\x6f\x6f\x9a\x75
grott-grott-1  |                 \x40\x46\xa5\x6e\x52\x61\x74\x75\x21\x7e\x14\x7b\xe0\x3a\x54\x47\x72\x6f\x77
grott-grott-1  |                 \x8b\x14\x74\x47\x72\x6f\x76\xf0\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
grott-grott-1  |                 \x6f\x77\x61\x74\x74\x47\x4e\x6f\x77\x61\x74\x74\x46\x72\x6f\x77\x61\x74\x74
grott-grott-1  |                 \x46\x26\x6f\x77\x61\x79\x78\x72\x7e\xde\x77\x61\x74\x7e\x47\x72\x6f\x3e\x61
grott-grott-1  |                 \x74\x74\x41\x72\x6f\x77\x3d\x74\x74\x47\x72\x6f\x77\x61\x75\x74\x47\x72\x04
grott-grott-1  |                 \x77\x61\x74\x79\x47\x72\x6f\x1c\x61\x74\x74\x46\x72\x6f\x77\x61\x74\x74\x47
grott-grott-1  |                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
grott-grott-1  |                 \x74\x47\x72\x6f\x77\x61\x53\x64\x47\x72\x77\x9a\x61\x74\x74\x47\x70\x6e\x77
grott-grott-1  |                 \x61\x74\x74\x79\x2d\x6f\x76\x61\x39\x6c\xab\x7e\x43\x77\x60\x74\x75\x46\x68
grott-grott-1  |                 \x6f\x77\x61\x74\x75\xb9\x72\x6f\x77\x61\x74\x74\x47\x3b\x6f\x77\x61\x28\x74
grott-grott-1  |                 \x47\x72\x6c\x7b\xa1\x74\x5b\x47\x73\x6f\x77\x61\x99\x74\xa6\x72\x6b\x77\x68
grott-grott-1  |                 \x74\x74\x47\x72\x6f\x76\x61\x74\x74\x74\x72\x5c\x77\x61\x74\x74\x47\x72\x6f
grott-grott-1  |                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x73\x61\x74\x74\x47
grott-grott-1  |                 \x72\x22\x49\x17\x74\x74\x46\xfe\x66\xb3\x68\xb0\x74\x47\x72\x0b\x35\xf1\x43
grott-grott-1  |                 \x24\x47\x72\x6f\x77\x7e\x15\x74\x47\x72\x6f\x7a\x69\x79\x71\x79\xf2\x6f\x77
grott-grott-1  |                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x7a\x45\x2a\x3a\x5f\x61\x74\x74\x47\x72
grott-grott-1  |                 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\xa1\xf0
grott-grott-1  |         - Grott automatic protocol detection
grott-grott-1  |         - Grott data record length 585
grott-grott-1  |         - layout   :  T060104X
grott-grott-1  |         - no matching record layout found, try generic
grott-grott-1  |         - Record layout used :  T06NNNNX
grott-grott-1  |         - Growatt data decrypted V2
grott-grott-1  |         - Grott Growatt data decrypted
grott-grott-1  |         - Growatt plain data:
grott-grott-1  |                 00d900060241010458474436444134314a4e00000000000000000000000000000000000000005
grott-grott-1  |                 05a4c30434752303254000000000000000000000000000000000000000017031b14121c020bb8
grott-grott-1  |                 0c3400010000000002f4000000000000018300000000000000000000000000000000000000000
grott-grott-1  |                 00000000154000004b90000008b138b092b0004000003aa091c0004000003a409070004000003
grott-grott-1  |                 9c0fdb0fa30f9b000000000000000000000154000a2be000000008000000510000007e0000000
grott-grott-1  |                 80000007e00000000000000000000000000000000000000000000000000000000000000000000
grott-grott-1  |                 000c0000006b0000000000000000000000080000000e1a40000000000000000018fb013401d70
grott-grott-1  |                 125000001660c7b0c814e2000000000ea60000000000191000000000000000000000000000000
grott-grott-1  |                 00003c00000000000100000000000001540000000d0c350cb10000000a0000004900000006000
grott-grott-1  |                 0005c00000000000000010000006b0000000d0000006b00000001000000000000000000000000
grott-grott-1  |                 00000000000000000000000000000000000000002710000018ed000000000201000000003e5f0
grott-grott-1  |                 001004d18ec0c2c00010001011a0000000001fe00000000000000490000005c000000030cc000
grott-grott-1  |                 2f0001000000ed00e100040009000000000001000000330033000000000000000000000000000
[... cut ...]
grott-grott-1  |                 -  pvenergytotal        :  8.1
grott-grott-1  |                 -  epvtotal             :  12.6
grott-grott-1  |                 -  epv1today            :  0.8
grott-grott-1  |                 -  epv1total            :  12.6
grott-grott-1  |                 -  epv2today            :  0.0
grott-grott-1  |                 -  epv2total            :  0.0
grott-grott-1  |                 -  pvtemperature        :  30.8
grott-grott-1  |                 -  pvipmtemperature     :  35.8
```

JohNan, 2023-03-27:
Here's a new screen shot from the site for reference

[image]

johanmeijer, 2023-04-13:
I am going to create a special layout for the MOD type inverters.

johanmeijer, 2023-08-13:
@JohNan 

I did not realized that the definition file for the MOD is already in the Grott 2.8.2. version. It seems to be the same as the MIN layout. 

If you specify invtype=min in the grott.ini [Generic] section you probably see usable data.  Can you please try this?

JohNan, 2023-08-14:
> @JohNan 
> 
> I did not realized that the definition file for the MOD is already in the Grott 2.8.2. version. It seems to be the same as the MIN layout. 
> 
> If you specify invtype=min in the grott.ini [Generic] section you probably see usable data.  Can you please try this?

You are right. Thank you for looking into it. 

I get a lot more data when using min type. And what I can tell it looks like correct values as well. But many fields are not mapped in the HA extension. 

I did however end up connecting a RPi directly to the Modbus interface to get faster updates. But the information may still be helpful for someone else.

Do you want me to post any logs or something before I revert back to the Modbus interface?

johanmeijer, 2023-08-15:
An original growatt data record from the grott log (with the /x format) will help with (future) debugging and formatting. 

Direct reading via the modbus interface is of course a good solution. I have not had time in the beginning to implement that and it was easier (and also fun) to use the existing hardware. Probably the setting / reading of the inverter register is a much more responsive then using an shinewifi, lan or link interface. 

An activity on my todo list is to hack the shinewifi module (with an ESP in it) and flash my own code on it. But there are more things or the list.

JohNan, 2023-08-16:
Modbus is more or less live, polling every 5 seconds. And setting values are a lot easier. I have a forked repo where I have adjusted it to work with my inverter: https://github.com/JohNan/Homeassistant-Growatt-Local-Modbus It's still a work in progress though :)

Here's the output
```
Inverter
grott-grott-1  |         - Growatt original Data:
grott-grott-1  |                 \x00\x07\x00\x06\x02\x41\x01\x04\x1f\x35\x2b\x41\x25\x35\x40\x76\x38\x21\x77
grott-grott-1  |                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
grott-grott-1  |                 \x3f\x2d\x2d\x44\x37\x00\x20\x5f\x45\x35\x74\x74\x47\x72\x6f\x77\x61\x74\x74
grott-grott-1  |                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x76\x7c\x64\x40\x58\x45\x75\x6a
grott-grott-1  |                 \xcc\x78\x73\x72\x6e\x77\x61\x66\x77\x51\xf6\x6f\x7f\x61\x74\x66\x44\x73\xa7
grott-grott-1  |                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
grott-grott-1  |                 \x72\x6f\x77\x61\x74\x64\x90\x72\x6f\x77\x12\x74\x74\x48\x1d\x7c\xf3\x68\x7d
grott-grott-1  |                 \x74\x4f\x72\x6f\x70\x5b\x7c\x83\x47\x75\x6f\x77\x67\x32\x7c\xa0\x72\x67\x77
grott-grott-1  |                 \x61\x73\x6b\x48\xd3\x60\x12\x6e\x2a\x74\x47\x72\x6f\x77\x61\x75\x72\x47\x72
grott-grott-1  |                 \x60\xa6\x60\x0d\x46\xc7\x72\x6f\x77\x6f\x74\x74\x87\x09\x6f\x77\xa1\xc2\x74
grott-grott-1  |                 \x47\x72\x6b\x77\x61\xb4\xc1\x47\x72\x6f\x77\x61\x74\x74\x46\x72\x6f\x77\x61
grott-grott-1  |                 \x74\x74\x47\x72\x6f\x77\x61\x5b\x74\x47\x7f\x2d\x77\x61\x74\x74\x47\x72\xe8
grott-grott-1  |                 \x5d\x61\x74\x74\x50\x72\x6f\x31\x0b\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
grott-grott-1  |                 \x72\x6b\x77\x61\x74\x7a\xb8\x88\x6f\x77\x61\x74\x74\x47\x72\x6f\x6f\xc1\x75
grott-grott-1  |                 \x12\x46\x8a\x6e\x2c\x61\x74\x75\xcb\x7e\x31\x7b\x23\x3a\x54\x47\x74\x6f\x77
grott-grott-1  |                 \x8b\x14\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
grott-grott-1  |                 \x6f\x77\x61\x74\x74\x47\x4e\x6f\x77\x61\x74\x74\x46\x72\x6f\x77\x61\x74\x74
grott-grott-1  |                 \x48\xa3\x6f\x77\x61\x64\x78\x72\x7e\xde\x77\x61\x74\x79\x47\x72\x76\xa3\x61
grott-grott-1  |                 \x74\x74\x6e\x72\x6f\x6d\x59\x74\x74\x47\x57\x6f\x77\x64\xf8\x74\x47\xb4\x1d
grott-grott-1  |                 \x77\x61\x74\x7b\x47\x72\x50\x30\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
grott-grott-1  |                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
grott-grott-1  |                 \x74\x47\x72\x6f\x77\x61\x53\x64\x47\x72\x77\xd0\x61\x74\x74\x4b\x73\x6e\x77
grott-grott-1  |                 \x61\x74\x74\x79\x11\x6f\x75\x61\x3d\x6c\xe1\x7e\x65\x77\x61\x74\x74\x46\x48
grott-grott-1  |                 \x6f\x77\x61\x74\x74\x47\x72\x6f\x76\x57\x74\x74\x5e\xa6\x6f\x77\x7b\x4c\x74
grott-grott-1  |                 \x47\x72\x6c\x7b\xfd\x74\x6d\x47\x42\x6f\x77\x60\x6a\x75\x4d\x72\x6d\x77\x68
grott-grott-1  |                 \x74\x74\x47\x72\x6f\x76\x61\x74\x74\x6a\x72\x42\x77\x61\x74\x74\x47\x72\x6f
grott-grott-1  |                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x73\x61\x74\x74\x47
grott-grott-1  |                 \x72\x26\x49\x17\x74\x6a\x46\xdb\x66\xb3\x68\xb0\x74\x47\x72\x0b\x35\xf1\x43
grott-grott-1  |                 \x24\x47\x72\x6f\x77\x7e\x16\x74\x47\x72\x6f\x7a\x65\x79\x74\x79\x1e\x6f\x77
grott-grott-1  |                 \x61\x74\x74\x45\x72\x63\x77\x61\x74\x7a\x45\x2a\x3a\x5f\x61\x74\x74\x47\x72
grott-grott-1  |                 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\xbe\x00

Datalogger
grott-grott-1  |         - Growatt original Data:
grott-grott-1  |                 \x00\x07\x00\x06\x01\x0e\x01\x20\x1f\x35\x2b\x41\x25\x35\x40\x76\x38\x21\x77
grott-grott-1  |                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
grott-grott-1  |                 \x3f\x2d\x2d\x44\x37\x00\x20\x5f\x45\x35\x74\x74\x47\x72\x6f\x77\x61\x74\x74
grott-grott-1  |                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x76\x7c\x64\x40\x58\x41\x77\xa9
grott-grott-1  |                 \x74\x74\x47\xfc\x6f\x77\x69\x8e\x74\x47\x7b\x6f\x77\x61\x7d\x74\x47\x72\x6f
grott-grott-1  |                 \x71\x61\x74\x74\x41\x72\x6f\x77\x6f\x8b\x8b\xbc\xbc\x90\x88\x9d\x08\x74\x47
grott-grott-1  |                 \x74\xb4\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x8d\x90\x8b\xe9\x8b
grott-grott-1  |                 \x8b\xbb\x3f\x90\x88\x9a\x3d\x8b\xb8\x8f\x24\x88\x9e\x89\xd0\x47\x72\x6d\x64
grott-grott-1  |                 \x9e\x8b\x8b\x63\x72\x6f\x77\x61\x8b\x8b\xb3\x6f\x90\x88\x9e\x8f\x74\x47\x73
grott-grott-1  |                 \x9c\x77\x61\x7b\xe5\x47\x72\x60\xe1\x61\x74\x7b\xd6\x72\x6f\x68\xd9\x74\x74
grott-grott-1  |                 \xd2\x4e\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
grott-grott-1  |                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
grott-grott-1  |                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
grott-grott-1  |                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
grott-grott-1  |                 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6b\x66
```