johanmeijer/grott #319: PVOutput Voltage - Use PV voltage instead of grid voltage

thorstenfleischmann, 2023-03-29:
Currently the "gridvoltage" is sent to pvoutput as voltage.. which is more or less a constant. I think the pv voltage would be better

Yogi4301, 2023-03-31:
Also the pv-voltage is nearly constant. In case of the gridvoltage you can see how the inverter pushes its energy into the grid by increasing the voltage. Maybe the pv-voltage can be added as additional value. As I know there are some plans in that point.

johanmeijer, 2023-03-31:
I used grid voltage while that was interesting for my installation. I have an older pv installation that is very sensitive for voltage level. At 237 volt it stops working.

With one of the next updates I will make all pvoutput Vx values configurable. So you can choose your own values to be registered by pvoutput.