johanmeijer/grott #321: SPH no export to grid values

BarmanMaz, 2023-04-01:
Hello , 
I have just instllaed the grott integration ( wonderfull job) , I'm using the add-on for ha, so the extension are buil-in into grott_ha.py.

My intallation is with two SPH 8000 inverter and one shine master via LAN

All fine expect for some sensors data still report zeros -> Export to grid power and energy.

[image]

[image]

The add-on is working fine, I can see the value updated to growatt web page,  where i can also found the return to grid, but on my sensor nothing, still zero's.

[image]

Can someone please give me an hints to how solve it?

Many thanks

BarmanMaz, 2023-04-02:
Hi All,
I have done more research today.
this is what i got:
 - MQTT jsonmsg: 
			 {"device": "KHH0B23036", "time": "2023-04-02T14:40:45", "buffered": "no",
			 "values": {"datalogserial": "DXD6BJ300P", "pvserial": "KHH0B23036",
			 "pvstatus": 5, "pvpowerin": 8300, "pv1voltage": 3323, "pv1current": 11,
			 "pv1watt": 3913, "pv2voltage": 3670, "pv2current": 11, "pv2watt": 4397,
			 "pvpowerout": 8573, "pvfrequentie": 4998, "pvgridvoltage": 3969,
			 "pvgridcurrent": 12, "pvgridpower": 9462, "pvgridvoltage2": 4017,
			 "pvgridcurrent2": 13, "pvgridpower2": 0, "pvgridvoltage3": 3986,
			 "pvgridcurrent3": 13, "pvgridpower3": 0, "totworktime": 70474883,
			 "eactoday": 149, "pvenergytoday": 149, "eactotal": 111636, "epvtotal":
			 110419, "epv1today": 99, "epv1total": 52458, "epv2today": 110, "epv2total":
			 57961, "pvtemperature": 673, "pvipmtemperature": 668, "pvboosttemp": 625,
			 "bat_dsp": 3711, "eacharge_today": 12, "eacharge_total": 13066,
			 "batterytype": 1, "uwsysworkmode": 5, "systemfaultword0": 0,
			 "systemfaultword1": 0, "systemfaultword2": 0, "systemfaultword3": 0,
			 "systemfaultword4": 0, "systemfaultword5": 0, "systemfaultword6": 0,
			 "systemfaultword7": 256, "pdischarge1": 0, "p1charge1": 0, "vbat": 3710,
			 "SOC": 96, "pactouserr": 0, "pactousertot": 0, "pactogridr": 0,
			 "pactogridtot": 0, "plocaloadr": 8200, "plocaloadtot": 8200, "spdspstatus":
			 5, "spbusvolt": 7063, "etouser_tod": 0, "etouser_tot": 0, "etogrid_tod": 0,
			 "etogrid_tot": 0, "edischarge1_tod": 44, "edischarge1_tot": 41572,
			 "eharge1_tod": 91, "eharge1_tot": 46503, "elocalload_tod": 165,
			 "elocalload_tot": 111183}}

I found this:   "pactogridr": 0,  ( power to grid) is zero even if growat web display this

[image]

also, "pactouserr": 0, ( power to user/home)   even if i have load at home.
 
what i'm doing wrong?
please help me
Many thanks
Roberto

johanmeijer, 2023-04-02:
Do you the hex record information (growatt original data) for me. Lets see if the information is in the record.

BarmanMaz, 2023-04-02:
Thanks for the reply, 
I'm a newbie, how i can dump
the hex?
don't know how to.
KR

BarmanMaz, 2023-04-03:
Hi,
Following the hex infos.

- Growatt original Data:
		 \x00\x01\x00\x06\x01\x20\x22\x1b\x03\x2a\x2b\x41\x23\x3e\x47\x77\x42\x3f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x34\x76\x70\x77\x4e\x62\x45\x77\x97\x46\x4c\x7f\x40\x5b\x59\x57\x4c\x58
		 \x7f\x47\x5e\x44\x4f\x42\x4c\x6b\x44\x58\x43\x4f\x46\x58\x6a\x46\x5b\x46\x4f
		 \x41\x58\x6a\x40\x57\x47\x4f\x4d\x58\x6a\x43\x5a\x44\x4f\x44\x58\x6a\x43\x5b
		 \x43\x4f\x47\x58\x6a\x40\x5f\x4e\x4f\x43\x58\x71\x4b\x5f\x59\x51\x58\x40\x71
		 \x41\x41\x41\x4d\x47\x40\x7f\x5c\x5a\x5b\x51\x5a\x4d\x6b\x5f\x5f\x59\x58\x58
		 \x59\x77\x5c\x58\x5b\x53\x46\x4d\x69\x41\x43\x45\x52\x44\x5a\x7e\x5e\x5d\x45
		 \x58\x5a\x43\x6b\x41\x41\x46\x4d\x46\x5a\x76\x5e\x5e\x59\x59\x58\x59\x73\x4a
		 \x41\x4e\x4d\x41\x44\x7e\x5c\x5b\x5b\x4c\x41\x44\x70\x5c\x5f\x5b\x4c\x44\x5a
		 \x77\x5e\x5b\x4e\x4f\x4d\x58\x75\x43\x59\x45\x56\x5a\x44\x6b\x45\x5e\x4e\x56
		 \x5a\x42\x6b\x41\x59\x44\x56\x5a\x40\x6b\x46\x57\x40\x57\x5a\x46\x6b\x41\x5f
		 \x47\x54\x41\x5a\x70\x5e\x5c\x43\x56\x4c\x5a\x70\x5e\x5d\x42\x55\x46\x5a\x73
		 \x5e\x5d\x43\x58\x46\x5a\x72\x5e\x5f\x59\x51\x58\x44\x69\x42\x43\x47\x4f\x44
		 \x58\x74\x45\x5c\x44\x59\x5a\x47\x70\x5e\x5c\x4e\x58\x5a\x42\x6b\x41\x56\x4f
		 \x4f\x40\x58\x74\x4b\x56\x59\x50\x58\x18\x55
	 - Grott automatic protocol detection
	 - Grott data record length 296
	 - layout   :  T06221b
	 - no matching record layout found, try generic
	 - Record layout used :  T06221b
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
- Growatt plain data:
		 000100060120221b44584436424a3330305000000000000000000000000000000000000000000
		 04317040309102a00f632383832342e36382c383531332e36382c3637342e322c2d3434312e35
		 2c2d3238302e392c2d3135332e302c2d3134342e332c2d3230392e372c3639302e302c3436332
		 e362c3334382e352c302e392c2d302e392c2d302e372c3232392e332c3233302e392c3232392e
		 372c332e312c322e312c312e382c2d34382e392c3530392e342c2d3530372e302c2d302e302c3
		 4392e392c32313632372e302c373139372e362c333633372e342c343837362e322c3330303535
		 2e372c333437382e372c323534322e342c323439322e352c302e302c302e302c302e302c33373
		 333382e33372c3339392e362c3339382e342c3339392e312c6c12
	 - Grott data ack record or data record not defined no processing done
	 - Growatt packet received:
		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.30.33.8', 5279), raddr=('192.168.1.198', 56917)>
	 - Data less then minimum record length, data not processed
	 - ('172.30.32.2', 37622) has connected
	 - ('172.30.32.2', 37622) has disconnected

I Have another question: why i have this " - Data less then minimum record length, data not processed "

Many thaks for the help

johanmeijer, 2023-04-03:
Yes that is the type of information I need, but ..... 

You send me the information of a SDM630 meter I think. Also interesting, especially why is Grott not processing this one. I like to know your datalogger type (shinewifi, shinelan, shinelink, 4g) while that is also a new one for me (at least number in the record). 

What I also need is the information you sent but then with a record that is processed by grott (with decoded mqtt information)

johanmeijer, 2023-04-03:
If I proces the record you sent as a SDM630, I see this information: 

 {"device": "SDM630", "time": "2023-04-03T20:33:17", "buffered": "no",
                         "values": {"datalogserial": "DXD6BJ300P", "active_energy": "28824.68",
                         "reactive_energy": "8513.68", "activePowerL1": "674.2", "activePowerL2":
                         "-441.5", "activePowerL3": "-280.9", "reactivePowerL1": "-153.0",
                         "reactivePowerL2": "-144.3", "reactivePowerL3": "-209.7", "apperentPowerL1":
                         "690.0", "apperentPowerL2": "463.6", "apperentPowerL3": "348.5",
                         "powerFactorL1": "0.9", "powerFactorL2": "-0.9", "powerFactorL3": "-0.7",
                         "voltageL1": "229.3", "voltageL2": "230.9", "voltageL3": "229.7",
                         "currentL1": "3.1", "currentL2": "2.1", "currentL3": "1.8", "power":
                         "-48.9", "active_power": 0, "reverse_active_power": "-48.9",
                         "apparent_power": "509.4", "reactive_power": "-507.0", "power_factor":
                         "-0.0", "frequency": "49.9", "posiActivePower": "21627.0",
                         "reverActivePower": "7197.6", "posiReactivePower": "3637.4",
                         "reverReactivePower": "4876.2", "apparentEnergy": "30055.7",
                         "totalActiveEnergyL1": "3478.7", "totalActiveEnergyL2": "2542.4",
                         "totalActiveEnergyL3": "2492.5", "totalRectiveEnergyL1": "0.0",
                         "totalRectiveEnergyL2": "0.0", "totalRectiveEnergyL3": "0.0",
                         "total_energy": "37338.37", "l1Voltage2": "399.6", "l2Voltage3": "398.4",
                         "l3Voltage1": "399.1", "device": "SDM630"}}

I am not sure if that is in line with what you expect. 

I also do not know if this information should be in the SPH record.

BarmanMaz, 2023-04-03:
Thanks a lot for the support.
I'm struggling to have the entire logs, hex plus + parsing data.
I'm using the grott add-on in my hassio HA, there is a way to export the complete log into a file? Can you please tell me how?
The data i posted i git it from the log page on HA, don't understand why they look wrong.
I have two sph8000 hybrid inverters 2 battery pack and a Smart Energy Manager SEM-E via LAN (https://www.ginverter.com/products/sem-sem-e-sem-d)
What basically is missing are all the value is the sensors related to the grid, import/export.
Thanks again.
Rob

BarmanMaz, 2023-04-04:
Hi,

This is the data logger:

[image]

[image]

The inverter are : 

[image]

This is the logs: 

[grott-log.txt](https://github.com/johanmeijer/grott/files/11150120/grott-log.txt)

thanks again for the support really appreciate.
Rob

johanmeijer, 2023-04-13:
You have a nice configuration ;) 

If you have only 1 sph the import / export values are provided in the SPH record (and I think also processed by the HA add on). 
In your configuration I think growatt (the datalogger in your case the shine master(  does not sent the export / import values with the SPH record but with the SDM630 records. Seems logical to me. 

Grott can handle these SDM630 but is not able to do this with the shinemaster at this moment. I will change that, that is no a big change but I need some time (little bit busy at work). At the end you will then receive 3 mqtt records 1 for each sph and 1 for the SDM630. We have to see how and of the HA add on can handle this. We need the creater of the add on for this I think. 

But first thing first you need the SDM630 record to get the information you need.

BarmanMaz, 2023-04-15:
Hi Johan,
Thanks, I was awaiting your answer, as, for me, you did the best and clever solution to integrate grovatt into HA.
I will wait your update, what about the logs? Do you need more of it?
Do you have a "buy me a coffe" link?
Thanks

johanmeijer, 2023-04-24:
Can you please put the layout file t06221b.json form the examples library in your grott main directory? 

See: 
https://github.com/johanmeijer/grott/blob/Beta-(2.8.x)/examples/Record%20Layout/T06221b.json 

If everything works right this will process your SDM630 records and generate MQTT messages for this. Can you very form the log if this what  you expected. 

I think we have to do some additional work to get this information in HA with the plugin/extension. @egguy can you please help with this? 

If you want to buy a coffee the link is at the readme or use this one: 
https://www.paypal.com/donate?business=RQFS46F9JTESQ&item_name=Grott+&currency_code=EUR

BarmanMaz, 2023-04-24:
Hi Johan,
thanks, grott main directory you mean the docker ones or the grott directory where grott.ini is located?

johanmeijer, 2023-04-25:
Ah docker. Can you put things in your docker container directory your self (or put in a reference to a file outside in the run command?).

I can also create a new image, but that will take a couple of days,

BarmanMaz, 2023-04-26:
Hi Johan, here I am.
please see the attached log made with the egguy add-on  build with your latest beta 2.8.x , it seems nothing changed.
Please let me know if i can do somethings.
Many thanks
[grott log 2.txt](https://github.com/johanmeijer/grott/files/11330196/grott.log.2.txt)

egguy, 2023-04-26:
Hello,
The extension should handle multiples inverters + the SDM.

The extension works like this:
[image]

The JSON I add to the JSON the last push date to help tracking if the intervention work

As for the SDM630, I didn't fully map the measures to the units. So, it's still treated as text values in HA.

BarmanMaz, 2023-04-26:
Hello,
thanks a lot for the clear explanation, will you be able to generate a new version with this improvement into?

Many thanks

BarmanMaz, 2023-05-08:
Hello, 
sorry for pushing, do you have any news of this task?
Many thanks