johanmeijer/grott #322: Somethings wrong with pvoutput after noon

thorstenfleischmann, 2023-04-04:
I am not sure what is happening, but something is wrong with PVOutput. I have the same problem every day.

Until 13:00 everything seems to be fine:

[image]

Next screenshot is from 13:45:

[image]

Now there is an entry at 7:15 which wasn't there before ... and also the record values seem to be from noon.

Let me know if you need logs or anything

Yogi4301, 2023-04-05:
Thats a bug in the ShineWifi-X I also had before - using the server time solved the problem for me. You can use the option gtime=server

thorstenfleischmann, 2023-04-06:
> Thats a bug in the ShineWifi-X I also had before ([#303 (comment)](https://github.com/johanmeijer/grott/issues/303#issuecomment-1450366867) using the server time solved the problem for me. You can use the option gtime=server - also you have to set the correct timezone in the server or container.

Thanks. Does this workaround affect the data at the Growatt servers? Only pvoutput seems to be corrupted. The data at growatt/shineApp is correct

thorstenfleischmann, 2023-04-06:
Sorry, I think I found the answer in the linked issue:

> The buffered records are sent to growatt, but grott will ignore them (not sent to mqtt, pvoutput) if time=server. 

I'll have a try.