johanmeijer/grott #325: mqttinverterintopic = True not working?

DraneLixX, 2023-04-09:


TheAfrogo, 2023-04-11:
I have the same issue.
mqttinverterintopic = true in the mqtt area in the grott.ini doe's not work.

DraneLixX, 2023-04-11:
> I have the same issue. mqttinverterintopic = true in the mqtt area in the grott.ini doe's not work.

Got it fixed, didn't even want to post this anymore that's why there is no description, sorry for that.
The Docs are pretty messy and need a rewrite, but you need to remove the "mqtt" before "nverterintopic" so it should look like this = inverterintopic = True
I was just about to give up when it got to me when you look in the logs when it starts up it list all possible settings in there its described as mqttinverterintopic = true a better reference is just the IP Setting when starting it debugs it as this:

```
_MQTT:
        nomqtt                   False
        mqttip:                  127.0.0.1
        mqttport:                1883
        mqtttopic:               energy/growatt
        mqttmtopic:              True
        mqttmtopicname:          energy/growatt
        mqttinverterintopic:     True
        mqtttretain:             False
        mqtttauth:               True
        mqttuser:                grott_proxy
        mqttpsw:                 **secret**
```
but in the grott.ini it look's like this:
```
[MQTT]
# Mqtt parameters definitions
# Be aware nomqtt = True means no MQTT processing will be done!!!!!!

nomqtt = False
ip = 127.0.0.1
port = 1883
mtopic = True
mtopicname = energy/growatt
inverterintopic = True
tretain = True
auth = True
user = grott_proxy
password = **secret**
```

johanmeijer, 2023-04-13:
Ok this seems to be a bug. Sorry for that. 

But is is hobby project ( i earn no money with it) so please have a little bit patience with me. If you found a bug please let me (kindly) know and I will try to fix it.

dranelixx, 2023-04-13:
> Ok this seems to be a bug. Sorry for that. 
> 
> But is is hobby project ( i earn no money with it) so please have a little bit patience with me. If you found a bug please let me (kindly) know and I will try to fix it. 

Im sorry i was so disrespectful you did a really great job im verry thankful for your solution.
Till now it works flawless for me 🤩

johanmeijer, 2023-04-24:
I re-read this issue and look in the coding and think it is only a documentation flaw (lack of documentation). 

inverterintopic = True/False is what needs to be specified in the grott.ini.  Internally grott it is translated to mqttinverterintopic but that is for alignment with the environmental variable (that is gmqttinveterintopic). It very hard to keep consistent with naming especially if you already work a couple of years changing an application. 

I will update the documentation (not my favorite part of  the project).

dranelixx, 2023-04-24:
From my point of view, it would be enough for now to correct the [Multi Inverter Page](https://github.com/johanmeijer/grott/wiki/Multi-Inverter-support)

because there, it's documented as `mqttinverterintopic="True"` when it should be `inverterintopic="True"`

I don't know if it's the same for docker, but I would think so then it would need to be changed from `-e gmqttinverterintopic="True"` to  `-e inverterintopic="True"`

Also, maybe try to feed ChatGPT4 with your documentation, maybe you can save a lot of time? Cuz GPT could rewrite it for you, :) it's pretty capable.

Again, I'm very thankful for your work since I set it up it work's flawless.