johanmeijer/grott #327: Incorrect discharge values

Jense35J, 2023-04-18:
My sensor input from the discharge of the battery is in watts but it is shown in Kw, I have tried to change the units to W but that did not help.
anyone knows a fix to that?

[image]

[image]

johanmeijer, 2023-04-24:
Is this a sensor setting in HA? @egguy do you recognize this?

Jense35J, 2023-04-24:
It is the sensors I get from My growatt inverter, and I have tried to change the sensor settings from Kw to W but the value that are showing in kw 🤔