johanmeijer/grott #333: incorrect values grott-server

T1gg1, 2023-04-28:
Hello,

I started a grott-server in a Dockercontainer and that works fine BUT.....
I started the container and it shows me onetime the correct output but the following updates shows not the correct output.
for example:
First start pvpowerout: 220watt
next update pvpowerout: 20watt.......(proxymode) in the growatt app it shows 225watt and on the inverter the same....225watt
how can that be?

T1gg1, 2023-04-28:
I guess they were statusmessages because they come nearly seconds. Now the inverter send every 5 minutes a message and that is correct.

closed 2023-04-28 by T1gg1

johanmeijer, 2023-04-28:
@T1gg1 it were probably "buffered"(Historical) records. IF I looked at the time it says it was a record from 07:46...

T1gg1, 2023-04-28:
OMG THank you :D yeah I should read exactly :D