johanmeijer/grott #334: API call Grott-server

T1gg1, 2023-04-28:
Heyho!
If I put it:
XXX.XXX.XXX.XXX:5782/datalogger?datalogger=XXXXXXXXXX&command=register&register=03&value=50

Then the grott-server-console shows "- Grotthttpserver - http response send:  400 text/body b'invalid datalogger id ' "

Can anybody help?
BR

johanmeijer, 2023-04-28:
Works fine for me (with postman). What do want to change with register 3? Seems to be 0 for my both inverters.

T1gg1, 2023-04-28:
https://forum.iobroker.net/assets/uploads/files/1550402978050-growatt-pv-inverter-modbus-rs485-rtu-protocol-v3.05-%E5%AE%A2%E6%88%B7%E7%89%88.pdf

in this link, that is the max output percentage
Do you have another list?

T1gg1, 2023-04-30:
@johanmeijer how do you use that? I thougth I have to use the server with that over the browser. Where I can find the right register?

johanmeijer, 2023-04-30:
You have to sets of register, the datalogger and the inverter. 

The datalogger registers say something about the connection with the internet and how the data is sent to the outside world. 
The dataloggers registers are not very good documented. You find some here in the grottserver discussion thread (#98), 

The inverter register are documented in the growatt documentation as the holding registers. And to be honest they are not all documented well either for all inverters. You find also more in the grottserver discussion thread. 

I am not sure if the document you pointing to is the right document for your inverter. This more the document for the "older" type of inverters (o.a. TL-S)
The newer type there is a othor document (eg: https://www.scribd.com/document/489508943/Growatt-Inverter-Modbus-RTU-Protocol-II-V1-20-English). 
For the SPH type inverters there is also a n other document.

T1gg1, 2023-05-18:
@johanmeijer 
Okay I test it but still I got the following error:
<html><body>

Grotthttpserver - http response send:  400 text/body b'invalid datalogger id '
--

</body>
</html>

johanmeijer, 2023-05-18:
Which tool do you use to sent the httprequest?

T1gg1, 2023-05-18:
my browser

johanmeijer, 2023-05-18:
First of all you can not set registers via the browser (browser get a get command and you need to use a put httprequest. 

Having said that the command as you show should not give an invalid logger id. 

If you use the command: 

XXX.XXX.XXX.XXX:5782/datalogger 

What do you see? Do you use the first device id for data datalogger command (for me it starts with JPC)? 
Example: 

     {"JPCxxxxxxx": {"ip": "192.168.0.206", "port": 52684, "protocol": "06", "QNBxxxxxxx": {"inverterno": "01", "power": 0}}

T1gg1, 2023-05-18:
{}

that is my response ;)

T1gg1, 2023-05-18:
yes. In the server log there I can see the data from the datalogger. I can see all on my mqtt server as well.

T1gg1, 2023-05-18:
@johanmeijer Do you know what Modbus Register 03 is? :D 4 is the update intervall
I wrote 100 to 03 and I don´t know if its right or if something is broken now.
I don´t find any updated version

T1gg1, 2023-05-18:
Or can I reset only the register 3?

T1gg1, 2023-05-18:
I had panic and did a factoryreset on both😅
Register 3 is still 100 so I think that is unused, too
Thank you

T1gg1, 2023-05-19:
@johanmeijer 
I found out that register 123 of my inverter is the exportlimitation.
Can I set it via the Api or do I need the right register from the datalogger?

johanmeijer, 2023-05-19:
You can set it directly in the inverter. Vua due grott api inverter request.

Be aware the inverter interface is a little bit instable (not a Grott thing, also true for doing it via growatt website). Sometimes you have to try it multiple times.

T1gg1, 2023-05-19:
I set the exportlimitation via CT to 40% but the inverter produces the full power. Do you know why?
Or should I limit the active power? Then it is register 03 as I mentioned😅🙈