johanmeijer/grott #335: Docker image multiplatform support (not only AMD64)

SavGRY, 2023-05-02:
Hi everyone, 
I was having problems reading data from my Growatt Server to a Libre Computer (Version: S905X) with RaspBian as OS, but the
service Grott only supports **amd64** platforms and mine happens to be **arm64/v8.**
Right now the way things stand is that in the docker-compose.yaml (the one that comes in `./docker/docker-compose.yaml` ) starts to to build from the **ledidobe/grott** image. This image supports only AMD64 architecture and this problem (photo below) comes out when tying to docker compose up:
[image]

I came up with changing the docker-compose.yaml file in the following way:
```yaml
version: '3.3'
services:
    grott:
        build: .
        container_name: grott
        restart: unless-stopped
        ports:
            - "5279:5279"
        volumes:
            - './grott.ini:/app/grott.ini'
```
So to resolve this problem now the compose file will find the Dockerfile to build. Mind I'm posting the Dockerfile for my project structure folder, so your mileage may vary

```yaml
FROM python:3.7-slim

ENV LANG en_US.UTF-8
ENV LC_ALL en_US.UTF-8

# Run updates
RUN apt-get clean && apt-get update && apt-get upgrade -y

# Set the locale
RUN apt-get install -y locales && locale-gen en_US.UTF-8

#Install required python packages 
RUN pip install paho-mqtt
RUN pip install requests 
RUN pip install influxdb
RUN pip install influxdb-client 

COPY ./grott/grott.py /app/grott.py
COPY ./grott/grottconf.py /app/grottconf.py
COPY ./grott/grottdata.py /app/grottdata.py
COPY ./grott/grottproxy.py /app/grottproxy.py
COPY ./grott/grottsniffer.py /app/grottsniffer.py
COPY grott.ini /app/grott.ini

WORKDIR /app
CMD ["python", "-u", "grott.py", "-v", "-t"]
```

So in my configuration I moved the dockerfile that was inside the docker folder (its name should be changed from dockerfile to Dockerfile with capital "D" for best practice and conventional naming) in the project root folder, and then all the grott related scripts and file inside a grott folder:

```
grott-project/
├─ docker-compose.yaml
├─ Dockerfile
├─ get-docker.sh
├─ grott.ini
└─ grott/
    ├─ docker/
    │   └─ ... all the file from the existing repo
    ├─ examples/
    │   └─ ... all the files from the existing repo
    ├─ grottconf.py
    ├─ grottdata.py
    ├─ grott.ini
    ├─ grottproxy.py
    ├─ grott.py
    ├─ Grottserver_history.txt
    ├─ grottserver.py
    ├─ grott.service
    ├─ grottsniffer.py
    └─ Version_history.txt
```

So at the end: Is there a way to make a ledidobe/grott image available for other platforms? If not, my suggestion is to revisit the docker-compose.yaml file with the local image build and not from the ledidobe/grott on docker hub .

johanmeijer, 2023-05-13:
The thing is I want to test the image I build and at this moment I do not have an arm64 device available. Although I have RPI4 I still (again) run it with a 32bits version of the OS. 

To be honest I thought that a docker image created for/on an arm32 would also run on an arm64 device (and it does but only with a 32bit OS I think). 

I suppose you have tested the grottrpi build on you ARM64 and this did not work? 

I know there should be also a possibility to create a multiple platform Image but I have to dive into that. 

Maybe in the near future I will upgrade to a 64bits OS again (and use an older RPI for the ARM32 build.

SavGRY, 2023-05-13:
Yeah, we needed a working image for arm64 and the current one wasn't working so we created this issue. We've also created a pull request to generate docker images whenever there's a push to main in the #336 PR with github Actions, using docker buildx. You can have a look at it. The image for arm64 is working and currently in use in our office.

johanmeijer, 2023-05-20:
I have created a beta beta test image with multiple architecture support. 

Image: ledidobe/grott:2.8.2 

This should support: 

    AMD64 (x86-64)
    ARM (v6/v7)  
    ARM64

Maybe somebody can try? I already being able to start a linux/x86 and linux/armv6.