johanmeijer/grott #337: Sniff out of range

hofikhof, 2023-05-13:
I used grott with success and after reinstall OS to bullseye I see this errors - grott crashing ....
Why???

May 13 10:33:07 kotel grott[77465]: Grott sniff mode started
May 13 10:33:25 kotel grott[77465]: Traceback (most recent call last):
May 13 10:33:25 kotel grott[77465]:   File "/home/pi/grott/grott.py", line 47, in <module>
May 13 10:33:25 kotel grott[77465]:     sniff.main(conf)
May 13 10:33:25 kotel grott[77465]:   File "/home/pi/grott/grottsniffer.py", line 30, in main
May 13 10:33:25 kotel grott[77465]:     self.ipv4 = IPv4(self.eth.data)
May 13 10:33:25 kotel grott[77465]:   File "/home/pi/grott/grottsniffer.py", line 93, in __init__
May 13 10:33:25 kotel grott[77465]:     version_header_length = raw_data[0]
May 13 10:33:25 kotel grott[77465]: IndexError: index out of range

hofikhof, 2023-05-13:
> I used grott with success and after reinstall OS to bullseye I see this errors - grott crashing .... Why???
> 
> May 13 10:33:07 kotel grott[77465]: Grott sniff mode started May 13 10:33:25 kotel grott[77465]: Traceback (most recent call last): May 13 10:33:25 kotel grott[77465]: File "/home/pi/grott/grott.py", line 47, in May 13 10:33:25 kotel grott[77465]: sniff.main(conf) May 13 10:33:25 kotel grott[77465]: File "/home/pi/grott/grottsniffer.py", line 30, in main May 13 10:33:25 kotel grott[77465]: self.ipv4 = IPv4(self.eth.data) May 13 10:33:25 kotel grott[77465]: File "/home/pi/grott/grottsniffer.py", line 93, in **init** May 13 10:33:25 kotel grott[77465]: version_header_length = raw_data[0] May 13 10:33:25 kotel grott[77465]: IndexError: index out of range

I fixed it by editing in code grottsniffer.py about line 90 and it is working

#Unpacks IPV4 packet
class IPv4:

    def __init__(self, raw_data):
        try:
          version_header_length = raw_data[0]
          self.version = version_header_length >> 4
          self.header_length = (version_header_length & 15) * 4
          self.ttl, self.proto, src, target = struct.unpack('! 8x B B 2x 4s 4s', raw_data[:20])
          self.src = self.ipv4addr(src)
          self.target = self.ipv4addr(target)
          self.data = raw_data[self.header_length:]
        except IndexError:
            print("Index is out of range")

rany2, 2023-05-13:
@hofikhof You could also add a guard at the start, like this:

```
#Unpacks IPV4 packet
class IPv4:

def __init__(self, raw_data):
      if not raw_data:
            return

      version_header_length = raw_data[0]
      self.version = version_header_length >> 4
      self.header_length = (version_header_length & 15) * 4
      self.ttl, self.proto, src, target = struct.unpack('! 8x B B 2x 4s 4s', raw_data[:20])
      self.src = self.ipv4addr(src)
      self.target = self.ipv4addr(target)
      self.data = raw_data[self.header_length:]

```

johanmeijer, 2023-05-13:
Ah you still using the sniffer mode. I stopped developing and testing that one because I did not like the port forwarding (to complex and error prone). It works as long as it works. And if somebody detect a problem and a solution I will add that to Grott. 

Do you have a reason not to go the proxy version? I think Grott is stable enough to run 24*7 and if Grott does not work  the data is being buffered in the inverters datalogger and send later.

hofikhof, 2023-05-14:
> Ah you still using the sniffer mode. I stopped developing and testing that one because I did not like the port forwarding (to complex and error prone). It works as long as it works. And if somebody [...]

Earlier I can set ip on shinelanbox with admin/admin pass and now I can't access to lanbox, maybe somebody changed pass. I tried to reset shine lanbox but I can't login to this device. I dost know why pass not be accepted then I need sniff mode. :(

johanmeijer, 2023-05-15:
You can also try to set ip of the shinelan box via the Growatt website,