johanmeijer/grott #339: On TL3 settings epv3today and evp3total is missing

shocker2, 2023-05-14:
Hello,
  I have just started using Grott and works great! Playing with the config I have seen that two values are missing epv3total and epv3today.

This is the output of the config:
```
grott        |   - Grott automatic protocol detection
grott        |   - Grott data record length 585
grott        |   - layout   :  T060104XTL3
grott        |   - no matching record layout found, try generic
grott        |   - Record layout used :  T06NNNNXTL3
grott        |   - Growatt data decrypted V2
grott        |   - Grott Growatt data decrypted
grott        |   - Growatt plain data:
grott        |           00410006024101045847443542475033564b00000000000000000000000000000000000000004
grott        |           6414b35434548303759000000000000000000000000000000000000000017050e0a3320020000
grott        |           007c00010001444d12e7004100007add0f87002c000044521a1a00330000851e0000000000000
grott        |           000000000000000000000000000000000000000000000000000000000000000000000013f0a13
grott        |           88095f007800007074095d007e000075fa088d0076000064e60fff0fa10f9c000000a90000ba8
grott        |           200cbe2610000004b0000346600000019000030280000003f00004aff00000000000000000000
grott        |           0000000000000000000000000000000000000000000000000000000000000000af8d02ba03360
grott        |           2b7000000000d930d6c4e2000000003640e000000000000000000000000000000000000002400
grott        |           3c0000000000000000000000000000000000000000007d00f9000000000000000000000000000
grott        |           0000000000000000000000000000000000000000012de004012de00000f92002a0f9200001a0a
grott        |           00331a0a000000000000000000000000000000000000000000000000000000000000000000000
grott        |           000000000000000000000000022000000020000000000000000000000540000000008ae49b600
grott        |           00000000170005000d00170005000e000000000000132200c1004000810000000700000000000
grott        |           00000000000000000000000000000000000000000000000000000000000000000000000000000
grott        |           0000000000013f1b000000e500017a6b000000000000000000000000000000000000000000000
grott        |           0000000000043b8
grott        |   - Growatt new layout processing
grott        |           - decrypt       :  True
grott        |           - offset        :  6
grott        |           - record layout :  T06NNNNXTL3
grott        | 
grott        |   - Grott data record date/time processing started
grott        |   - date-time:  2023-05-14T10:51:32
grott        |   - Grott values retrieved:
grott        |           -  datalogserial        :  hidden
grott        |           -  pvserial             :  hidden
grott        |           -  recortype1           :  0
grott        |           -  recortype2           :  124
grott        |           -  pvstatus             :  1
grott        |           -  pvpowerin            :  8302.1
grott        |           -  pv1voltage           :  483.9
grott        |           -  pv1current           :  6.5
grott        |           -  pv1watt              :  3145.3
grott        |           -  pv2voltage           :  397.5
grott        |           -  pv2current           :  4.4
grott        |           -  pv2watt              :  1749.0
grott        |           -  pv3voltage           :  668.2
grott        |           -  pv3current           :  5.1
grott        |           -  pv3watt              :  3407.8
grott        |           -  pvpowerout           :  8167.4
grott        |           -  pvfrequentie         :  50.0
grott        |           -  pvgridvoltage        :  239.9
grott        |           -  pvgridcurrent        :  12.0
grott        |           -  pvgridpower          :  2878.8
grott        |           -  pvgridvoltage2       :  239.7
grott        |           -  pvgridcurrent2       :  12.6
grott        |           -  pvgridpower2         :  3020.2
grott        |           -  pvgridvoltage3       :  218.9
grott        |           -  pvgridcurrent3       :  11.8
grott        |           -  pvgridpower3         :  2583.0
grott        |           -  Vac_RS               :  409.5
grott        |           -  Vac_ST               :  400.1
grott        |           -  Vac_TR               :  399.6
grott        |           -  pvenergytoday        :  16.9
grott        |           -  pvenergytotal        :  4774.6
grott        |           -  totworktime          :  1855.8
grott        |           -  epv1today            :  7.5
[... 82 lines cut ...]
grott        |                   "voltage_l1": 2147, "voltage_l2": 2385, "voltage_l3": 2389, "Current_l1":
grott        |                   189, "Current_l2": 56, "Current_l3": 47, "act_power_l1": 40674,
grott        |                   "act_power_l2": -7020, "act_power_l3": -10593, "app_power_l1": 40891,
grott        |                   "app_power_l2": 13438, "app_power_l3": 11303, "react_power_l1": -2949,
grott        |                   "react_power_l2": -11556, "react_power_l3": -4240, "powerfactor_l1": 997,
grott        |                   "powerfactor_l2": -510, "powerfactor_l3": -926, "pos_rev_act_power": 23061,
grott        |                   "pos_act_power": 23061, "rev_act_power": 23061, "app_power": 24353,
grott        |                   "react_power": -17321, "powerfactor": 688, "frequency": 499, "L1-2_voltage":
grott        |                   3947, "L2-3_voltage": 4132, "L3-1_voltage": 3934, "pos_act_energy": 221131,
grott        |                   "rev_act_energy": 1522, "pos_act_energy_kvar": 161, "rev_act_energy_kvar":
grott        |                   65650, "app_energy_kvar": 232176, "act_energy_kwh": 222654,
grott        |                   "react_energy_kvar": 65811}}
grott        |   - Grott MQTT topic used : energy/growatt
grott        |   - MQTT message message sent
```

My device is MID 20KTL3-X1 20kW with ShineWiFi-X firmware 3.1.0.5 as a datalogger and SDM630-Modbus v3 for metering.
In Grott config I have:
```
[Generic]
invtype=tl3
includeall = True
```

Thank you!

shocker2, 2023-05-16:
> If you use docker you have to add the ginvtype and gincludeall environmentals.

Thanks for the feedback! I have already added includeall = True in the grott.ini but I don't have anything in docker. On Grott startup I can see that the includeall = True is already defined, where should I add this in docker?

shocker2, 2023-05-16:
I have added it but nothing changed. I have attached the log file as well.

In Grott I have:
```
grott        |  invtype:                 "tl3"
grott        |  include_all:             True
```

In grott.ini I have:
```
[Generic]
invtype=tl3
includeall = True
```

And the docker-compose is started with:
```
  grott:
    container_name: grott
    image: ledidobe/grott:2.7.8
    volumes:
      - ./grott/output:/tmp/grottlog/
      - ./grott/config/grott.ini:/app/grott.ini
    ports:
[grott.log](https://github.com/johanmeijer/grott/files/11490680/grott.log)

      - 5279:5279
    environment:
     - ginvtype="tl3"
     - gincludeall=True
```

johanmeijer, 2023-05-18:
Hmm strange. I will test it. At this moment my docker/linux is inoperational. I hope I have it up and running shortly so I can help you.

johanmeijer, 2023-05-18:
I you copy t06NNNNXTL3.json layout file from examples/layout to your grott main directory it will work (I think). 

You still have to use includeall= parameter. 

For docker you have to make an external file reference, something like this in your docker compose volumes section: 

    - ./grott/config/tNNNNXTL3.json:/app/tNNNNXTL3.json

shocker2, 2023-05-18:
> I you copy t06NNNNXTL3.json layout file from examples/layout to your grott main directory it will work (I think).
>
> You still have to use includeall= parameter.
>
> For docker you have to make an [...]

I'll test in 15min

shocker2, 2023-05-18:
grott        |  invtype:                 "tl3"
grott        |  invtypemap:              {}
grott        |  include_all:             True

shocker2, 2023-05-18:
I don't have the values from the inverter now, only the ones from the smart meter SDM630 ModBus v3.

```
grott        |   - Grott values retrieved:
grott        |           -  datalogserial        :  3VK
grott        |           -  pvserial             :  07Y
grott        |           -  voltage_l1           :  214.2
grott        |           -  voltage_l2           :  233.3
grott        |           -  voltage_l3           :  234.1
grott        |           -  Current_l1           :  25.1
grott        |           -  Current_l2           :  7.9
grott        |           -  Current_l3           :  5.5
grott        |           -  act_power_l1         :  5392.8
grott        |           -  act_power_l2         :  1620.5
grott        |           -  act_power_l3         :  1230.5
grott        |           -  app_power_l1         :  5431.6
grott        |           -  app_power_l2         :  1867.9
grott        |           -  app_power_l3         :  1333.1
grott        |           -  react_power_l1       :  -137.7
grott        |           -  react_power_l2       :  -907.4
grott        |           -  react_power_l3       :  -437.7
grott        |           -  powerfactor_l1       :  1.0
grott        |           -  powerfactor_l2       :  0.9
grott        |           -  powerfactor_l3       :  0.9
grott        |           -  pos_rev_act_power    :  8243.8
grott        |           -  pos_act_power        :  8243.8
grott        |           -  rev_act_power        :  8243.8
grott        |           -  app_power            :  8416.8
grott        |           -  react_power          :  -1529.7
grott        |           -  powerfactor          :  1.0
grott        |           -  frequency            :  49.9
grott        |           -  L1-2_voltage         :  388.2
grott        |           -  L2-3_voltage         :  405.0
grott        |           -  L3-1_voltage         :  388.8
grott        |           -  pos_act_energy       :  22939.1
grott        |           -  rev_act_energy       :  182.7
grott        |           -  pos_act_energy_kvar  :  16.5
grott        |           -  rev_act_energy_kvar  :  6707.4
grott        |           -  app_energy_kvar      :  24079.7
grott        |           -  act_energy_kwh       :  23121.8
grott        |           -  react_energy_kvar    :  6724.0
grott        |   - MQTT jsonmsg: 
grott        |                   {"device": "3VK", "time": "2023-05-18T16:09:14", "buffered": "no",
grott        |                   "values": {"datalogserial": "3VK", "pvserial": "07Y",
grott        |                   "voltage_l1": 2142, "voltage_l2": 2333, "voltage_l3": 2341, "Current_l1":
grott        |                   251, "Current_l2": 79, "Current_l3": 55, "act_power_l1": 53928,
grott        |                   "act_power_l2": 16205, "act_power_l3": 12305, "app_power_l1": 54316,
grott        |                   "app_power_l2": 18679, "app_power_l3": 13331, "react_power_l1": -1377,
grott        |                   "react_power_l2": -9074, "react_power_l3": -4377, "powerfactor_l1": 999,
grott        |                   "powerfactor_l2": 874, "powerfactor_l3": 944, "pos_rev_act_power": 82438,
grott        |                   "pos_act_power": 82438, "rev_act_power": 82438, "app_power": 84168,
grott        |                   "react_power": -15297, "powerfactor": 983, "frequency": 499, "L1-2_voltage":
grott        |                   3882, "L2-3_voltage": 4050, "L3-1_voltage": 3888, "pos_act_energy": 229391,
grott        |                   "rev_act_energy": 1827, "pos_act_energy_kvar": 165, "rev_act_energy_kvar":
grott        |                   67074, "app_energy_kvar": 240797, "act_energy_kwh": 231218,
grott        |                   "react_energy_kvar": 67240}}
grott        |   - Grott MQTT topic used : energy/growatt
```

Docker started with:
```
   volumes:
      - ./grott/output:/tmp/grottlog/
      - ./grott/config/grott.ini:/app/grott.ini
      - ./grott/config/t06NNNNXTL3.json:/app/t06NNNNXTL3.json
```

```
docker-compose# cat ./grott/config/t06NNNNXTL3.json 
{"T06NNNNXTL3": {
[... 22 lines cut ...]
    "pvgridvoltage3"    : {"value" :342, "length" : 2, "type" : "num", "divide" : 10},                
    "pvgridcurrent3"    : {"value" :346, "length" : 2, "type" : "num", "divide" : 10},                
    "pvgridpower3"      : {"value" :350, "length" : 4, "type" : "num", "divide" : 10},  
    "Vac_RS"            : {"value" :358, "length" : 2, "type" : "num", "divide" : 10},                
    "Vac_ST"            : {"value" :362, "length" : 2, "type" : "num", "divide" : 10},                
    "Vac_TR"            : {"value" :366, "length" : 2, "type" : "num", "divide" : 10},                
    "pvenergytoday"     : {"value" :370, "length" : 4, "type" : "num", "divide" : 10},                  
    "pvenergytotal"     : {"value" :378, "length" : 4, "type" : "num", "divide" : 10},
    "totworktime"       : {"value" :386, "length" : 4, "type" : "num", "divide" : 7200},
    "epv1today"         : {"value" :394, "length" : 4, "type" : "num", "divide" : 10},                
    "epv1total"         : {"value" :402, "length" : 4, "type" : "num", "divide" : 10},                
    "epv2today"         : {"value" :410, "length" : 4, "type" : "num", "divide" : 10},                 
    "epv2total"         : {"value" :418, "length" : 4, "type" : "num", "divide" : 10},
    "epv3today"         : {"value" :426, "length" : 4, "type" : "num", "divide" : 10},                 
[... cut ...]
    "pvboottemperature" : {"value" :538, "length" : 2, "type" : "num", "divide" : 10},          
    "temp4"             : {"value" :542, "length" : 2, "type" : "num", "divide" : 10},          
    "uwBatVolt_DSP"     : {"value" :546, "length" : 2, "type" : "num", "divide" : 10},          
    "pbusvolt"          : {"value" :550, "length" : 2, "type" : "num", "divide" : 1},                  
    "nbusvolt"          : {"value" :554, "length" : 2, "type" : "num", "divide" : 1}                                
     } }
```

johanmeijer, 2023-05-18:
Strange this all seem to be oke for me. Have you already seen a t06xxxx04 record in your log?

shocker2, 2023-05-18:
There are, but there are no inverter values seen.
Attached is the full log.
[t06xxxx04.log](https://github.com/johanmeijer/grott/files/11508181/t06xxxx04.log)

johanmeijer, 2023-05-18:
Ok something went wrong there: 

      - layout   :  T060104X"TL3"
      
 There are quotes around TL3.

johanmeijer, 2023-05-18:
I think it is in the invtype= parm

shocker2, 2023-05-18:
Indeed I hard the quotes in docker env. I have removed them and now the inverter is showsing the values but epv3 is still missing. From the log:

```
grott        |  Pulled 'gincludeall=True' from the environment
grott        |  Pulled 'ginvtype=tl3' from the environment

grott        |  invtype:                 tl3
grott        |  include_all:             True

grott        |   - Growatt new layout processing
grott        |           - decrypt       :  True
grott        |           - offset        :  6
grott        |           - record layout :  T06NNNNTL3
grott        | 
grott        |   - Grott data record date/time processing started
grott        |   - date-time:  2023-05-18T17:39:37
grott        |   - Grott values retrieved:
grott        |           -  datalogserial        :  the_3VK
grott        |           -  pvserial             :  the_07Y
grott        |           -  recortype1           :  0
grott        |           -  recortype2           :  124
grott        |           -  pvstatus             :  1
grott        |           -  pvpowerin            :  2233.8
grott        |           -  pv1voltage           :  455.5
grott        |           -  pv1current           :  1.2
grott        |           -  pv1watt              :  546.6
grott        |           -  pv2voltage           :  402.5
grott        |           -  pv2current           :  1.1
grott        |           -  pv2watt              :  442.7
grott        |           -  pv3voltage           :  636.8
grott        |           -  pv3current           :  1.7
grott        |           -  pv3watt              :  1082.5
grott        |           -  pvpowerout           :  2200.3
grott        |           -  pvfrequentie         :  50.0
grott        |           -  pvgridvoltage        :  234.9
grott        |           -  pvgridcurrent        :  3.4
grott        |           -  pvgridpower          :  798.6
grott        |           -  pvgridvoltage2       :  236.8
grott        |           -  pvgridcurrent2       :  3.6
grott        |           -  pvgridpower2         :  852.4
grott        |           -  pvgridvoltage3       :  213.1
grott        |           -  pvgridcurrent3       :  3.4
grott        |           -  pvgridpower3         :  724.5
grott        |           -  Vac_RS               :  401.9
grott        |           -  Vac_ST               :  393.2
grott        |           -  Vac_TR               :  390.4
grott        |           -  pvenergytoday        :  52.7
grott        |           -  pvenergytotal        :  5057.5
grott        |           -  totworktime          :  1918.3
grott        |           -  epv1today            :  16.8
grott        |           -  epv1total            :  1428.2
grott        |           -  epv2today            :  9.9
grott        |           -  epv2total            :  1288.5
grott        |           -  epvtotal             :  4764.5
grott        |           -  pvtemperature        :  67.9
grott        |           -  pvipmtemperature     :  66.9
grott        |           -  pvboottemperature    :  60.4
grott        |           -  temp4                :  0.0
grott        |           -  uwBatVolt_DSP        :  0.0
grott        |           -  pbusvolt             :  3285
grott        |           -  nbusvolt             :  3298
```

shocker2, 2023-05-27:
@johanmeijer did you had time to check the template?

johanmeijer, 2023-05-29:
I will look at this week

shocker2, 2023-06-25:
> I will look at this week

Hi Johan, did you manage to check this out?

johanmeijer, 2023-06-29:
Oops no. It is at the do list (but moved a bit down). At this moment I am abroad for work / Holidays. It will look at it when I am back mid July (put a not in my agenda). Sorry for the delay.

shocker2, 2023-06-29:
Thank you for the feedback. No rush, just trying to keep this up to not forget :) Let me know if you need any help, logs, or access from my side to ease the fix/troubleshooting.