johanmeijer/grott #341: Grottserver not Working

T1gg1, 2023-05-18:
Hey,

I tried grottserver. It is working if I set the ip to server.growatt.com in the .ini file
But if I set it to localhost or my grottserver ip then I got this error message in the log:

Traceback (most recent call last):

  File "grott.py", line 35, in <module>

    proxy.main(conf)

  File "/app/grottproxy.py", line 129, in main

    inputready, outputready, exceptready = ss(self.input_list, [], [])

ValueError: filedescriptor out of range in select()

Can anyone help?

johanmeijer, 2023-05-18:
It should work with grottserver ip specified, at least at my place and a lot of others I think.  I will try local host. On which OS/system do you run grott / grottserver (windows, linux, docker)?

T1gg1, 2023-05-18:
I run it in a docker container

johanmeijer, 2023-05-18:
Both Grott and Grottserver? 

The grottserver container is enough, that has both grott and grottserver in it. 
So run only grottserver and set the dest IP address in the datalogger to grottserver IP with port 5279 (grott port), And it should work.

johanmeijer, 2023-05-18:
if you use to run both it will be something like: 

inverter/datalogger => grott => [grott => grottserver]

johanmeijer, 2023-05-18:
yes that have both. So use the ip / protnummer of the grottserver container to point to directly from the datalogger.

You have to start it with to ports externalized (the grottt port: 5279 and the grottserver api port: 5782). 

    docker run -d -p 5279:5279 -p 5782:5782 --name grottserver -e gmqttip="your.mqttip.address.number" ledidobe/grottserver:beta
    
 please stop the grott container because that is using the 5279 port as well.

T1gg1, 2023-05-18:
It works well if I put in the server.growatt.com ip. I can see all the data. But if I want to cut it from the cloud, there comes no data and the stick in the cloud is offline

T1gg1, 2023-05-18:
So again. I use ONLY the ledidobe/grottserver version 2.7.8
in my grott.ini I have ip:server.growatt.com
Everything works fine. I can see the data and use it via mqtt. I see the stick online on the growattcloud server.
Still api XXX.XXX.XXX.XXX:5782/datalogger output is {} Issue: #334 

If I change the ip from server.growatt.com to localhost or the ip from the server where the container runs, I got the error from above in the log, I do not get any updates from the stick and the stick is offline on the growatt cloud server.

johanmeijer, 2023-05-18:
Please stay friendly, I am patience with you too!)

I did not expect that grottserver image is being used with an external grott.ini. The internal one is configured right for most configurations, and changes can be made by using the environmental variables. 

You only use the grott part of the grottserver image at this moment. 

In your grott.ini you have to specify (port of grottserver), to make use of the grottserver part. 

    [Growatt]
    ip = localhost
    port = 5781

T1gg1, 2023-05-18:
Sorry I only summarized. I didn't mean it unkindly.

I got this error now:

	 - Can't establish connection with remote server.

	 - Closing connection with client side ('172.17.0.1', 37600)

	 - Grott - grottproxy forward error :  [Errno 111] Connection refused

	 - Can't establish connection with remote server.

	 - Closing connection with client side ('172.17.0.1', 37604)

	 - Grott - grottproxy forward error :  [Errno 111] Connection refused

johanmeijer, 2023-05-18:
Hmm ok. an you try to run it without the external grott.ini?

johanmeijer, 2023-05-18:
Oke ;) 

I run grottserver container with a time zone specified at the run command: 

     -e TZ=Europe/Amsterdam

T1gg1, 2023-05-18:
Or can I set the time via the grott.ini?