johanmeijer/grott #345: 3600TL-XH not compatible?

bartpaeleman, 2023-05-29:
I think my problem is similar to the existing case #311. 
My Growatt config is defined as follows:
INVERTER: 3600TL-XH (ShineLan + ShineRF)
BATTERIES: Growatt ARK 10.2H (LiFePO4)
PVSerial (anonimized): ADJ0B00000

Grott, MQTT and Home Assistant containers running on Docker, there is communication established (can see the messages come in).

I defined the inverter in my Grott ini as an sph-type (tried the others too) and can’t seem to get the values correctly interpreted.  Also tried the above configuration template but that doesn’t work since my system seems to look for different template files.

Is there anything I can do to get this working or do I give up?

Excerpt from my GROTT Log:

```
- Growatt original Data:
		 \x00\xc5\x00\x06\x02\x41\x50\x03\x09\x33\x2c\x42\x20\x40\x46\x75\x30\x24\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x2e\x33\x2b\x44\x36\x76\x41\x5f\x45\x34\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x75\x61
		 \x74\x74\x3b\x72\x6e\x76\x98\x74\x74\x47\x16\x6f\x77\x46\x64\x74\x47\xfe\xcf
		 \x74\x89\x35\x38\x76\x5c\x5f\x77\x3b\x35\x36\x06\x72\x66\x77\x60\x74\x74\x44
		 \x9a\x6f\x4b\x61\x48\x74\x8f\x72\x0b\x77\x61\x35\x30\x0d\x42\x2d\x46\x52\x44
		 \x46\x12\x72\x6f\x77\x61\x74\x75\x47\x16\x6f\x77\x61\x74\x54\x67\x52\x3f\x21
		 \x41\x3d\x1a\x31\x17\x1d\x03\x04\x06\x54\x67\x72\x6f\x64\x8d\x76\x75\x40\x95
		 \x6f\x72\x61\x69\x74\x4a\x72\x75\x77\x47\x74\x75\x40\x42\x65\x22\x73\xfa\x60
		 \x59\x75\x5f\x7d\x34\x66\xfa\x53\x6c\x68\x47\x6b\x21\x66\xc9\x66\x71\x70\xc2
		 \x7d\x96\x54\x0c\x7c\xe5\x61\x46\x74\x75\x72\x5d\x77\x53\x74\x46\x47\x40\x6f
		 \x45\x61\x46\x74\x75\x72\x5d\x77\x65\x74\x70\x4e\x90\x6f\x77\x20\x38\x36\x06
		 \x42\x5b\x47\x56\x44\x45\x77\x4b\x6e\x46\x61\x74\x74\x47\x61\xf3\x77\x53\x7d
		 \xc0\x4e\x90\x67\x33\x69\x62\x74\x53\x72\x6a\x7e\x0e\x7c\x88\x60\x62\x48\x67
		 \x46\x64\x53\x57\x55\x7f\x50\x71\x74\x10\x47\x72\x6e\x93\x61\x8b\x3a\x67\x72
		 \x90\x39\x41\x74\x8b\x09\x52\x6f\x88\x2f\x54\x79\x4f\x72\x6f\x78\x60\x74\x50
		 \x47\x72\x6f\x77\x61\x74\x7f\xff\x7e\x5b\x77\x61\x35\x30\x0d\x42\x2d\x46\x52
		 \x44\x46\x12\x42\x5f\x47\x51\x44\x44\x77\x42\x5f\x47\x51\x44\x44\x77\x42\x5f
		 \x47\x51\x44\x44\x47\x72\x6e\x83\x61\x74\x75\xd7\x72\x6e\x77\x61\x61\x08\x47
		 \x72\x6d\x2f\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x51\xda\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x10\x47\x7d\x4f\x77\x60\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x23\x72\x0b\x77\x60\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x36\x2f\x29\x44\x44\x77
		 \x42\x5f\x45\x50\x45\x4c\x77\x42\x5a\x46\x61\x74\x2e\x02\x30\x2e\x77\x61\x22
		 \x37\x06\x33\x6f\x73\x6b\x29\x74\x43\x72\x6f\x75\xbd\x74\x74\x47\x72\x6f\x67
		 \x65\x74\x74\x47\x72\x5f\x77\x61\x75\x77\x47\x78\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x23\x3e
	 - Grott automatic protocol detection
	 - Grott data record length 585
	 - layout   :  T065003XSPH
	 - no matching record layout found, try generic
	 - Record layout used :  T065003XSPH
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00c50006024150034e41433541343232424b00000000000000000000000000000000000000004
		 1444a304231333032550000000000000000000000000000000000000000000000000000020000
		 007c000101f9000000640000271000008ca003e8414c312e30005a41424100090001000003e80
		 03c003c00c80064000041444a3042313330325500000000000100640000000020202050562049
		 6e7665727465722020000013ec020107e70005001d000d001a0026000107300a55128e141e073
		 00a55128e141e07300a55128e141e07a309e2137e139200320032003200320032003200320032
		 003200320004000409e20000414c42413034303730313039013100000000139c003209b409e20
		 844081600140005096f08fc2710271027102710271027100064000001e400ff4e2000ff4e2000
		 ff4e2000ff4e200d0800000f0100240000000000000bb80c34000041444a30423133303255303
		 0303030303030303030303030303030303030000001f40000019000010000157c000002580000
		 000000000000000016a8000000000000000000000064000f20000100000000000000000000000
		 00000000064006400010000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000005958483030303030323131383030353100005a45424100005643414100040a5d
		 0004000002dc00000000001004000000003000000103000a00000000000000000000000000000
[... 133 lines cut ...]
			 {"device": "ADJ0B00000", "time": "2023-05-29T13:28:11", "buffered": "no",
			 "values": {"pvstatus": 1, "pvpowerin": 16272, "pv1voltage": 0, "pv1current":
			 0, "pv1watt": 0, "pv2voltage": 2332, "pv2current": 69, "pv2watt": 16272,
			 "pvpowerout": 0, "pvfrequentie": 0, "pvgridvoltage": 2439, "pvgridcurrent":
			 0, "pvgridpower": 0, "pvgridvoltage2": 0, "pvgridcurrent2": 0,
			 "pvgridpower2": 0, "pvgridvoltage3": 9482, "pvgridcurrent3": 1075,
			 "pvgridpower3": [3340566528](tel:3340566528), "totworktime": 34890, "eactoday": 70157,
			 "pvenergytoday": 70157, "eactotal": 0, "epvtotal": 0, "epv1today": 66,
			 "epv1total": 35267, "epv2today": 0, "epv2total": 0, "pvtemperature": 350,
			 "pvipmtemperature": 350, "pvboosttemp": 0, "bat_dsp": 413, "eacharge_today":
			 0, "eacharge_total": 60, "batterytype": 0, "uwsysworkmode": 0,
			 "systemfaultword0": 39, "systemfaultword1": 0, "systemfaultword2": 16517,
			 "systemfaultword3": 0, "systemfaultword4": 26, "systemfaultword5": 0,
			 "systemfaultword6": 14561, "systemfaultword7": 0, "pdischarge1": 0,
			 "p1charge1": 117571585, "vbat": 8321, "SOC": 0, "pactouserr": 5111808,
			 "pactousertot": 0, "pactogridr": 0, "pactogridtot": 0, "plocaloadr": 0,
			 "plocaloadtot": 4047, "spdspstatus": 257, "spbusvolt": 0, "etouser_tod":
			 [1401552927](tel:1401552927), "etouser_tot": 2559950, "etogrid_tod": 134807569, "etogrid_tot":
			 [1114408](tel:1114408), "edischarge1_tod": 0, "edischarge1_tot": 0, "eharge1_tod":
			 443023360, "eharge1_tot": [1082458112](tel:1082458112), "elocalload_tod": 954269696,
			 "elocalload_tot": 198597}}
	 - MQTT message message sent
```

_Originally posted by @bartpaeleman in https://github.com/johanmeijer/grott/discussions/311#discussioncomment-6029077_

johanmeijer, 2023-05-29:
See #311 the temp solution seems to be working.I will add in permanently in the update

bartpaeleman, 2023-05-29:
Thanks @johanmeijer, I tried to do what was mentioned on that thread but it doesn’t seem to fly here. 
I also used the T06NNNNXMIN.json layout file to the examples/layout directory but it doesn’t work. I also used the trick to bond the example file into the app folder on the Docker container but still no luck.

johanmeijer, 2023-05-29:
Can you try the experimental ledidobe/grott:2.8.2 image (it is the first try with multiple architecture support). You can use ginvtype=min with that build.

bartpaeleman, 2023-05-29:
Sure thing. Let me see if I can get that going right away.

bartpaeleman, 2023-05-29:
Getting this in the log after installing that image:

	 - Growatt original Data:
		 \x00\x4b\x00\x06\x02\x41\x50\x04\x09\x33\x2c\x42\x20\x40\x46\x75\x30\x24\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x2e\x33\x2b\x44\x36\x76\x41\x5f\x45\x34\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x75\x6a
		 \xcc\x78\x73\x72\x6e\x77\x61\x51\xdc\x47\x72\x6f\x77\x61\x74\x74\x47\x7b\xcf
		 \x77\x46\x74\x74\x62\xda\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x69\xdb\x72\x6f\x77\x61\x74\x74\x50\x9e\x7c\xf5\x68\x1e
		 \x74\x5d\x72\x6f\x60\x8e\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x4e\x18\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x72\xeb\x65\x40\x5d\x15\x72\x6f\x77\x3e\x74\x75\x44\xc5\x6f\x76\x73\x4e\x74
		 \x47\x72\x6f\x77\x61\xfc\x3e\x47\x72\x6f\x18\x61\x74\xfd\xb7\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x78\x74\x47\xfb\x07\x77\x61\x74\x75\x47\x72\x34
		 \x04\x61\x74\x74\xc7\x72\x6e\x30\xad\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x00\x77\x61\x74\x74\xb8\x88\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x75
		 \x2e\x46\x28\x6f\x77\x61\x74\x75\xdc\x7d\xf0\x77\x61\x3a\x54\x47\x63\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6d\x81\x63\x82\x74\x32\x72
		 \x6f\x77\x61\x74\x74\x47\x4e\x6f\x77\x61\x74\x74\x46\x72\x6f\x77\x61\x74\x74
		 \x5a\xee\x6f\x77\x61\x01\x78\x72\x7e\xde\x77\x61\x74\x59\x47\x72\x2f\xfc\x61
		 \x74\x74\x60\x72\x6f\x4f\x8f\x74\x74\x47\x72\x6f\x77\x66\x76\x74\x46\x52\xc8
		 \x77\x61\x74\x07\x47\x72\xaa\x42\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x53\x64\x47\x72\x60\xd2\x61\x74\x74\x47\x73\x6e\x77
		 \x61\x74\x74\x14\x6d\x6f\x7e\x61\x45\x7b\xe3\x75\x9a\x77\x67\x74\x72\x46\x6e
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x70\xbb\x74\x74\x07\xf9\x6f\x77\x59\x9a\x74
		 \x47\x72\x6c\x70\xce\x74\x5d\x47\x62\x6f\x77\x60\x77\x74\xaa\x72\x6d\x77\x71
		 \x74\x74\x47\x72\x6f\x76\x61\x74\x74\x75\x72\x5d\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x76\x61\x74\x74\x47
		 \x72\x5e\x24\x55\x74\x0c\x46\xfd\x66\xb3\x68\xb0\x74\x47\x72\x0b\x2f\xa1\x3d
		 \xb4\x47\x72\x6f\x77\x49\xd4\x74\x47\x72\x6f\x7b\x94\x78\x85\x14\x4c\x6f\x77
		 \x61\x76\x74\x46\x72\x6f\x77\x61\x74\x74\x47\x73\x6f\x77\x61\x55\x74\xb1\x73
		 \xee\x79\x71\x74\x74\x47\x7b\x6f\x70\x61\x76\x74\x47\x37\xc8
	 - Grott automatic protocol detection
	 - Grott data record length 585
	 - layout   :  T065004XMIN
	 - no matching record layout found, try generic
	 - Record layout used :  T06NNNNXMIN
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 004b0006024150044e41433541343232424b00000000000000000000000000000000000000004
		 1444a304231333032550000000000000000000000000000000000000000000000000000020bb8
		 0c340001000025a8000000000000000009a00027000025a800000000000000000000000000000
		 00000001d9c00000000000017ec1382096a001a000017ef000000000000000000000000000000
		 00096a00000000000000000000000000001d9c043429520000005f000103b70001123a0000000
		 00000884a0000006f000089f000000000000000000000000c000089680000000100005b730000
		 0080000147cc00000000000000000000006f00000000fffa00000000000000000000015a015a0
		 0000000019b0f9f00004e200011000000000000000000000000000002f602f600750000000000
		 00003c0000000000010000000000001d9c000000750c350cb10000002d0000408b00000027000
		 038ee0000000000000702000120a7000000730000c53500000000000000000000000000000000
		 0000000000000000000000000000000000000000271000000fa500000000010100000000531f0
		 00900310fa407f500060006011c000000000000000007da0000408b000038ee0000000307af00
		 2900100000010300ed00020010000000000001000000320032000000000000000000000000000
		 0000000000000000100000000003153340078018f09c409c40000006458c049c00000000028a0
		 000000000cf50cf1533e00000002000100000000000000010000002100f601810e10000000090
		 0070002000045a7
	 - Growatt new layout processing
		 - decrypt       :  True
		 - offset        :  6
		 - record layout :  T06NNNNXMIN
	 - Grott data record date/time processing started
	 - no or no valid time/date found, grott server time will be used (buffer records not sent!)
	 - Grott values retrieved:
		 -  pvserial             :  ADJ0B000000
		 -  pvstatus             :  1
		 -  pvpowerin            :  964.0
		 -  pv1voltage           :  0.0
		 -  pv1current           :  0.0
[... 241 lines cut ...]
		 \x74\x47\x72\x6f\x77\x61\x74\x8b\xb8\x8d\x95\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x6e\x73\x8b\x8b\xa1\x9d\x90\x88\x9e\x8e\x74\x47\x73
		 \x9b\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\xa3\x97\x74\x74
		 \x1a\xf2\x6f\x77\x61\x74\x74\x46\x46\xb2\x77\x61\x74\x74\x47\x73\x5d\x01\x61
		 \x75\x40\x99\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x49\x79
	 - Grott automatic protocol detection
	 - Grott data record length 238
	 - layout   :  T065020
	 - no matching record layout found, try generic
	 - Record layout used :  T065020
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 004d000600e650204e41433541343232424b00000000000000000000000000000000000000004
		 1444a30423133303255000000000000000000000000000000000000000000000000000000a000
		 0000840000096c00000000000000000000001e000000000000000000000000000000000000000
[... cut ...] 000
		 000000001912ffffe6effffffffa000001f40000000000000000000000000000d4f600005d800
		 0000000000134dd0000000000013276000134de00000000000000000000000000000000000000
		 0000000000280d
	 - Grott data ack record or data record not defined no processing done
	 - Growatt packet received:
		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.0.220', 5279), raddr=('192.168.0.105', 1029)>
	 - Data less then minimum record length, data not processed

johanmeijer, 2023-05-29:
Oké the inverter seems to be handled OK. The smart meter not yet (t065020) not yet. Which wife device you use (shinelink)?

johanmeijer, 2023-05-29:
Oh wait this a historical record (grott should recognize this as well, bur is not in it yet). I guess normal processing for both inverter and smart meter should be processed well.

bartpaeleman, 2023-05-30:
I have an RF stick in the inverter and a Shine Lanbox plugged into my router,

bartpaeleman, 2023-06-02:
@johanmeijer could it be that the values are not always correct? I see the following coming on from Grott specificall the battery state of charge (bms_soc):

```
		 -  bms_warninfo         :  0
		 -  bms_soc              :  53
		 -  bms_batteryvolt      :  214.6
		 -  bms_batterycurr      :  4.7
		 -  bms_batterytemp      :  41.5
		 -  bms_maxcurr          :  25.0:
```

The actual value on the Growatt page show this (which I believe to be true):
[image]

My HA sensors show this up ad down jumping behaviour accross the timeline:
[image]

Is this something I did wrong on my end or is it an issue in Grott?

johanmeijer, 2023-06-08:
I am not sure. If we want to know if this is a Grott issue or sent by the inverter we need the log to look deeper into it.

bartpaeleman, 2023-06-10:
Would that be the Docker log Johan? I can tgry to see if I can get that. It seems to occur on multiple values that are also in my home assistant setup as sensors.