johanmeijer/grott #347: Grott - grottproxy forward error: [Errno 110] operation timed out

Yogi4301, 2023-06-01:
Actually I get often the above shown error during the communication with Growatt. Due to the fact that buffered events are send to Growatt again there is no loss of data.
I have set servertime due to a bug in the Shinewifi-x - therefore no data is send to MQTT and PVoutput.org in these cases.
Actually I have copied the sources and I am building the docker-container myself, so that I could reduce the logging and sent more data to PVoutput.org (V7, V8..).

Can you give me hint on how to change the behaviour, so that events are sent to MQTT an PVoutput in the case Growatt is not responding?

gajotnt, 2023-06-08:
I think theres problems with growatt servers, and for some reason GROTT will only decode the info and forward it to MQTT after reply from server.
So if growatt goes offline we get nothing... -.-

daath, 2023-06-08:
Yes, it definitely seems like a bug - it seems like it got the data, but nothing is put in MQTT :(

Yogi4301, 2023-06-08:
The growatt servers are down - and if growatt is down, no data is send to MQTT and PVoutput.org. Thats the reason why I ask how to change grott accordingly.

gajotnt, 2023-06-08:
Just got some data in home assistant.

But yes this needs to be addressed.

rcvaneck, 2023-06-08:
The error code is changed from 110 to 111. Looks like a block on Growatt side. 

- Closing connection with client side ('192.168.1.166', 37083)
	 - Grott - grottproxy forward error :  [Errno 111] Connection refused

gajotnt, 2023-06-08:
Not getting that error. Still getting 110 from time to time

Yogi4301, 2023-06-08:
Getting lots of errors - but still 110
- ```
Growatt data decrypted V2 ( 2023-06-08T20:37:51 )
         - Grott data ack record or data record not defined no processing done
         - Grott - grottproxy forward error :  [Errno 110] Operation timed out
         - Can't establish connection with remote server.
         - Closing connection with client side ('192.168.178.69', 52734)
         - Grott - grottproxy forward error :  [Errno 110] Operation timed out
         - Can't establish connection with remote server.
         - Closing connection with client side ('192.168.178.88', 18176)
         - Grott - grottproxy forward error :  [Errno 110] Operation timed out
         - Can't establish connection with remote server.
         - Closing connection with client side ('192.168.178.69', 52854)
         - Grott - grottproxy forward error :  [Errno 110] Operation timed out
         - Can't establish connection with remote server.
         - Closing connection with client side ('192.168.178.88', 13027)
         - Grott - grottproxy forward error :  [Errno 110] Operation timed out
         - Can't establish connection with remote server.
         - Closing connection with client side ('192.168.178.69', 52876)
         - Grott - grottproxy - Invalid data record received, processing stopped for this record
         - Grott connection error
         - Grott - grottproxy - Invalid data record received, processing stopped for this record
         - Grott connection error
         - Grott - grottproxy - Invalid data record received, processing stopped for this record
         - Grott connection error
         - Grott - grottproxy - Invalid data record received, processing stopped for this record
         - Grott connection error
         - Grott - grottproxy - Invalid data record received, processing stopped for this record
         - Grott - grottproxy - Invalid data record received, processing stopped for this record
         - Grott - grottproxy - Invalid data record received, processing stopped for this record
         - Grott - grottproxy - Invalid data record received, processing stopped for this record
         - Grott - grottproxy - Invalid data record received, processing stopped for this record
         - Grott - grottproxy forward error :  [Errno 110] Operation timed out
         - Can't establish connection with remote server.
         - Closing connection with client side ('192.168.178.88', 21953)
         - Grott - grottproxy - Invalid data record received, processing stopped for this record
         - Grott - grottproxy - Invalid data record received, processing stopped for this record

```

muppet3000, 2023-06-08:
Regardless of whether the servers are down or not grott should continue to push data to other endpoints e.g. mqtt, it shouldn't 'bomb out' if Growatt servers are unavailable. 

I've tried looking through the codebase base before I find it really difficult to parse, I may try and look through it again over the weekend to see if I can help out with a PR, but I'm short on time at the minute.

It would be really handy if @johanmeijer could explain whether some communication is needed to Growatt in order for grott to work or if this is just a case of an ordering problem?

johanmeijer, 2023-06-08:
Grott by it self is a proxy so read the data  and forward it. A proxy needs connection with both sides. So the inverter (datalogger) and growatt server. We nee a server (growatt server) to keep the communication going. If there is no connection, and no communication the inverter will not sent any data at all (it only tries to set up a connection). 

This is how Grott is initial designed (without the need to really understand the protocol is being used). 

After some reversed engineering of the used protocol, I also created a Grottserver (see #98) that emulates the growatt server and removes the need for growatt server. 

Be aware I also think that it should not be a problem that you lose contact with the growatt server. The inverter (datalogger) is buffering the data and will resent it the moment the connection is established again. This will work oke for MQTT and Influxdb. For PVoutput this can cause problems while the inverter will try to sent all the buffered data in a short time and then you will hit the update limits of PVOutput.

gajotnt, 2023-06-08:
@muppet3000 can this be added to your HA addon? :)

daath, 2023-06-08:
Looks like my communication has been restored, and my inverters now show up in Home Assistant, though there is a huge gap in the data, although kWh generated today is there ... It was a yet another good day for solar in Denmark: 42.6 kWh generated on a 6.6 kWp westward pointing setup 🤪

muppet3000, 2023-06-08:
@johanmeijer - Thank you for the detailed response I now understand the situation. I'll take a look at the Grottserver option and see if there's something clever I can do (and return a PR for) so that it can intelligently determine if the growatt servers are online and use those if possible (for people that want to maintain connectivity to the growatt servers) and if they are unavailable then divert to using the grottserver. 
It will take me a bit of time to work out a clean way of doing it, but I think it would be quite an elegant solution.

@gajotnt - I don't maintain that add-on, I maintain the home assistant integration for MQTT here: https://github.com/muppet3000/homeassistant-grott

I also have my communication restored (about 1 hour ago, 21:00 UK time, 20:00 UTC) and everything is working nicely again.

gajotnt, 2023-06-08:
@muppet3000 sorry I mixed up your integration with the addon --'

gajotnt, 2023-06-08:
> > @muppet3000 sorry I mixed up your integration with the addon --'
> 
> No problem, you can actually use my integration to have all the data from MQTT presented nicely in HA and egguy's one to run grott. I need to update the documentation to describe that to everyone as it's really useful.

At the time I went with the addon because it was way simpler, no need to setup, docker.
Will try your integration to check it out :)

muppet3000, 2023-06-08:
> At the time I went with the addon because it was way simpler, no need to setup, docker.
> Will try your integration to check it out :)

Yeah I completely understand that situation, I don't run a version of HA that supports the use of add-ons so I made the integration so it could be used with every version of HA. Ultimately it's just consuming and formatting the data from MQTT but in the future I plan to extend it for configuring the inverter via HA as well (once I've proven we can do config via grott).

Yogi4301, 2023-06-13:
Actually growatt is down again - I get error 110 like last time

gajotnt, 2023-06-13:
Moving to solar Assistant.

gajotnt, 2023-06-13:
Just finished installing Solar Assistant... never going back to ShinePhone/Growattserver

daath, 2023-06-13:
Solar Assistant looks cool, but AFAIK it doesn't work well with multiple inverters, if they are different models ...