johanmeijer/grott #350: Publish the record format as metadata to MQTT

shampeon, 2023-06-09:
Since the keys are not consistent between inverter models, it'd be great to publish in MQTT the record layout used to decode the data.  So e.g. the HA plugin could read the correct keys when creating sensors specific to the inverter model.

For example, in a MIN inverter, the battery voltage is `bms_batteryvolt` but in an SPH inverter it's `vbat`.

Proposed format:

~~~ json
{
  "device": "DEVICE", 
  "record_format":"RECORD_FORMAT", 
  "time": "TIME", 
  "buffered": "BUFFERED",
  "values": {...}
}
~~~

Relates to https://github.com/muppet3000/homeassistant-grott/issues/15

muppet3000, 2023-06-09:
I'd argue the two shouldn't have to be related. I believe all of the values in grott are reverse engineered from various technical documentation. This isn't hard for me to implement in the HA integration so I'm happy to do that there.
For convenience I'd say this would be nice but I believe the intention of grott is to leave everything in the most raw state possible, if people want to format it afterwards they can.

johanmeijer, 2023-08-13:
Is this still an issue?

shampeon, 2023-08-13:
The HA plugin now correctly handles the different keys, so yeah, we can close.

closed 2023-08-13 by shampeon