johanmeijer/grott #352: MOD 10KTL3-XH + Battery ARK-XH

amritpal7205, 2023-06-13:
Hi,

My setup:
Inverter: MOD 10KTL3-XH
Batteries: ARK-XH
Logger: ShineWiFi-X,
Inverter is also connected to an energy meter over RS485

I am new to all this with solar panels but have been using Home Assistant for a long time. Really happy that I found grott to get the inverter data into Home Assistant. However, my setup with MOD 10KTL3-XH and batteries ARK-XH doesn't seem to be fully supported (assuming that I am not doing something wrong).

Looking at the logs, my inverter serial is identfied correctly: QDL1XXXXXX, as well as the datalogger: XGD6CMXXXX (X - for the purpose of anonymization, are otherwise also correct). The chosen layout, however, varies: T060120 (see log below) but also T060103X (log further down). In the latter case the log says "no matching record layout found, try generic. Record layout used :  T060103X". In both cases I miss information about: battery charge/discharge, load consumption, export to grid, import from grid. Looking at the grottconf.py, I can see that this information is available for some other inverters.

Any possibility to get the full information for my setup? I have some basic python, json etc (far from an expert though). So, I'll be to some fiddeling, if needed. Also, let me know if you need any more information.

Thanks for all the great work.

LOGS below (I have anonymized the serial numbers - hope it is OK)
--------------
- Growatt original Data:
		 \x00\x5e\x00\x06\x01\x0e\x01\x20\x1f\x35\x2b\x41\x22\x39\x38\x77\x37\x3c\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x3e\x33\x2d\x45\x37\x0c\x39\x5f\x45\x59\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x76\x72\x7a\x47\x6a\x58\x77\xa9
		 \x74\x74\x47\xfc\x6f\x77\x68\x21\x74\x47\x7b\x3a\x77\x61\x7d\x17\x47\x72\x6f
		 \x7d\x61\x74\x74\x4d\x72\x6f\x77\x71\x8b\x8b\xb9\x31\x90\x88\x9a\x08\x74\x47
		 \x76\x1d\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x8d\x90\x80\xda\x8b
		 \x8b\xbe\x7f\x90\x88\x92\xd6\x8b\xb8\x8d\x37\x88\x9e\x8a\x41\x47\x72\x6e\x6d
		 \x9e\x8b\x8a\x75\x72\x6f\x77\x61\x8b\x8b\xa3\x18\x90\x88\x9e\xbd\x74\x47\x73
		 \x9b\x77\x61\x64\x5d\x47\x72\x7f\x42\x61\x74\x64\x72\x72\x6f\x7d\xe9\x74\x74
		 \x45\xd5\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\xbf\xeb
	 - Grott automatic protocol detection
	 - Grott data record length 278
	 - layout   :  T060120
	 - Record layout used :  T060120
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 005e0006010e012058474436434d4c30455300000000000000000000000000000000000000005
		 1444c31434b4b303238000000000000000000000000000000000000000017060e00183700c800
		 00008e0000095500000955000009630000000a0000000a00000010fffffe43fffffb7c0000047
		 2000000000000000000000000fffff7bbfffff90dfffff3a2ffffff58fffffe350000011affff
		 fe3200000000ffffe46affffffc9000001f400001029000010350000103500000a88000002a70
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 0000000000000d09c
	 - Growatt new layout processing
		 - decrypt       :  True
		 - offset        :  6
		 **- record layout :  T060120**
	 - Grott data record date/time processing started
	 - date-time:  2023-06-14T00:24:55
	 - Grott values retrieved:
		 -  datalogserial        :  XGD6CXXXXX
		 -  pvserial             :  QDL1XXXXXX
		 -  voltage_l1           :  238.9
		 -  voltage_l2           :  238.9
		 -  voltage_l3           :  240.3
		 -  Current_l1           :  1.0
		 -  Current_l2           :  1.0
		 -  Current_l3           :  1.6
		 -  act_power_l1         :  -44.5
		 -  act_power_l2         :  -115.6
		 -  act_power_l3         :  113.8
		 -  app_power_l1         :  0.0
		 -  app_power_l2         :  0.0
		 -  app_power_l3         :  0.0
		 -  react_power_l1       :  -211.7
		 -  react_power_l2       :  -177.9
		 -  react_power_l3       :  -316.6
		 -  powerfactor_l1       :  -0.2
		 -  powerfactor_l2       :  -0.5
		 -  powerfactor_l3       :  0.3
		 -  pos_rev_act_power    :  -46.2
		 -  pos_act_power        :  -46.2
		 -  rev_act_power        :  -46.2
		 -  app_power            :  0.0
		 -  react_power          :  -706.2
		 -  powerfactor          :  -0.1
		 -  frequency            :  50.0
		 -  L1-2_voltage         :  413.7
		 -  L2-3_voltage         :  414.9
		 -  L3-1_voltage         :  414.9
		 -  pos_act_energy       :  269.6
		 -  rev_act_energy       :  67.9
	 - MQTT jsonmsg: 
			 {"device": "XGD6CXXXXX", "time": "2023-06-14T00:24:55", "buffered": "no",
			 "values": {"datalogserial": "XGD6CXXXXX", "pvserial": "QDL1XXXXXX",
			 "voltage_l1": 2389, "voltage_l2": 2389, "voltage_l3": 2403, "Current_l1":
			 10, "Current_l2": 10, "Current_l3": 16, "act_power_l1": -445,
			 "act_power_l2": -1156, "act_power_l3": 1138, "app_power_l1": 0,
			 "app_power_l2": 0, "app_power_l3": 0, "react_power_l1": -2117,
			 "react_power_l2": -1779, "react_power_l3": -3166, "powerfactor_l1": -168,
			 "powerfactor_l2": -459, "powerfactor_l3": 282, "pos_rev_act_power": -462,
			 "pos_act_power": -462, "rev_act_power": -462, "app_power": 0, "react_power":
			 -7062, "powerfactor": -55, "frequency": 500, "L1-2_voltage": 4137,
			 "L2-3_voltage": 4149, "L3-1_voltage": 4149, "pos_act_energy": 2696,
			 "rev_act_energy": 679}}
	 - Grott MQTT topic used : energy/growatt
	 - MQTT message message sent
	 - Grott Send data to PVOutput disabled 
	 - Grott Send data to Influx disabled 
	 - Grott extension processing started :  grott_ha
	 - Grott extension processing ended :  0
	 - Growatt packet received:
		  <socket.socket fd=5, family=2, type=1, proto=0, laddr=('172.30.33.11', 5279), raddr=('192.168.1.40', 33734)>
	 - Data less then minimum record length, data not processed 

---------------------------

 - Growatt original Data:
		 \x00\x0a\x00\x06\x02\x41\x01\x03\x1f\x35\x2b\x41\x22\x39\x38\x77\x37\x3c\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x3e\x33\x2d\x45\x37\x0c\x39\x5f\x45\x59\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x76\x72\x7a\x47\x68\x5e\x75\x61
		 \x74\x74\x3b\x72\x6e\x77\xf8\x8b\x8b\x47\x16\x6f\x77\x46\x64\x74\x46\xf4\xcf
		 \x71\x21\x30\x3a\x76\x5c\x5f\x77\x3b\x36\x30\x05\x72\x6c\x77\x60\x74\x74\x41
		 \x32\x6f\x4b\x61\x48\x74\x23\x72\x0b\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x70\x47\x16\x6f\x77\x61\x74\x54\x67\x52\x3f\x21
		 \x41\x3d\x1a\x31\x17\x1d\x03\x04\x06\x54\x67\x72\x6f\x62\x79\x76\x77\x40\x95
		 \x6f\x71\x61\x7a\x74\x47\x72\x44\x77\x61\x74\x77\x4b\x01\x7e\x92\x73\xfa\x60
		 \x59\x78\x8b\x66\x84\x66\xfa\x53\x6c\x65\x93\x70\x91\x66\xc9\x66\x71\x7a\x5b
		 \x65\x6a\x54\x0c\x7c\xe5\x61\x46\x74\x75\x72\x5d\x77\x53\x74\x46\x47\x40\x6f
		 \x45\x61\x46\x74\x75\x72\x5d\x77\x53\x74\x46\x56\x6c\x6f\x77\x25\x3a\x35\x06
		 \x42\x5e\x42\x50\x44\x44\x77\x41\x6e\x46\x61\x74\x74\x47\x61\xf3\x77\x53\x64
		 \xba\x56\x6c\x61\x26\x6f\x75\x74\x53\x72\x6a\x67\x36\x7b\xe4\x60\x6c\x48\x69
		 \x47\xb9\x53\xb9\x55\x91\x50\x96\x74\x6a\x47\x72\x6e\x93\x61\x8b\x3a\x67\x72
		 \x90\x39\x41\x74\x8b\x09\x52\x6f\x88\x2f\x54\x4a\x46\x72\x6f\x78\x60\x74\x10
		 \x47\x73\x6f\x77\x61\x74\x7f\xff\x7e\x5b\x77\x61\x25\x30\x0b\x43\x2c\x3c\x2a
		 \x44\x46\x7f\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6e\x77\x61\x5c\x48\x47
		 \x72\x6d\x2f\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x51\xda\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x10\x47\x78\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x23\x72\x0b\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x3e\x26\x2a\x47\x44\x77
		 \x42\x5f\x45\x53\x47\x46\x77\x42\x3b\x41\x61\x74\x2e\x02\x30\x2e\x77\x61\x22
		 \x37\x06\x33\x6f\x73\x6b\x5e\x74\x43\x72\x6d\x77\x60\x74\x74\x47\x72\x6f\x67
		 \x65\x74\x74\x47\x72\x5f\x77\x61\x75\x77\x47\x78\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x57\x89
	 - Grott automatic protocol detection
	 - Grott data record length 585
	 - layout   :  T060103X
	 - no matching record layout found, try generic
	 - Record layout used :  T060103X
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:

amritpal7205, 2023-06-13:
Should add, I am using grott through: [https://github.com/egguy/grott-home-assistant-add-on](url), using the edge-variant.

johanmeijer, 2023-06-15:
T060120 is normally  a smartmeter record e.g sdm630 do you have that also or this record created by the  inverter/battery.

The T060103x is a announcement record from the inverter that is only sent at startup. It is not a datarecord there should be also a t060104x record. That arevthe real datarecords and sent periodical (1× 5 minutes). 

With a 10ktl3 you should also specify invtype=tl3 in the grott.ini. the data record layout is being used then is t060104xtl3.

johanmeijer, 2023-08-13:
I did not realized that the definition file for the MOD is already in the Grott 2.8.2. version. It seems to be the same as the MIN layout. 

If you specify invtype=min in the grott.ini [Generic] section you probably see usable data.

jmalm, 2023-09-06:
@amritpal7205 , were you able to get battery information with your MOD-10KTL3-XH using grott? The only battery value that I get is `uwBatVolt_DSP`, which I don't know what it is (and neither does homeassistant-grott).

jmalm, 2023-09-07:
I switched to branch Beta-(2.8.x) and invtype=min and am now getting all the information. Thanks!

closed 2023-09-28 by johanmeijer