johanmeijer/grott #353: grott an grottserver docker images out of sync?

catch56, 2023-06-16:
I'm currently running a fork of grottserver, but would like to move back to the docker container now that the PRs I was relying on are in 2.8.1, however it looks like there is not a new grottserver image available:

i.e.  https://hub.docker.com/r/ledidobe/grott has 2.8.2

https://hub.docker.com/r/ledidobe/grottserver still has 2.7.8

Have I missed something, and the separate grottserver image is no longer necessary? Or is it actually out of date and I should wait for the latest tag to be built?

johanmeijer, 2023-06-16:
Your right they are out of sync at this moment. It is on the list to update. the grott 2.8.2 version support multiple architectures if this is working fine I will create also a new grottserver version.

johanmeijer, 2023-06-16:
Do you need specific functionality from 2.8.2? There has not been any change in the grottserver part!

catch56, 2023-06-16:
https://github.com/johanmeijer/grott/pull/285 is the one I'm after :)

johanmeijer, 2023-06-16:
Ah oke. Yes I have to build a new docker image.

catch56, 2023-06-16:
OK there is no hurry I just had the container tabs open for a while and was hopeful I could drop my fork :)

catch56, 2024-02-21:
The new docker images included the change I needed, and I'm successfully running the beta (for the past twelve hours). Thanks!