johanmeijer/grott #357: No response when reading or writing registers with the grottserver API

bsipos-precognox, 2023-06-22:
Hello,

I'm trying to use the API to read and set some registers on my inverter, but I never get a reponse. I assume it's an issue on my side, because this feature seems to work for other users. 

Example call to the grottserver: `GET http://localhost:5782/inverter?command=register&register=15&inverter=XXXXXXXX`
Response : `no or invalid response received`

Meanwhile in the grottserver logs:
```
Grotthttpserver - wait for GET response
Grotthttpserver - wait for GET response
Grotthttpserver - wait for GET response
Grotthttpserver - wait for GET response
Grotthttpserver - wait for GET response
Grotthttpserver - wait for GET response
Grotthttpserver - http response send:  400 text/body b'no or invalid response received'
...
```

So the response from the datalogger just times out. I have the same result with PUT requests. Other communication from the datalogger to the grottserver works fine, the periodic updates are logged (I attached a partial log file). [grottserver_logs.txt](https://github.com/johanmeijer/grott/files/11834326/grottserver_logs.txt)

Do you have any suggestions what should I check on my side? If you need more information I'll be happy to provide it.

johanmeijer, 2023-06-22:
Do you use blockcmd= parameter in grott? Then every request will be blocked.

A time out for the datalogger does not happens very often (not for get or put).

Setting / reading the inverter registers give more often time outs. That seems to caused by the communication between the inverter and datalogger (happens also when you try it via the growatt website.

bsipos-precognox, 2023-06-22:
Thank you for the response. Yes, I suspected that the problem with the communication is between the inverter's wifi stick and the datalogger, but I double checked this and it seems to be fine. The device state is "Normal" on the Shinelink box's admin page (screenshot below), and the LED on the wifi stick is showing the same (slowly flashing blue). 

([image])

blockmd is false (I didn't change the default), is this the correct setting if I want to to send commands with grottserver? Anything else I can try, or are growatt's inverters this unreliable?

johanmeijer, 2023-06-29:
I think we have to accept taht the inverters (communication between inverter and wifishine) are unreliable. 

I will investigate in the future if we can find out a better way to communicate (or at least detect that there is no or rubbish response).  There is a register with baudrate setting maybe changing this can help or stops the communication for ever :(

If others has idea's please let me know.

rany2, 2023-06-29:
@bsipos-precognox any chance you could test my fork? If it works with my fork I think I have an idea what may have fixed it and will ask you to try changes on johan's branch https://github.com/rany2/grott

bsipos-precognox, 2023-06-30:
@rany2 Thanks for the suggestion, I'll test it and report back.

johanmeijer, 2023-06-30:
@rany2 I am very curious what your solution is. 

For me it is simple sending a request and waiting for a response. You can change the wait for response time.

I do not see  much other possibilities to change the behaviour. (Unless it is possible to change the communication between inverter and datalogger by setting specific parameters/registers). 

The fact that the datalogg response are always handled prompt and correctly and the inverter request not indicates there is something going wrong in the (serial) communication between datalogger and inverter. But I can be wrong of course.

rany2, 2023-06-30:
@johanmeijer I had a couple of situations where when I try to receive data from datalogger, it returns two response in the same data packet. So I ended up changing the mechanism to read data, first I get modbus header to know length of data packet then I read from socket the exact length we need to get the rest of the data packet for processing.

Also the HTTP server's wait to get command response was changed so that it does it via polling instead of busy loop and I had TCP_NODELAY enabled which also helped with reliability. It seems to have helped as I have HA poll my grott constantly and never had it fail in this way, these are my logs you could see there is only one error for getting inverter command at line 2722: https://gist.github.com/rany2/5f93ac882b38497a0dc0b08f792a080e.

johanmeijer, 2023-06-30:
Thanks I going to look at it by the next update. 

I also think that firmware version of the wifishine datalogger can make a difference.

bsipos-precognox, 2023-07-09:
@rany2 I have the same result with your fork: `b'no or invalid response received`

bsipos-precognox, 2023-07-09:
I'm posting the verbose logs from Grottserver after trying to read a register (the timestamps were added by me, and the inverter ID was removed): [grottserver_logs_verbose.txt](https://github.com/johanmeijer/grott/files/11995664/grottserver_logs_verbose.txt)

The request was: `GET /inverter?command=register&register=15&inverter=XXX`

As you can see, there is some response after the command is sent, but Grottserver still returns an error. @johanmeijer do you have an idea what's going wrong here?

rany2, 2023-07-13:
@bsipos-precognox Just try to upgrade the firmware on the datalogger as well, you need to do that too.

cyberdiamond, 2023-07-28:
I am having the same issues of not being able to read any data from the inverter?

I have updated the firmware to the latest that is available and they have changed the usual admin/admin so now I am locked out of the shinelan box locally.

If there is any pattern I am using a SPH5000 with batteries and Eastron SDM230v2 meter connected to ShineLink (combination of ShineLanBox and ShineRFStick @ 433mhz)

@rany2 also then tried your fork and no different.

Although my get and put commands to the shinelink datalogger are working fine.

XXXXX'd out the real data

$ curl -X GET "http://127.0.0.1:5782/inverter?command=register&register=1091&inverter=NTCIAXXXXX"
no or invalid response received

Log from the same command:

Jul 27 21:53:53 grott grott[9517]: - GrottHttpServer - Get received
Jul 27 21:53:53 grott grott[9517]: - GrottHttpServer - inverter get received:  {'command': ['register'], 'register': ['1091'], 'inverter': ['NTCIAXXXXX']}
Jul 27 21:53:53 grott grott[9517]: - Grott - get command: register
Jul 27 21:53:53 grott grott[9517]: - GrottHttpServer - selected deviceid: 51
Jul 27 21:53:53 grott grott[9517]: - GrottHttpServer - command created:
Jul 27 21:53:53 grott grott[9517]:         \x00\x01\x00\x06\x00\x24\x51\x05\x09\x33\x2c\x42\x20\x45\x40\x75\x45\x57\x77\x
Jul 27 21:53:53 grott grott[9517]:         61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6b
Jul 27 21:53:53 grott grott[9517]:         \x34\x65\x37\x63\x0b
Jul 27 21:54:00 grott grott[9517]: - GrottHttpServer - 127.0.0.1 - "GET /inverter?command=register&register=1091&inverter=NTCIAXXXXX HTTP/1.1" 400 -
Jul 27 21:54:00 grott grott[9517]: - GrottHttpServer - http response send: 400 text/plain b'no or invalid response received\r\n'

cyberdiamond, 2023-07-29:
@bsipos-precognox 

Which datalogger are you using?
Did you update the datalogger and if so are you now locked out of being able to log in locally?

Can you run these commands:

curl -X GET "http://127.0.0.1:5782/datalogger"
curl -X GET "http://127.0.0.1:5782/inverter"

They should return the same information and can you report back the output.

I have a hunch but before I go digging to much it would be good to compare.

cyberdiamond, 2023-07-30:
https://user-images.githubusercontent.com/83278161/247954319-375994d9-e469-4541-83b9-442a425267cf.png

@johanmeijer  @rany2 

In [bsipos-precognox](https://github.com/bsipos-precognox) image above you can see: Device Addr is 080

I am not seeing that in his verbose logs anywhere I only see: Grotthttpserver: selected deviceid : 51

Which I also see in my debug logs but my Device Addr is 081

Reading registers from the inverter does work perfectly when running grott in proxy mode and reading the data from the server.growatt.com web page.

Have either of you known anyone with the grott server working with the ShineLink or has it only with ShineWifi devices?

bsipos-precognox, 2023-07-30:
@cyberdiamond Yes I upgraded the datalogger, and yes I'm now locked out (current firmware is 7.0.0.9). Also, the upgrade didn't solve any of the other issues.

cyberdiamond, 2023-07-30:
See here to get back in: https://github.com/johanmeijer/grott/issues/348

I thought the problem may have been it was using 51 as the inverterid (which in hex is decimal 81 which is my inverter Device Addr) but your logs showed inverterid of 51 also but your Device Addr in your screenshot is 080.

Can you confirm you have the ShineLink setup that has the ShineLanBox that connects to your router?

johanmeijer, 2023-08-02:
I am not sure but there might be a difference is device addresses. Can it be that 50 (80) is the device address of the Shinelink itself and 51(81) of the inverter?

johanmeijer, 2023-08-02:
Can you do a simple get for only http//your_grottserverip/datalogger. Let see what is shows for connect devices and their id? 

Is their a difference in the inverter id you use ([NTCIAXXXXX) and the one shown in your shinelink configuration (DFFxxxxx).

johanmeijer, 2023-08-02:
Id you have a SDM what is the device address of the SDM?

cyberdiamond, 2023-08-05:
$ curl -X GET "http://127.0.0.1:5782/datalogger"
{"NAC5Axxxxx": {"ip": "192.168.1.22", "port": 1029, "protocol": "06", "NTCIAxxxxx": {"inverterno": "51", "power": 0}}}

$ curl -X GET "http://127.0.0.1:5782/inverter"
{"NAC5Axxxxx": {"ip": "192.168.1.22", "port": 1029, "protocol": "06", "NTCIAxxxxx": {"inverterno": "51", "power": 0}}}

That was not my ShineLink details that picture is bsipos-precognox ShineLink details. (you can see his inverter address in the picture is 80 mine is 81).
You can see both in his logs in text attachment above and in my logs it shows both using inverterno: 51

My inverter details as shown in the ShineLink are in this picture:
[image]

cyberdiamond, 2023-08-05:
> If you have a SDM what is the device address of the SDM?

I do have an Eastron SDM230v2 meter connected directly to the sph5000 inverter (via rs485 / modbus) but do not see it listed anywhere in shinelink or as a device on the server.growatt.com website but when I checked directly on the meter in the meterbox it shows on its display Addr 001 (Modbus ID) I think the SPH5000 uses Modbus ID / Addr 002

johanmeijer, 2023-08-05:
I think I found it. Seems to be a timing issue. Grottserver stops waiting on the response to soon. 

You probably see the response if you give the command to display all received register values: 

curl -X GET "http://127.0.0.1:5782/inverter?command=regall&inverter=NTCIAXXXXX

I am going to change the wait time in grottserver. Do you use the python version or the docker image?

johanmeijer, 2023-08-05:
I hace update grottserver.py in the beta version. Can you try it?

cyberdiamond, 2023-08-06:
Thanks, I tried the new version but no different besides taking a bit longer to come back with no response, see logs below:

Grottserver starting log:

 - Grottserver - Version: 0.0.12a
 - GrottHttpserver - Ready to listen at: 0.0.0.0:5782
 - Grottserver - Ready to listen at: 0.0.0.0:5781
 - GrottHttpserver - server listening
 - Grottserver - server listening

Grott starting log:

_Generic:
	version:             	 2.8.2
	verbose:             	 True
	trace:               	 False
	config file:         	 grott.ini
	minrecl:             	 1
	decrypt:             	 True
	compat:              	 False
	invtype:             	 sph
	invtypemap:          	 {}
	include_all:         	 False
	blockcmd:            	 False
	noipf:               	 False
	time:                	 auto
	sendbuf:             	 True
	timezone:            	 local
	valueoffset:         	 6
	offset:              	 6
	inverterid:          	 automatic
	mode:                	 proxy
	grottip              	 default
	grottport            	 5279
_MQTT:
	nomqtt               	 False
	mqttip:              	 192.168.1.12
	mqttport:            	 1883
	mqtttopic:           	 energy/growatt
	mqttmtopic:          	 False
	mqttmtopicname:      	 energy/meter
	mqttinverterintopic: 	 False
	mqtttretain:         	 False
	mqtttauth:           	 True
	mqttuser:            	 mqtt-user
	mqttpsw:             	 **secret**
_Growatt server:
	growattip:           	 0.0.0.0
	growattport:         	 5781
_PVOutput:
	pvoutput:            	 False
	pvdisv1:             	 False
	pvtemp:              	 False
	pvurl:               	 https://pvoutput.org/service/r2/addstatus.jsp
	pvapikey:            	 yourapikey
	pvinverters:         	 1
	pvsystemid:          	 systemid1
_Influxdb:
	influx:             	 True
	influx2:            	 True
	database:           	 grottdb
	ip:                 	 192.168.1.15
	port:               	 8086
	user:               	 grott
	password:           	 **secret**
	organization:       	 influxdata
	bucket:             	 grottdb/autogen
	token:              	 **secret**
_Extension:
	extension:          	 True
	extname:            	 grottext
	extvar:             	 {'none': 'none'}

Grott proxy mode started
Hostname : grott
IP :  192.168.1.14 , port :  5279 

	 - ('192.168.1.22', 1029) has connected

--------------------------------------------------------------------------------------------

Inverter commands that don't work:

Command:

$ curl -X GET "http://127.0.0.1:5782/inverter?command=regall&inverter=NTCIA*****"
{}

Grotterver Log:

	 - Grotthttpserver - Get received 
	 - Grotthttpserver - inverter get received :  {'command': ['regall'], 'inverter': ['NTCIA*****']}
	 - Grotthttpserver: get command:  regall
127.0.0.1 - - [06/Aug/2023 11:47:59] "GET /inverter?command=regall&inverter=NTCIA***** HTTP/1.1" 200 -
	 - Grotthttpserver - http response send:  200 text/body b'{}'

Grott Log:

Nothing logged for this command?

cyberdiamond, 2023-08-06:
Inverter commands that don't work:

Command:

$ curl -X GET "http://127.0.0.1:5782/inverter?command=regall&inverter=NTCIA*****"
{}

Grotterver Log:

	 - Grotthttpserver - Get received 
	 - Grotthttpserver - inverter get received :  {'command': ['regall'], 'inverter': ['NTCIA*****']}
	 - Grotthttpserver: get command:  regall
127.0.0.1 - - [06/Aug/2023 11:47:59] "GET /inverter?command=regall&inverter=NTCIA***** HTTP/1.1" 200 -
	 - Grotthttpserver - http response send:  200 text/body b'{}'

Grott Log:

Nothing logged for this command?

cyberdiamond, 2023-08-06:
Inverter commands that don't work:

Command:

$ curl -X GET "http://127.0.0.1:5782/inverter?inverter=NTCIA*****&command=register&register=15"
no or invalid response received

Grottserver Log:

	 - Grotthttpserver - Get received 
	 - Grotthttpserver - inverter get received :  {'inverter': ['NTCIA*****'], 'command': ['register'], 'register': ['15']}
	 - Grotthttpserver: get command:  register
	 - Grotthttpserver: selected deviceid : 51
	 - Grott - data decrypted V2
	 - Grotthttpserver: command created :
		 \x00\x01\x00\x06\x00\x24\x51\x05\x09\x33\x2c\x42\x20\x45\x40\x75\x45\x57\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x78\x61\x7b\xb1\xc8
	 - Grotthttpserver - wait for GET response
	 - Grottserver - get response from queue:  127.0.0.1_45242 msg: 
		 \x00\x01\x00\x06\x00\x24\x51\x05\x09\x33\x2c\x42\x20\x45\x40\x75\x45\x57\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x78\x61\x7b\xb1\xc8
	 - Grotthttpserver - wait for GET response
	 - Grotthttpserver - wait for GET response
	 - Grotthttpserver - wait for GET response
	 - Grotthttpserver - wait for GET response
	 - Grotthttpserver - wait for GET response
	 - Grottserver - Data received from : 127.0.0.1:45242
	 - Grottserver - Original Data:
		 \x00\xe0\x00\x06\x00\x5e\x51\x29\x09\x33\x2c\x42\x20\x45\x40\x75\x45\x57\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x22\x2e\x22\x4d\x35\x76\x46\x5d\x40\x59\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x2f\x20\x37\x0e\x33\x5e\x40\x51
		 \x47\x42\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x79\xea\xa5\xdb
	 - Grott - data decrypted V2
	 - Grottserver - Plain record: 
		 00e00006005e51294e41433541313432373800000000000000000000000000000000000000004
		 d59433941313432373800000000000000000000000000000000000000004e5443494131373033
		 3600000000000000000000000000000000000000000dadd7b4
	 - Grottserver - 5129 data record received
	 - Grottserver - Response: 
		\x00\xe0\x00\x06\x00\x03\x51\x29\x47\xe9\x0a
	 - Grottserver - Put response on queue:  127.0.0.1_45242  msg: 
		 \x00\xe0\x00\x06\x00\x03\x51\x29\x47\xe9\x0a
	 - Grottserver - get response from queue:  127.0.0.1_45242 msg: 
		 \x00\xe0\x00\x06\x00\x03\x51\x29\x47\xe9\x0a
	 - Grotthttpserver - wait for GET response
	 - Grotthttpserver - wait for GET response
	 - Grotthttpserver - wait for GET response
	 - Grotthttpserver - wait for GET response
	 - Grotthttpserver - wait for GET response
	 - Grotthttpserver - wait for GET response
	 - Grottserver - Data received from : 127.0.0.1:45242
	 - Grottserver - Original Data:
		 \x00\xe1\x00\x06\x00\x20\x01\x16\x09\x33\x2c\x42\x20\x45\x40\x75\x45\x57\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \xf3\xc4
	 - Grott - data decrypted V2
	 - Grottserver - Plain record: 
		 00e10006002001164e41433541313432373800000000000000000000000000000000000000009
		 cb3
	 - Grottserver - 16 - Ping response: 
		 \x00\xe1\x00\x06\x00\x20\x01\x16\x09\x33\x2c\x42\x20\x45\x40\x75\x45\x57\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \xf3\xc4
	 - Grottserver - Put response on queue:  127.0.0.1_45242  msg: 
		 \x00\xe1\x00\x06\x00\x20\x01\x16\x09\x33\x2c\x42\x20\x45\x40\x75\x45\x57\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
[... 78 lines cut ...]
	 - Record layout used :  T060116SPH
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00e10006002001164e41433541313432373800000000000000000000000000000000000000009
		 cb3
	 - Grott data ack record or data record not defined no processing done

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('192.168.1.14', 5279), raddr=('192.168.1.22', 1029)>
	 - Growatt original Data:
		 \x00\xe1\x00\x06\x00\x20\x01\x16\x09\x33\x2c\x42\x20\x45\x40\x75\x45\x57\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \xf3\xc4
	 - Grott automatic protocol detection
	 - Grott data record length 40
	 - layout   :  T060116SPH
	 - no matching record layout found, try generic
	 - Record layout used :  T060116SPH
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00e10006002001164e41433541313432373800000000000000000000000000000000000000009
		 cb3
	 - Grott data ack record or data record not defined no processing done

cyberdiamond, 2023-08-06:
Datalogger command that works just for reference:

Command:

$ curl -X GET "http://127.0.0.1:5782/datalogger?datalogger=NAC5A*****&command=register&register=4"
{"value": "5"}

Grotterver Log:

	 - Grotthttpserver - Get received 
	 - Grotthttpserver - datalogger get received :  {'datalogger': ['NAC5A*****'], 'command': ['register'], 'register': ['4']}
	 - Grotthttpserver: get command:  register
	 - Grott - data decrypted V2
	 - Grotthttpserver: command created :
		 \x00\x01\x00\x06\x00\x24\x01\x19\x09\x33\x2c\x42\x20\x45\x40\x75\x45\x57\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x73\x61\x70\x1b\xfc
	 - Grotthttpserver - wait for GET response
	 - Grottserver - get response from queue:  127.0.0.1_45242 msg: 
		 \x00\x01\x00\x06\x00\x24\x01\x19\x09\x33\x2c\x42\x20\x45\x40\x75\x45\x57\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x73\x61\x70\x1b\xfc
	 - Grottserver - Data received from : 127.0.0.1:45242
	 - Grottserver - Original Data:
		 \x00\x01\x00\x06\x00\x25\x01\x19\x09\x33\x2c\x42\x20\x45\x40\x75\x45\x57\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x73\x61\x75\x41\xe8\x49
	 - Grott - data decrypted V2
	 - Grottserver - Plain record: 
		 00010006002501194e41433541313432373800000000000000000000000000000000000000000
		 004000135af3b
	 - Grottserver - 0119 record received, no response needed
	 - Grotthttpserver - wait for GET response
127.0.0.1 - - [06/Aug/2023 11:52:22] "GET /datalogger?datalogger=NAC5A*****&command=register&register=4 HTTP/1.1" 200 -
	 - Grotthttpserver - http response send:  200 text/body b'{"value": "5"}'

Grott Log:

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('192.168.1.14', 5279), raddr=('192.168.1.22', 1029)>
	 - Growatt original Data:
		 \x00\x01\x00\x06\x00\x24\x01\x19\x09\x33\x2c\x42\x20\x45\x40\x75\x45\x57\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x73\x61\x70\x1b\xfc
	 - Grott automatic protocol detection
	 - Grott data record length 44
	 - layout   :  T060119SPH
	 - no matching record layout found, try generic
	 - Record layout used :  T060119SPH
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00010006002401194e41433541313432373800000000000000000000000000000000000000000
		 00400046fbb
	 - Grott data ack record or data record not defined no processing done

	 - Growatt packet received:
		  <socket.socket fd=5, family=2, type=1, proto=0, laddr=('127.0.0.1', 45242), raddr=('127.0.0.1', 5781)>
	 - Growatt original Data:
		 \x00\x01\x00\x06\x00\x25\x01\x19\x09\x33\x2c\x42\x20\x45\x40\x75\x45\x57\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x73\x61\x75\x41\xe8\x49
	 - Grott automatic protocol detection
	 - Grott data record length 45
	 - layout   :  T060119SPH
	 - no matching record layout found, try generic
	 - Record layout used :  T060119SPH
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00010006002501194e41433541313432373800000000000000000000000000000000000000000
		 004000135af3b
	 - Grott data ack record or data record not defined no processing done

johanmeijer, 2023-08-06:
Hmm Strange. 

Are you able to use the growatt website to create the same get command? Then we can see in the grott log how it looks like. Please specify minrecl=1 in the [Generic] section of grott.ini so we will get all communication information. 

I do not have a shinelink so I can not test it (maybe I can find a cheap second-hand one on the internet, I will have a look) so I need some help. 
Are there others with a shinelink facing the same problems?

cyberdiamond, 2023-08-06:
Using the server.growatt.com website to read inverter:
Start Address 15 End Address 15 then pressing the Advanced Read button.

Grott log:

	 - Growatt packet received:
		  <socket.socket fd=6, family=2, type=1, proto=0, laddr=('192.168.1.14', 5279), raddr=('192.168.1.22', 1032)>
	 - Growatt original Data:
		 \x00\x01\x00\x06\x00\x24\x51\x05\x09\x33\x2c\x42\x20\x45\x40\x75\x45\x57\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x78\x61\x7b\xb1\xc8
	 - Grott automatic protocol detection
	 - Grott data record length 44
	 - layout   :  T065105SPH
	 - no matching record layout found, try generic
	 - Record layout used :  T065105SPH
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00010006002451054e41433541313432373800000000000000000000000000000000000000000
		 00f000fc58f
	 - Grott data ack record or data record not defined no processing done

	 - Growatt packet received:
		  <socket.socket fd=5, family=2, type=1, proto=0, laddr=('192.168.1.14', 35504), raddr=('47.91.67.66', 5279)>
	 - Growatt original Data:
		 \x00\x2c\x00\x06\x00\x26\x51\x05\x09\x33\x2c\x42\x20\x45\x40\x75\x45\x57\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x78\x61\x7b\x74\x46\x6e\xa3
	 - Grott automatic protocol detection
	 - Grott data record length 46
	 - layout   :  T065105SPH
	 - no matching record layout found, try generic
	 - Record layout used :  T065105SPH
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 002c0006002651054e41433541313432373800000000000000000000000000000000000000000
		 00f000f00011ccc
	 - Grott data ack record or data record not defined no processing done

johanmeijer, 2023-08-08:
Ok, Thanks. I am going to look at the data and test with to see how grottserver reacts (and why it not works for you).

johanmeijer, 2023-08-10:
I can not really find the issue. If Grottserver is sending the command. The response is a T065129 record from the inverter. I then it starts a "ping" again.  I do not see that when you doing it via the Growatt server. Then I see the reply with the register value.  

At this moment I always assumed that the "29" records where some kind of announce records and that a simple ack is enough to reply. But maybe it is more and I (grottserver) has to react differently to this record during initialization fase. 

Can you send the grott log when connected to the growatt server during start up (till the first T066129 record has been processed? It would be nice if the inverter api call is also included I can then replay the whole communication. 

I do understand if you do not want to publish this on Github. You can sent it with email to grott@ledidobe.nl.

johanmeijer, 2023-08-19:
@cyberdiamond is it possible to create a grott log with time stamps in it? I like to see how fast the reaction on the commands is.

cyberdiamond, 2023-08-20:
@johanmeijer did you get my first email with the grott log running in proxy mode (that one did have timestamps)?
I have just emailed you again with 2 logs, this time one of grott in server mode, the other of the grottserver both with timestamps. I ran a datalogger and an inverter get command registers 4 & 15 as above in them.

johanmeijer, 2023-08-20:
@cyberdiamond I received them. It is still not totally clear what causing the problem but I have some idea's to change things in Grottserver. I will work on that the coming days.

johanmeijer, 2023-09-04:
@bsipos-precognox @cyberdiamond 

Just published grottserver v0.0.14d. This fixes a bug with the register put command when using a shinlink box.  

It have some other improvements (timing, invalid response filtering) as well. 

Can you please test it.

cyberdiamond, 2023-09-06:
Tested it and reading from the registers on the inverter now works through the shinelanbox mostly works.
When the reading doesn't work with the error message: no or invalid response received
It seems like it is processing a normal proxy record at that time and when I wait for it to finish that and try again it works (not sure if I captured that in any logs, but I can do that if it would help)
Put / writing registers does not work.

I sent you an email with the log files.

Cheers

johanmeijer, 2023-09-06:
Oh ok. Strange it works for me (I have also a shinelink now).

I am going to look at it. 

Thanks for testing.

cyberdiamond, 2023-09-09:
This is now working, I also added pull request #406

johanmeijer, 2023-09-09:
Grottserver now also writes registers to thevinverter if the shinelink logger is used. For nog limitted to register 1024. After the pr from @cyberdiamond above all registers will be possible. I will merge the pr soon.