johanmeijer/grott #359: Divider not applied to MQTT and InfluxDB JSON messages

KevinEeckman, 2023-06-28:
Hi, 

Is there a reason you chose not to apply the dividers descibed in the examples/ folder to the generated MQTT and InfluxDB jsons ?

See [this section](https://github.com/johanmeijer/grott/wiki/Grott-advanced-(customize-behaviour)#change-the-data-processing-and-grott-output-for-mqtt-and-influxdb) of the wiki

johanmeijer, 2023-06-29:
My main reason was (is) to sent the unmodified data to MQTT and Influxdb (the raw data as it was received from the inverter) and do the processing by the reporting tools. The receiver can decide how to process. 

The dividers are only used for Grott log messages at this moment. 

If you think it is of added value we think about making a "switch"(parameter) to also use the dividers in the MQTT and influxdb output.

KevinEeckman, 2023-07-03:
Hi @johanmeijer, implemented in the attached PR as discussed.

johanmeijer, 2023-07-07:
I am going to look into this one the coming weeks (I am enjoying holidays now)