johanmeijer/grott #362: bms_batterycurr for T06NNNNXMIN needs to be parsed as type "numx" instead of "num"

eliasrm87, 2023-07-07:
Hi, 

I'm starting to use this project to integrate my inverter with Home Assistant. As my inverter is a MIN 5000TL-XH, I'm running the beta branch as Linux service.

While reviewing all the values, I found that bms_batterycurr jumps to absurd values, as high as 650 Amps every time the battery (ARK 2.5H-A1 HV) is being discharged.

After a quick look to the configuration for T06NNNNXMIN, I've found that the value for battery current is a signed number, where negative values signify that the battery is discharging and parsing it as "num" is not applying the sign correctly.

I've updated my configuration to "numx" and it is now reporting sensible values.

Thank you very much for all the hard work.

johanmeijer, 2023-07-07:
Thanks, I am going to change that.

johanmeijer, 2023-12-04:
Changed in 2.8.3 (now published)

eliasrm87, 2023-12-04:
Thank you very much!