johanmeijer/grott #363: Big difference between PVoutput and influxdb graphs/values

moedtm, 2023-07-09:
Hello,
I'm experiencing strange differences between MQTT/PVoutput values and influxdb. See the graph attached. The graph in influxdb is still unprocessed, so there is a 10x factor present. Even then, the shape roughly correspond, but do not match. I've selected 5:00pm (17:00) for reference. As you can see the reported value at that time differs hugely: 161 watt vs 2100 watt. The 161 watt matches the value reported in the Shine app so is likely correct.

Aggregating function in influxdb is set to mean, but min/max/median does not make a difference.

Model of the inverter used: MOD 5000 TL3-X

Can you help me debug this further?

[image]
[image]

johanmeijer, 2023-07-09:
What is your update frequecy have you changed that?  Standard pvoutput will only handle one request per 5 minutes. That might flayten things out.

moedtm, 2023-07-10:
I believe no changes in frequency were made, my full grott.ini is

```
[Generic]
invtype=tl3

[Growatt]

[MQTT]
ip = mosquitto
port = 1883
topic = energy/growatt
auth = True
user = grott
password = XX

[PVOutput]
pvoutput = True
apikey = XX
systemid=XX

[influx]
influx = true
influx2 = true
ip = http://influxdb
port = 8086
token  = XX
org  = XX
bucket = XX

[extension]
```