johanmeijer/grott #366: CSV can not be opened with Ms Excel, undefined separator prohibits column creation

geve09, 2023-07-11:
In the Netherlands we use the comma as decimal symbol and a dot as digit grouping symbol, an example: 123.456.789,00
MS Excel uses in this case ";" as a separator. Excel In formulas I have to use the ";" as well. =if(a1=2;0;1).
Opening the comma separated file results in all the data concentrated in column A. Every time you open a csv file in Excel you have to use the text wizard (many clicks) the get the data divided over the columns.
The CSV format support this defined seperator by putting in the first line of the CSV "sep=,"
Add a line to grottcsv.py
```
    with open_makedirs(csvfile, 'a') as f:
        if os.path.getsize(csvfile) == 0:
            f.write('sep=,\n')               # <---
            f.write(csvheader + '\n')
        f.write(csvline)

```
Create new csv and Excel opens with all data spread over the cells as expected.
Note: this might cause problems for other scripts opening the csv, as they do not expect the separator definition line.

johanmeijer, 2023-08-13:
Is this am ral issue that needs to be solved or just for information?

MannyLama, 2023-09-08:
I don't see this as a problem. CSV's get delivered different ways. It is to the consumer to read the CSV correctly. Just import your CSV the correct way into Excel.

A possible solution is mention above but this doesn't seem to me as a real problem.

geve09, 2023-09-10:
You probably have no probem, because your Windows delimeter is equal to comma. Opening the csv with Excel will show all fields in cells. For me everything is packed in the first column. I have to rename the file extension to .txt and use the Excel import wizard every time I open the file.
The problem is the regional settings in Windows (used by Excel) do not comply with the setting used in csv. When creating a csv one should check for the system delimiter and use this. Or override the delimiter with csv parameter "sep=,". Thus Windows (Excel) will understand the layout of the csv.

MannyLama, 2023-09-10:
I do have the same "problem" but there are plenty of options on how to change your delimiter in Excel.
I agree it would be nice to set your delimiter of choose in Grott, but for me it would be a nice to have and it would not provide additional functionality to Grott. 
Please refer to this site for all you possibilities in Excel on how to change your delimiter: https://www.ablebits.com/office-addins-blog/change-excel-csv-delimiter/

johanmeijer, 2023-09-10:
The "challenge"is that I have not created the extension. 

This is contributed by somebody else. I am a little bit hesitant to change.

rany2, 2023-09-10:
@johanmeijer I would suggest you use https://docs.python.org/3/library/csv.html, it is part of python standard library. Creating CSVs like grottcsv.py does is wrong because it doesn't properly escape fields, etc.